home-assistant/operating-system #1975: Failed to Update from HA OS 8.1 to 8.2

jhemmond, 2022-06-14:
### Describe the issue you are experiencing

Updating from HA OS 8.1 to 8.2 from the UI stops without presenting any errors on screen. Here's a log snapshot from Supervisor:

> 22-06-14 17:15:25 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Failed marking slot kernel.0 non-bootable: Failed marking 'kernel.0' as bad: Failed to run grub-editenv: Child process exited with code 1

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

8.1

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Once Home Assistant Update notifications appears, click on Settings and navigate to the notification
2. Hit Install and wait
3. About 2 minutes later, the "Installing" status clears up with no errors, but HA is still on old version (8.1)
...

### Anything in the Supervisor logs that might be useful for us?

```txt
22-06-14 17:14:52 INFO (MainThread) [supervisor.os.manager] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/8.2/haos_generic-x86-64-8.2.raucb
22-06-14 17:14:55 INFO (MainThread) [supervisor.host.info] Updating local host information
22-06-14 17:14:56 INFO (MainThread) [supervisor.host.services] Updating service information
22-06-14 17:14:56 INFO (MainThread) [supervisor.host.network] Updating local network information
22-06-14 17:14:57 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
22-06-14 17:14:57 INFO (MainThread) [supervisor.host.manager] Host information reload completed
22-06-14 17:15:23 INFO (MainThread) [supervisor.os.manager] Completed download of OTA update file /data/tmp/hassos-8.2.raucb
22-06-14 17:15:25 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Failed marking slot kernel.0 non-bootable: Failed marking 'kernel.0' as bad: Failed to run grub-editenv: Child process exited with code 1
```

### Anything in the Host logs that might be useful for us?

```txt
nope. Just repetitive lines:

[896672.789023] loop0: detected capacity change from 0 to 448826
[896672.794188] loop0: detected capacity change from 448826 to 448824
[896852.500261] loop0: detected capacity change from 0 to 448826
[896852.509336] loop0: detected capacity change from 448826 to 448824
[897017.528835] loop0: detected capacity change from 0 to 448826
[897017.533953] loop0: detected capacity change from 448826 to 448824
[897044.948195] audit: type=1334 audit(1655241295.712:1262): prog-id=513 op=LOAD
[897044.948317] audit: type=1334 audit(1655241295.712:1263): prog-id=514 op=LOAD
[897045.456066] audit: type=1334 audit(1655241296.220:1264): prog-id=515 op=LOAD
[897045.456124] audit: type=1334 audit(1655241296.220:1265): prog-id=516 op=LOAD
[897075.099182] loop0: detected capacity change from 0 to 448826
[897075.107075] loop0: detected capacity change from 448826 to 448824
[897075.518114] audit: type=1334 audit(1655241326.283:1266): prog-id=0 op=UNLOAD
[897075.518128] audit: type=1334 audit(1655241326.283:1267): prog-id=0 op=UNLOAD
[897075.865103] audit: type=1334 audit(1655241326.630:1268): prog-id=0 op=UNLOAD
[897075.865118] audit: type=1334 audit(1655241326.630:1269): prog-id=0 op=UNLOAD
```

### System Health information

## System Health

version | core-2022.6.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.12
os_name | Linux
os_version | 5.15.41
arch | x86_64
timezone | America/Iqaluit

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4978
Installed Version | 1.25.5
Stage | running
Available Repositories | 1113
Downloaded Repositories | 3

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.1
-- | --
update_channel | stable
supervisor_version | supervisor-2022.05.3
agent_version | 1.2.1
docker_version | 20.10.14
disk_total | 219.4 GB
disk_used | 4.6 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | SSH & Web Terminal (11.0.1), Mosquitto broker (6.1.2), File editor (5.3.3), Samba share (9.7.0), Zigbee2MQTT (1.25.2-1), AdGuard Home (4.6.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
views | 5
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | June 9, 2022 at 2:13 PM
-- | --
current_recorder_run | June 14, 2022 at 12:33 PM
estimated_db_size | 100.02 MiB
database_engine | sqlite
database_version | 3.34.1

</details>

### Additional information

My host is Dell Wyse 7020. Runs HA comfortably with less than 10% utilization. My upgrades post 7.6 experience have been quite bumpy.

> Going from 7.6 to 8.0. I got hit by the GRUB2 bugs and my instance had to be scraped and restored from backup.
> Naturally, I skipped 8.0, and updated from 7.6 > 8.1. This was fine, but anytime I reboot, GRUB defaults to "Slot A" which contained 7.6, and I had to manually hook a monitor to the box to select Slot B for 8.1 to load

kamm0r, 2022-06-16:
I'm still running OS 7.6 since also 8.1 won't update.
"Check Home Assistant configuration" has run without errors or warning.

## System Health

version | core-2022.6.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.12
os_name | Linux
os_version | 5.10.103-v8
arch | aarch64
timezone | Europe/Rome

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.25.5
Stage | running
Available Repositories | 1113
Downloaded Repositories | 29

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.6
-- | --
update_channel | stable
supervisor_version | supervisor-2022.05.3
agent_version | 1.2.1
docker_version | 20.10.9
disk_total | 28.8 GB
disk_used | 8.0 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.3), Check Home Assistant configuration (3.10.2), Duck DNS (1.15.0), Mosquitto broker (6.1.2), Samba share (9.7.0), Terminal & SSH (9.4.0), Home Assistant Google Drive Backup (0.108.2), CEC Scanner (3.0), InfluxDB (4.4.1), eWeLink Smart Home (1.2.6), ZeroTier One (0.13.2), Grafana (7.5.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 17
views | 10
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 7 giugno 2022 12:05
-- | --
current_recorder_run | 16 giugno 2022 08:36
estimated_db_size | 194.77 MiB
database_engine | sqlite
database_version | 3.34.1

</details>

<details><summary>Sonoff</summary>

version | 3.0.5 (200f243)
-- | --
cloud_online | 2 / 2
local_online | 2 / 2

</details>

Reprakkucik, 2022-06-16:
I'm still running OS 7.6 since also 8.1, 8.2 won't update.

paskyecijano, 2022-06-17:
Same error for me

Home Assistant Operating System update failed with: Installation error: Failed marking slot kernel.1 non-bootable: Failed marking 'kernel.1' as bad: Failed to run grub-editenv: Child process exited with code 1

Any solution??

agners, 2022-06-25:
Can you check the rauc logs by connecting to the underlying host OS (see [SSH access to the host](https://developers.home-assistant.io/docs/operating-system/debugging#ssh-access-to-the-host)), then, after an upgrade attempt, run the command:

```
journalctl -u rauc.service
```

jhemmond, 2022-06-26:
Update @agners 👍
While I was rebooting the host with USB connected to load the SSH Authorized Keys it hit me. Why not update the OS when it was on 7.6 (Slot A) in GRUB? So I did. The update from 7.6 -> 8.2 was successful and now my box reboots correctly into 8.2 every time.

Here's the summary of my issue and workaround:

- Upgrading from 7.6 to 8.0 was a mess. I had to reformat the SSD drive and install 7.6
- Upgrade from 7.6 to 8.1 was successful. However, anytime I rebooted my host GRUB would default to 7.6 which was annoying. I had to manually select GRUB Slot B to load 8.1
- Upgrade from 8.1 to 8.2 would result in the error up in my original bug post above
- Finally, if i let the host choose GRUB Slot A and load 7.6, then upgrade to 8.2 straight, the upgrade was successful. The host is also booting 8.2 (GRUB Slot B) every time I reboot without having to intercept the OS boot menu.

Sorry I couldn't grab the rauc logs for you. You can probably close this bug because my issue is resolved.

astirling01, 2022-06-27:
I’m having the same issue. Cant update From 8.1 to 8.2. Same error

astirling01, 2022-06-27:
Supervisor shows this…

22-06-27 01:39:33 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Failed marking slot kernel.1 non-bootable: Failed marking 'kernel.1' as bad: Failed to run grub-editenv: Child process exited with code 1

astirling01, 2022-07-06:
Can anyone help? Still not updating after successfully updating core. Seems to be exactly the same error as listed above

luisfmrosa, 2022-07-14:
Another one with issues on NUC since HA OS 8.2: my server is a NUC 8iBEH and is located in a remote site. Since upgrade to 8.2 on 09/07/2022, server is down. I came today to the site and I'm checking it: when I boot it, either I get black screen, or it stalls in startup menu (F2 to Enter Setup, F7 to Update BIOS, F10 to Enter Boot Menu).

I tried to flash 8.2 from website with the same results. I will try again before flashing 8.1 ...

balilu91, 2022-07-27:
Having same issue. Coudn't update from 8.1 to 8.2 and now I have the same issue to update from 8.1 (still) to 8.4

`Home Assistant Operating System update failed with: Installation error: Failed marking slot kernel.1 non-bootable: Failed marking 'kernel.1' as bad: Failed to run grub-editenv: Child process exited with code 1`

Running on NUC 8iBEH

## System Health

version | core-2022.7.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.41
arch | x86_64
timezone | Europe/Malta
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4608
Installed Version | 1.26.2
Stage | running
Available Repositories | 1152
Downloaded Repositories | 30

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.1
-- | --
update_channel | stable
supervisor_version | supervisor-2022.07.0
agent_version | 1.2.1
docker_version | 20.10.14
disk_total | 109.3 GB
disk_used | 69.8 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Node-RED (13.2.0), InfluxDB (4.5.0), Studio Code Server (5.2.0), Grafana (7.6.0), Duck DNS (1.15.0), MariaDB (2.5.0), Mosquitto broker (6.1.2), Home Assistant Google Drive Backup (0.108.2), SSH & Web Terminal (11.0.2), Samba share (10.0.0), FTP (4.6.0), phpMyAdmin (0.8.1), Vaultwarden (Bitwarden) (0.17.1), Nginx Proxy Manager (0.12.1), Matrix (0.10.0), Glances (0.16.0), Frigate NVR (Full Access) (3.1), Promtail (2.1.0), Loki (1.10.2), Uptime Kuma (0.2.0), chrony (2.4.0), ESPHome (2022.6.2), AdGuard Home (4.6.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 19
views | 8
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | July 12, 2022 at 12:48 PM
-- | --
current_recorder_run | July 27, 2022 at 10:08 PM
estimated_db_size | 2045.61 MiB
database_engine | mysql
database_version | 10.6.8

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

bobcroft, 2022-07-29:
I too am having the same problem on my NUC running HassOS, yet my two Wyse thin clients and also running HassOS updated to 8.4 with no issues.  Has anyone found a solution yet please.  The update concerned is 8.2 to 8.4

jhemmond, 2022-07-29:
Can you explain what versions do you have installed on GRUB?

I found that if a box has 8.1 and 8.2, then manually loading 8.1 from GRUB
the upgrading straight to 8.4 works better than upgrading fro. 8.2 to 8.4.
let me know if that works.

astirling01, 2022-07-29:
Are you able to point me to a step by step guide as to how to do this? I don’t know what a GRUB is :D:D

jhemmond, 2022-07-29:
I'll try :) I wish I could make a video.

1) Reboot the HA Host (not HA, the machine). Google should have step by
step from the UI

2) Make sure you have a display connected to your host when the rebooting
occurs. The GRUB menu should appear showing Slot A and Slot B

3) Choose one and see if it is the lower version. Try until you get the
lower one

4) Once the host is booted and HA is up, go through your normal upgrade
process

Let me know how it goes!

balilu91, 2022-07-29:
I managed to fix it based on the comments from agners in [2031](https://github.com/home-assistant/operating-system/issues/2031)

agners, 2022-08-04:
Closing as the fix in #2036 should fix this problem. I'll backport that fix to the next release 8.

closed 2022-08-04 by agners

jhemmond, 2022-08-04:
Thank you for the fix!