home-assistant/operating-system #1978: Upgrade from HAos 7.6 to 8.x not possible

oliverbuschner, 2022-06-19:
### Describe the issue you are experiencing

Upgrade from HAOS 7.6 to any 8.x version is not possible.

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

7.6

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. click on "install" button for OS update
2. wait for reboot
3. still old version installed

### Anything in the Supervisor logs that might be useful for us?

```txt
not as far as I see
```

### Anything in the Host logs that might be useful for us?

```txt
not as far as I see
```

muntoris, 2022-06-19:
I wonder, if there is a way to manual update the HA OS.

agners, 2022-06-25:
Can you post the Supervisor logs after pressing the upgrade button?

wingedseed, 2022-06-27:
> Can you post the Supervisor logs after pressing the upgrade button?

22-06-27 20:32:04 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Failed updating slot boot.0: failed to run slot hook: Child process exited with code 1

muntoris, 2022-06-27:
here we go .. supervisor log after an upgrade try:

22-06-27 14:49:56 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-audio with version 2022.05.0
22-06-27 14:49:56 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin
22-06-27 14:49:58 INFO (SyncWorker_0) [supervisor.docker.audio] Starting Audio ghcr.io/home-assistant/aarch64-hassio-audio with version 2022.05.0 - 172.30.32.4
22-06-27 14:49:58 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-observer with version 2021.10.0
22-06-27 14:49:58 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-multicast with version 2022.02.0
22-06-27 14:49:58 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin
22-06-27 14:49:59 INFO (SyncWorker_1) [supervisor.docker.multicast] Starting Multicast ghcr.io/home-assistant/aarch64-hassio-multicast with version 2022.02.0 - Host
22-06-27 14:49:59 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
22-06-27 14:50:00 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 1 Home Assistant secrets
22-06-27 14:50:00 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/raspberrypi4-64-homeassistant with version 2022.6.7
22-06-27 14:50:01 INFO (MainThread) [supervisor.os.manager] Detect Home Assistant Operating System 7.6 / BootSlot A
22-06-27 14:50:02 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/3a26b21d repository
22-06-27 14:50:02 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
22-06-27 14:50:02 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
22-06-27 14:50:02 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/75a80a57 repository
22-06-27 14:50:04 INFO (MainThread) [supervisor.store] Loading add-ons from store: 64 all - 64 new - 0 remove
22-06-27 14:50:04 INFO (MainThread) [supervisor.addons] Found 9 installed add-ons
22-06-27 14:50:04 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-samba with version 10.0.0
22-06-27 14:50:04 INFO (SyncWorker_3) [supervisor.docker.interface] Attaching to bropat/eufy-security-ws with version 0.9.2
22-06-27 14:50:04 INFO (SyncWorker_2) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-letsencrypt with version 4.12.5
22-06-27 14:50:04 INFO (SyncWorker_5) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/node-red/aarch64 with version 12.0.2
22-06-27 14:50:04 INFO (SyncWorker_4) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-ssh with version 9.4.0
22-06-27 14:50:04 INFO (SyncWorker_6) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/influxdb/aarch64 with version 4.5.0
22-06-27 14:50:04 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to aler9/rtsp-simple-server with version v0.17.6
22-06-27 14:50:04 INFO (SyncWorker_7) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/grafana/aarch64 with version 7.6.0
22-06-27 14:50:04 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-configurator with version 5.3.3
22-06-27 14:50:04 INFO (MainThread) [supervisor.backups.manager] Found 97 backup files
22-06-27 14:50:04 INFO (MainThread) [supervisor.discovery] Loaded 0 messages
22-06-27 14:50:04 INFO (MainThread) [supervisor.ingress] Loaded 0 ingress sessions
22-06-27 14:50:04 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.SETUP
22-06-27 14:50:04 INFO (MainThread) [supervisor.resolution.check] System checks complete
22-06-27 14:50:04 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP
22-06-27 14:50:04 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-06-27 14:50:04 INFO (MainThread) [supervisor.jobs] 'ResolutionFixup.run_autofix' blocked from execution, system is not running - CoreState.SETUP
22-06-27 14:50:04 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP
22-06-27 14:50:04 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-06-27 14:50:04 INFO (MainThread) [__main__] Running Supervisor
22-06-27 14:50:04 INFO (MainThread) [supervisor.os.manager] Rauc: A - marked slot kernel.0 as good
22-06-27 14:50:04 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.INITIALIZE' starting 0 add-ons
22-06-27 14:50:04 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SYSTEM' starting 0 add-ons
22-06-27 14:50:04 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SERVICES' starting 3 add-ons
22-06-27 14:50:04 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-06-27 14:50:04 WARNING (SyncWorker_0) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
22-06-27 14:50:06 INFO (SyncWorker_0) [supervisor.docker.addon] Starting Docker add-on homeassistant/aarch64-addon-samba with version 10.0.0
22-06-27 14:50:07 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
[... cut ...]
22-06-27 14:52:39 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
22-06-27 14:52:39 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
22-06-27 14:52:39 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
22-06-27 14:52:39 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
22-06-27 14:52:39 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running

muntoris, 2022-06-27:
also the host log:

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.10.103-v8 (builder@f0f12117096a) (aarch64-buildroot-linux-gnu-gcc.br_real (Buildroot -g8d88bd9a3) 9.4.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP PREEMPT Mon Mar 28 11:44:31 UTC 2022
[    0.000000] random: fast init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] On node 0 totalpages: 1012736
[    0.000000]   DMA zone: 3792 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242688 pages, LIFO batch:63
[    0.000000]   DMA32 zone: 12032 pages used for memmap
[    0.000000]   DMA32 zone: 770048 pages, LIFO batch:63
[    0.000000] percpu: Embedded 32 pages/cpu s91544 r8192 d31336 u131072
[    0.000000] pcpu-alloc: s91544 r8192 d31336 u131072 alloc=32*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 996912
[    0.000000] Kernel command line: zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor rootwait systemd.machine_id=0d5c515dcae14dfe8f83a3c9bee22af7 cgroup_enable=memory fsck.repair=yes coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:42:26:CE vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootfstype=squashfs ro rauc.slot=A
[    0.000000] cgroup: Enabling memory control group subsystem
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x0000000037400000-0x000000003b400000] (64MB)
[    0.000000] Memory: 3815184K/4050944K available (12288K kernel code, 1994K rwdata, 4228K rodata, 3904K init, 1274K bss, 170224K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 39661 entries in 155 pages
[    0.000000] ftrace: allocated 155 pages with 5 groups
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x3b0/0x574 with crng_init=1
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000272] Console: colour dummy device 80x25
[    0.000843] printk: console [tty1] enabled
[    0.000901] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000943] pid_max: default: 32768 minimum: 301
[    0.001095] LSM: Security Framework initializing
[    0.001217] AppArmor: AppArmor initialized
[    0.001435] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.001514] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.004668] rcu: Hierarchical SRCU implementation.
[    0.005726] EFI services will not be available.
[    0.006259] smp: Bringing up secondary CPUs ...
[    0.007235] Detected PIPT I-cache on CPU1
[    0.007309] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.008473] Detected PIPT I-cache on CPU2
[... 613 lines cut ...]
[  174.750348] ext4 filesystem being remounted at /run/systemd/unit-root/etc/dropbear supports timestamps until 2038 (0x7fffffff)
[  175.083724] audit: type=1334 audit(1656334316.972:198): prog-id=17 op=LOAD
[  175.083860] audit: type=1334 audit(1656334316.972:199): prog-id=18 op=LOAD
[... cut ...]
[  175.624356] ext4 filesystem being remounted at /run/systemd/unit-root/etc/systemd/timesyncd.conf supports timestamps until 2038 (0x7fffffff)
[  175.624405] ext4 filesystem being remounted at /run/systemd/unit-root/etc/modules-load.d supports timestamps until 2038 (0x7fffffff)
[  205.125993] audit: type=1334 audit(1656334347.016:200): prog-id=16 op=UNLOAD
[  205.126019] audit: type=1334 audit(1656334347.016:201): prog-id=15 op=UNLOAD
[  206.007250] audit: type=1334 audit(1656334347.896:202): prog-id=18 op=UNLOAD
[  206.007274] audit: type=1334 audit(1656334347.896:203): prog-id=17 op=UNLOAD

muntoris, 2022-06-28:
I have the feeling, that when I click on the "upgrade" button, just nothing happens in the backend. The system reboots and thats it.
I also do not have the option to create an backup before upgrading. Is this normal? For every other upgrade/update I have that option.

rocheuf, 2022-06-29:
Hello,
Almost the same for me 
- Rpi4-64 (with SSD)
- Home assistant OS
- Stuck on 7.6 (no notification to update 8.x)

here is the supervisor log after update try 

22-06-29 12:00:38 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
22-06-29 12:00:38 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/sabeechen/hassio-google-drive-backup repository
22-06-29 12:00:38 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/home-assistant-addon repository
22-06-29 12:00:38 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/zigbee2mqtt/hassio-zigbee2mqtt repository
22-06-29 12:00:38 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository
22-06-29 12:00:38 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
22-06-29 12:00:41 INFO (MainThread) [supervisor.store] Loading add-ons from store: 68 all - 0 new - 0 remove
22-06-29 12:00:41 INFO (MainThread) [supervisor.store] Loading add-ons from store: 68 all - 0 new - 0 remove
22-06-29 12:03:43 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token

muntoris, 2022-07-09:
Nobody a suggestion or solution? I really do not want to setup the complete system from scratch ...

muntoris, 2022-07-16:
Not so easy to grab the logs BEFORE reboot.
I did now an update to version 8.0 (for trying) and the last lines before the reboot:

22-07-16 19:53:54 INFO (MainThread) [supervisor.api.middleware.security] /os/update access from core_ssh
22-07-16 19:53:54 INFO (MainThread) [supervisor.os.manager] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/8.0/haos_rpi4-64-8.0.raucb
22-07-16 19:54:03 INFO (MainThread) [supervisor.os.manager] Completed download of OTA update file /data/tmp/hassos-8.0.raucb

Are the log files somewhere stored or after the reboot overwritten?

Nafania, 2022-07-18:
I have the same issue

Here is the logs
`22-07-18 16:39:53 INFO (MainThread) [supervisor.os.manager] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/8.2/haos_rpi4-64-8.2.raucb
22-07-18 16:40:00 INFO (MainThread) [supervisor.os.manager] Completed download of OTA update file /data/tmp/hassos-8.2.raucb
22-07-18 16:40:24 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Failed updating slot boot.0: failed to run slot hook: Child process exited with code 1`

Tried to reboot host / supervisor - no changes.
Tried ha supervisor repair command - also didn't help.

agners, 2022-07-19:
> Are the log files somewhere stored or after the reboot overwritten?

The frontend only shows logs since the last boot (and a maximum of 100 lines for most log types).

However, underneath more logs are stored, but currently SSH access is required to access them.

If someone has [SSH access to the underlying OS](https://developers.home-assistant.io/docs/operating-system/debugging#ssh-access-to-the-host), please try to get the following with the following commands:

```
journalctl -u rauc.service
journalctl CONTAINER_NAME=hassio_supervisor --since yesterday
```

czrogers13, 2022-07-19:
I'm having the same issue, and have been since before 7.6. I'm on rpi3-64 as well and the only solution I've found to update is to wipe and reflash the SD card, which totally overwrites my current setup. Hoping to find a solution that doesn't involve upgrading hardware.

czrogers13, 2022-07-23:
@agners sorry for the delay here, took me a couple days to get around to getting the SSH set up. I also made the mistake[?] of trying the update from the SSH CLI before I tested it front end again. The interesting news is that the update did actually work from CLI, so apparently it's not a hardware thing. I'm now on 8.2, which had failed from the front end.

I went ahead and pulled the two command responses you asked for and have them below. They were pulled after the CLI update went through, so they may or may not be helpful for troubleshooting. Unfortunately, I've ruled myself out of front-end update testing until another update drops, so let me know if I can help in the meantime.

Command 1:

`May 10 14:37:43 homeassistant rauc[358]: rauc mark: marked slot kernel.0 as good
-- Boot 0ecab851763248ac8ac018c4fe53fcf4 --
May 16 11:47:53 homeassistant systemd[1]: Starting Rauc Update Service...
May 16 11:47:54 homeassistant rauc[339]: Using central status file /mnt/data/rauc.db
May 16 11:47:54 homeassistant systemd[1]: Started Rauc Update Service.
May 16 11:50:38 homeassistant rauc[339]: rauc mark: marked slot kernel.0 as good
-- Boot d5c2ba78ef3f47c18162f856dcea78d6 --
May 16 20:29:55 homeassistant rauc[334]: rauc mark: marked slot kernel.0 as good
May 23 22:18:29 homeassistant rauc[334]: rauc mark: marked slot kernel.0 as good
-- Boot 0ecab851763248ac8ac018c4fe53fcf4 --
Jun 08 01:43:48 homeassistant rauc[339]: Using deprecated 'Install' D-Bus Method (replaced by 'InstallBundle')
Jun 08 01:43:48 homeassistant rauc[339]: input bundle: /mnt/data/supervisor/tmp/hassos-8.1.raucb
Jun 08 01:43:48 homeassistant rauc[339]: Active slot bootname: A
Jun 08 01:43:48 homeassistant rauc[339]: installing /mnt/data/supervisor/tmp/hassos-8.1.raucb: started
Jun 08 01:43:48 homeassistant rauc[339]: installing /mnt/data/supervisor/tmp/hassos-8.1.raucb: Checking and mounting bundle...
Jun 08 01:43:48 homeassistant rauc[339]: Reading bundle: /mnt/data/supervisor/tmp/hassos-8.1.raucb
Jun 08 01:43:48 homeassistant rauc[339]: Detected CRL but CRL checking is disabled!
Jun 08 01:43:48 homeassistant rauc[339]: Verifying bundle signature... 
Jun 08 01:43:52 homeassistant rauc[339]: Verified detached signature by 'O = HassOS, CN = HassOS GitHub-1'
Jun 08 01:43:52 homeassistant rauc[339]: Mounting bundle '/mnt/data/supervisor/tmp/hassos-8.1.raucb' to '/run/rauc/bundle'
Jun 08 01:43:52 homeassistant rauc[339]: Configured loop device '/dev/loop0' for 117989376 bytes
Jun 08 01:43:52 homeassistant rauc[339]: Running bundle hook install-check
Jun 08 01:43:52 homeassistant rauc[339]: Marking target slot kernel.1 as non-bootable...
Jun 08 01:43:52 homeassistant rauc[339]: installing /mnt/data/supervisor/tmp/hassos-8.1.raucb: Updating slots...
Jun 08 01:43:52 homeassistant rauc[339]: installing /mnt/data/supervisor/tmp/hassos-8.1.raucb: Checking slot boot.0
Jun 08 01:43:53 homeassistant rauc[339]: Slot needs to be updated with /run/rauc/bundle/boot.vfat
Jun 08 01:43:53 homeassistant rauc[339]: Updating /dev/disk/by-partlabel/hassos-boot with /run/rauc/bundle/boot.vfat
Jun 08 01:43:53 homeassistant rauc[339]: Running custom slot install hook for boot.0
Jun 08 01:43:53 homeassistant rauc[339]: Running slot hook slot-install for boot.0
Jun 08 01:43:53 homeassistant rauc[339]: installing /mnt/data/supervisor/tmp/hassos-8.1.raucb: Updating slot boot.0
Jun 08 01:44:25 homeassistant rauc[339]: installing /mnt/data/supervisor/tmp/hassos-8.1.raucb: Updating slot boot.0 status
Jun 08 01:44:25 homeassistant rauc[339]: Checking image type for slot type: raw
Jun 08 01:44:25 homeassistant rauc[339]: installing /mnt/data/supervisor/tmp/hassos-8.1.raucb: Updating slot boot.0 done
Jun 08 01:44:25 homeassistant rauc[339]: Image detected as type: *.img
Jun 08 01:44:25 homeassistant rauc[339]: Slot needs to be updated with /run/rauc/bundle/kernel.img
Jun 08 01:44:25 homeassistant rauc[339]: Updating /dev/disk/by-partlabel/hassos-kernel1 with /run/rauc/bundle/kernel.img
Jun 08 01:44:25 homeassistant rauc[339]: opening slot device /dev/disk/by-partlabel/hassos-kernel1
Jun 08 01:44:25 homeassistant rauc[339]: installing /mnt/data/supervisor/tmp/hassos-8.1.raucb: Checking slot kernel.1
Jun 08 01:44:25 homeassistant rauc[339]: installing /mnt/data/supervisor/tmp/hassos-8.1.raucb: Updating slot kernel.1
Jun 08 01:44:25 homeassistant rauc[339]: writing data to device /dev/disk/by-partlabel/hassos-kernel1
Jun 08 01:44:30 homeassistant rauc[339]: installing /mnt/data/supervisor/tmp/hassos-8.1.raucb: Updating slot kernel.1 status
Jun 08 01:44:30 homeassistant rauc[339]: Checking image type for slot type: raw
Jun 08 01:44:30 homeassistant rauc[339]: installing /mnt/data/supervisor/tmp/hassos-8.1.raucb: Updating slot kernel.1 done
Jun 08 01:44:30 homeassistant rauc[339]: Image detected as type: *.img
Jun 08 01:44:30 homeassistant rauc[339]: Slot needs to be updated with /run/rauc/bundle/rootfs.img
Jun 08 01:44:30 homeassistant rauc[339]: installing /mnt/data/supervisor/tmp/hassos-8.1.raucb: Checking slot rootfs.1
Jun 08 01:44:30 homeassistant rauc[339]: Updating /dev/disk/by-partlabel/hassos-system1 with /run/rauc/bundle/rootfs.img
Jun 08 01:44:30 homeassistant rauc[339]: opening slot device /dev/disk/by-partlabel/hassos-system1
Jun 08 01:44:30 homeassistant rauc[339]: installing /mnt/data/supervisor/tmp/hassos-8.1.raucb: Updating slot rootfs.1
Jun 08 01:44:30 homeassistant rauc[339]: writing data to device /dev/disk/by-partlabel/hassos-system1
[... cut ...]
Jul 22 20:02:39 homeassistant systemd[1]: Stopped Rauc Update Service.
-- Boot c8cbd20124af4717a1ed1f8e4c2e65e5 --
Jul 22 20:02:46 homeassistant systemd[1]: Starting Rauc Update Service...
Jul 22 20:02:47 homeassistant rauc[360]: Using central status file /mnt/data/rauc.db
Jul 22 20:02:47 homeassistant rauc[360]: Using central status file /mnt/data/rauc.db
Jul 22 20:02:47 homeassistant systemd[1]: Started Rauc Update Service.
Jul 22 20:06:09 homeassistant rauc[360]: rauc mark: marked slot kernel.1 as good'

agners, 2022-07-27:
@czrogers13 At least on your last reboot, it seems the system rebooted into the new OS. Are you sure that updating failed?

Can you paste output of `ha os info` before and after the update?

saeedesmaili, 2022-08-05:
I have the same issue, and while running the aforementioned commands, it seems like I don't have `systemd` installed. How can I run those commands on my raspberry pi?

```
~ journalctl -u rauc.service
zsh: command not found: journalctl

~ apt-get install systemd
zsh: command not found: apt-get
```

ppohns, 2022-08-12:
same issue here - trying to upgrade from 7.6 to 8.4.

Nafania, 2022-08-30:
Here is my log

```
Aug 30 15:47:34 homeassistant rauc[376]: Using deprecated 'Install' D-Bus Method (replaced by 'InstallBundle')
Aug 30 15:47:34 homeassistant rauc[376]: Active slot bootname: A
Aug 30 15:47:34 homeassistant rauc[376]: input bundle: /mnt/data/supervisor/tmp/hassos-8.5.raucb
Aug 30 15:47:34 homeassistant rauc[376]: installing /mnt/data/supervisor/tmp/hassos-8.5.raucb: started
Aug 30 15:47:34 homeassistant rauc[376]: installing /mnt/data/supervisor/tmp/hassos-8.5.raucb: Checking and mounting bundle...
Aug 30 15:47:34 homeassistant rauc[376]: Reading bundle: /mnt/data/supervisor/tmp/hassos-8.5.raucb
Aug 30 15:47:34 homeassistant rauc[376]: Detected CRL but CRL checking is disabled!
Aug 30 15:47:34 homeassistant rauc[376]: Verifying bundle signature...
Aug 30 15:47:35 homeassistant rauc[376]: Verified detached signature by 'O = HassOS, CN = HassOS GitHub-1'
Aug 30 15:47:35 homeassistant rauc[376]: Mounting bundle '/mnt/data/supervisor/tmp/hassos-8.5.raucb' to '/run/rauc/bundle'
Aug 30 15:47:35 homeassistant rauc[376]: Configured loop device '/dev/loop9' for 121425920 bytes
Aug 30 15:47:35 homeassistant rauc[376]: Running bundle hook install-check
Aug 30 15:47:35 homeassistant rauc[376]: Marking target slot kernel.1 as non-bootable...
Aug 30 15:47:35 homeassistant rauc[376]: installing /mnt/data/supervisor/tmp/hassos-8.5.raucb: Updating slots...
Aug 30 15:47:35 homeassistant rauc[376]: installing /mnt/data/supervisor/tmp/hassos-8.5.raucb: Checking slot boot.0
Aug 30 15:47:35 homeassistant rauc[376]: Slot needs to be updated with /run/rauc/bundle/boot.vfat
Aug 30 15:47:35 homeassistant rauc[376]: Updating /dev/disk/by-partlabel/hassos-boot with /run/rauc/bundle/boot.vfat
Aug 30 15:47:35 homeassistant rauc[376]: installing /mnt/data/supervisor/tmp/hassos-8.5.raucb: Updating slot boot.0
Aug 30 15:47:35 homeassistant rauc[376]: Running custom slot install hook for boot.0
Aug 30 15:47:35 homeassistant rauc[376]: Running slot hook slot-install for boot.0
Aug 30 15:47:36 homeassistant rauc[200951]: cp: write error: No space left on device
Aug 30 15:47:36 homeassistant rauc[200951]: cp: write error: No space left on device
Aug 30 15:47:36 homeassistant rauc[200951]: cp: write error: No space left on device
Aug 30 15:47:36 homeassistant rauc[200951]: cp: write error: No space left on device
Aug 30 15:47:37 homeassistant rauc[200951]: cp: write error: No space left on device
Aug 30 15:47:37 homeassistant rauc[200951]: cp: write error: No space left on device
Aug 30 15:47:38 homeassistant rauc[200951]: cp: write error: No space left on device
Aug 30 15:47:38 homeassistant rauc[200951]: cp: write error: No space left on device
Aug 30 15:47:39 homeassistant rauc[200951]: cp: write error: No space left on device
Aug 30 15:47:39 homeassistant rauc[200951]: cp: write error: No space left on device
Aug 30 15:47:39 homeassistant rauc[200951]: cp: write error: No space left on device
Aug 30 15:47:39 homeassistant rauc[200951]: cp: write error: No space left on device
Aug 30 15:47:40 homeassistant rauc[200951]: cp: write error: No space left on device
Aug 30 15:47:47 homeassistant rauc[200951]: cp: write error: No space left on device
Aug 30 15:47:54 homeassistant rauc[376]: Installation error: Failed updating slot boot.0: failed to run slot hook: Child process exited with code 1
Aug 30 15:47:54 homeassistant rauc[376]: installing /mnt/data/supervisor/tmp/hassos-8.5.raucb: Installation error: Failed updating slot boot.0: failed to run slot hook: Child process exited with code 1
Aug 30 15:47:54 homeassistant rauc[376]: installing /mnt/data/supervisor/tmp/hassos-8.5.raucb: finished
Aug 30 15:47:54 homeassistant rauc[376]: installing `/mnt/data/supervisor/tmp/hassos-8.5.raucb` failed: 1
```

```
# df -h /mnt/data/supervisor/tmp/
Filesystem                Size      Used Available Use% Mounted on
/dev/mmcblk0p8           58.7G      9.7G     46.7G  17% /mnt/data
```

Nafania, 2022-10-06:
@agners Any ideas on how i can help you to investigate more?

paavo, 2022-12-10:
did someone found a solution / workaround?

ppohns, 2022-12-12:
Same here - reinstall.

agners, 2022-12-13:
It seems the boot partition run full, I wonder why that is. If someone observes it again, and has shell acces, can you check `df -h` and check the contents of `/mnt/boot`?

closed 2023-03-20 by github-actions[bot]

nonyhaha, 2024-07-25:
Did anybody manage to make this work? Without a backup and reinstall? Running this on a vm, I am not at home, and I would like it to work without too much time spent, especially without anything going wrong...