home-assistant/operating-system #1979: Odroid XU4 wont boot from eMMC

shajek, 2022-06-20:
### Describe the issue you are experiencing

Hello,

i have latest HaOS image (8.2) with Odroid XU4 with red emmc and it wont boot (only solid blue led). I try flash emmc with :

- ubuntu - work
- armbian  - work
- emmc recovery from hardkernel and android work. 
 
But when i flash HaOS image with balena to emmc it doesnt boot (from microsd it works, but i have lot of emmcs)

i see (https://github.com/home-assistant/operating-system/blob/dev/Documentation/boards/hardkernel/odroid-xu4.md) but in CLI i doesnt have DD

Thank you

### What operating system image do you use?

odroid-xu4 (Hardkernel ODROID-XU4)

### What version of Home Assistant Operating System is installed?

8.2

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. download image
2. flash to eMMC card

1. flash recovery image to emmc card
2. flash HaOS

still same result

### Anything in the Supervisor logs that might be useful for us?

```txt
-
```

### Anything in the Host logs that might be useful for us?

```txt
-
```

alex7614, 2022-06-27:
Hi
I have the same problem with N2. 

It worked perfectly at the beginning and following an OTA update (v7), the N2 stopped starting on the eMMC module but on the other hand, works with the eMMC in the SD CARD module.

I tried to reflash the old version with etcher at the time but it doesn't work anymore.

I also tested other versions which worked (v5.X and v6.x) and same thing. I tried to rewrite 0 on the whole module and same problem.

But like you, it works with Ubuntu or Armbian.

I personally think of one or more bad sectors of the module which cause this problem with HassOS image. I have never tried to reflash to another eMMC as I now use the eMMC in the SD CARD module.

shajek, 2022-06-27:
Bad sectors are not issue. I have 5+ new eMMC (red PCB with blue dot) and tens of used and all have same problem. Only way to make work eMMC for me is to flash recovery image from Hardkernel (to SD and boot it to write to eMMC), then flash SD and eMMC with same image of HaOS. When OS boots, type login to access to bash, and make DD from [THIS links](https://github.com/home-assistant/operating-system/blob/dev/Documentation/boards/hardkernel/odroid-xu4.md) (with haos 6.6 without error and with 8.2 with error during writing, but it works)

alex7614, 2022-06-27:
And after, when you update HassOS from OTA, does the eMMC still boot or do you have to reflash with the above method?

shajek, 2022-06-27:
When i flash 6.6. I upgrade it from HA (or OTA as you preffer) to 7.6 and next 8.2 without issue and normally booting.
When i flash 8.2 I dont have what to upgrade :) and wait for next release.

alex7614, 2022-06-27:
Thanks
I tried:
Flash with etcher HassOS 8.2 on SD and eMMC
Boot from SD with eMMC plug in
After login, I tried "dd if=/dev/mmcblk1 of=/dev/mmcblk0boot0 bs=512 skip=1 seek=0 count=16381"
Unplug SD and boot only on eMMC = not work (red)

Tried same but with "dd if=/dev/mmcblk0 of=/dev/mmcblk0boot0 bs=512 skip=63 seek=62 count=1440"
eMMC not work (red)

Maybe this method doesn't work with the N2??

shajek, 2022-06-27:
I mention it [here](https://github.com/home-assistant/operating-system/issues/389#issuecomment-1165777126) in my case when I boot with with eMMC and SD pluged, i have mmcblk0 as SD and mmcblk1 as eMMC - oposite of that example.
When i flash something else instead of recovery image on eMMC before HaOS it dosent work. Every time i must flash that doomed android before i can flash HaOS succesfully

alex7614, 2022-06-28:
In my case with N2, eMMC is mmcblk1 too. But commands not help for boot from eMMC.

However, it works in a strange way that I don't know how to explain....
If I insert the SD card at 100% in slot (SD card with HassOS on it), HassOS boots on the eMMC and the SD card is not recognized neither on HassOS nor on PETitboot (maybe a PIN problem ????)

If I insert the SD card 90% deep, HassOS boots on the SD card and both the SD card is detected in HassOS as well as the eMMC.

If on the other hand I totaly remove the SD card or if I insert a formatted SD card at 100%, no boot on the eMMC, red light...

Frankly I don't understand at all but suddenly it works......

closed 2022-10-03 by github-actions[bot]