home-assistant/operating-system #1980: RTC Clock Initialized Too Soon in Boot for ODroid N2

nbetcher, 2022-06-20:
### Describe the issue you are experiencing

Once the ODroid-N2 finishes booting, the clock is off due to the RTC clock not being used. The problem is that the RTC clock is compiled as a module, but also relies on I2C (which is also a module). As such, the RTC cannot be used at the command line once the OS is booted, unless the modules are removed from memory and reloaded; unfortunately that causes /dev/rtc0 to remain (non-functionally) and /dev/rtc1 to be created.

This issue (and the errors it produces for me) is described here: https://forum.odroid.com/viewtopic.php?p=326351#p326351

Please compile the ODroid N2's kernel with:

```
CONFIG_AMLOGIC_I2C_MASTER = y
CONFIG_RTC_DRV_PCF8563 = y
```

Or it may now be something like this:

```
CONFIG_AMLOGIC_I2C_MASTER = y
CONFIG_RTC_DRV_**MESON_VRTC** = y
```

Or create udev rules and a systemd unit as described here: https://wiki.odroid.com/accessory/add-on_boards/rtc_shield#configuration

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

8.2

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Flash image on ODroid N2 (**note: I am NOT using ODroid N2+**);
2. Boot device;
3. Note that http://homeassistant.local never finishes saying that it's "preparing";
4. Open developer tools, look at 'log' HTTP call;
5. Note that in the logs the clock is off;
6. Open console to device;
7. login;
8. look at dmesg and the "Invalid argument" errors produced.
9. Commands to hwclock always fail stating battery is low (my battery is NOT low; DMM measures the connection **TO THE MOTHERBOARD** as 3.36 volts) and while controlling the RTC it receives and, "Invalid argument"
...

### Anything in the Supervisor logs that might be useful for us?

```txt
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/udev.sh
[07:35:09] INFO: Using udev information from host
cont-init: info: /etc/cont-init.d/udev.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun supervisor (no readiness notification)
services-up: info: copying legacy longrun watchdog (no readiness notification)
s6-rc: info: service legacy-services successfully started
[07:35:09] INFO: Starting local supervisor watchdog...
22-03-11 07:35:11 INFO (MainThread) [__main__] Initializing Supervisor setup
22-03-11 07:35:11 INFO (MainThread) [supervisor.docker.network] Can't find Supervisor network, creating a new network
22-03-11 07:35:11 INFO (MainThread) [supervisor.bootstrap] Initializing Supervisor Sentry
22-03-11 07:35:11 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: odroid-n2
22-03-11 07:35:11 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor homeassistant/aarch64-hassio-supervisor with version 2022.05.3
22-03-11 07:35:11 INFO (SyncWorker_0) [supervisor.docker.supervisor] Connecting Supervisor to hassio-network
[... 115 lines cut ...]
22-03-11 00:37:51 WARNING (MainThread) [supervisor.jobs] 'Updater.fetch_data' blocked from execution, no supervisor internet connection
22-03-11 00:37:51 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
22-03-11 00:38:21 WARNING (MainThread) [supervisor.jobs] 'Updater.fetch_data' blocked from execution, no supervisor internet connection
22-03-11 00:38:21 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
22-03-11 00:38:51 WARNING (MainThread) [supervisor.jobs] 'Updater.fetch_data' blocked from execution, no supervisor internet connection
22-03-11 00:38:51 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
22-03-11 00:39:21 WARNING (MainThread) [supervisor.jobs] 'Updater.fetch_data' blocked from execution, no supervisor internet connection
22-03-11 00:39:21 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
22-03-11 00:39:51 WARNING (MainThread) [supervisor.jobs] 'Updater.fetch_data' blocked from execution, no supervisor internet connection
22-03-11 00:39:51 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
22-03-11 00:40:21 WARNING (MainThread) [supervisor.jobs] 'Updater.fetch_data' blocked from execution, no supervisor internet connection
22-03-11 00:40:21 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
22-03-11 00:40:51 WARNING (MainThread) [supervisor.jobs] 'Updater.fetch_data' blocked from execution, no supervisor internet connection
22-03-11 00:40:51 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
22-03-11 00:41:21 WARNING (MainThread) [supervisor.jobs] 'Updater.fetch_data' blocked from execution, no supervisor internet connection
22-03-11 00:41:21 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
22-03-11 00:41:51 WARNING (MainThread) [supervisor.jobs] 'Updater.fetch_data' blocked from execution, no supervisor internet connection
22-03-11 00:41:51 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

agners, 2022-06-25:
I don't think that compiled in is the problem here.

> Note: Once I fixed the time stored in the hardware RTC, this issue no longer occurs after reboot.

Doesn't that prove that prove that the system is able to actually read the RTC?

One thing to note is that the RTC has a bit which essentially says when the system should not trust the RTC value (e.g. because it has never been set, or because the voltage dropped to low and it was not able to reliably run). If that is the case, you'll see an error in the kernel boot logs ("Host logs" in the frontend).

nbetcher, 2022-06-25:
> I don't think that compiled in is the problem here.
>
>
> Doesn't that prove that prove that the system is able to actually read the RTC?
>
> One thing to note is that the RTC has a bit which essentially [...]

Correct, it did say that it couldn't trust the value due to the battery being low; however that's not exactly how the message reads. It says that the voltage is CURRENTLY low and as such cannot be relief upon. That is false, as I measured the voltage coming into the board and it was 3.3+v.

When I said that once I fixed the RTC value it worked, I failed to mention that I had to rmmod the RTC driver, then modprobe it, then use 'hwclock' with -f /dev/rtc1. Ideally this should have been able to be accomplished without having to do that. That does not seem like what a user should be expected to do in order to remediate their RTC battery dying, hence this issue.

closed 2022-09-30 by github-actions[bot]

juagicre, 2022-10-06:
I had issues while "preparing" HA due to certificate not valid because of the system time was not correct.
The problem is still happening with HA 2022.9 and 2022.10. 
I'm not familiar with modules or linux kernel but seems that RTC si somehow useless at all, in my case i'm trying to use a different NTP server, as the default one is banned by my ISP.