home-assistant/operating-system #1981: GitHub not accessible from Generic x86-64

MilesAheadToo, 2022-06-21:
### Describe the issue you are experiencing

When I run `ping github.com` from `Terminal` I get an error message `ping: bad address 'github.com'`. I can ping github.com from several other computers on the network, including Windows, Ubuntu, and RPi OS machines.
I then installed Lubuntu, (Windows) Tiny, and Proxmox on the host separately. In all cases, I can run `ping github.com` successfully. I then installed Generic x86-64 on the Proxmox server and got the same error when running `ping guthub.com`.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

8.2

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Install Lubuntu on the host, and run `ping github.com `successfully.
2. Install (Windows) Tiny on the host, and run `ping github.com `successfully. 
3. Install Proxmox on the host, create a VM following [these instructions](https://community.home-assistant.io/t/installing-home-assistant-os-using-proxmox-7/201835), and run `ping github.com `successfully
4. Install HA in the Proxmox VM using the instructions above, install `Terminal` and `run ping gtihub.com` with the error. Please note that the only add-on installed is `Terminal`.

In all instances, I also pinged other sites such as google.com, microsoft.com, cisco.com successfully. The only errors I got were for github.com.

### Anything in the Supervisor logs that might be useful for us?

```txt
Filtered on GitHub
22-06-21 15:59:36 INFO (MainThread) [supervisor.store.git] Cloning add-on https://github.com/hassio-addons/repository repository
...
```

### Anything in the Host logs that might be useful for us?

```txt
Host log filtered on GitHub
[    2.033012] sof-audio-pci-intel-apl 0000:00:0e.0:        download it from https://github.com/thesofproject/sof-bin/

DNS Full Logs
[INFO] 172.30.32.2:48256 - 41705 "A IN github.com. udp 28 false 512" FORMERR qr,rd 28 0.036633166s
[INFO] 172.30.32.2:55924 - 9369 "A IN github.com. udp 28 false 512" FORMERR qr,rd 28 0.042024412s
...
```

### System Health information

I could not add the system health because of 65536 character size limitation.

MilesAheadToo, 2022-06-21:
System Health

## System Health

version | core-2022.6.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.12
os_name | Linux
os_version | 5.15.45
arch | x86_64
timezone | Europe/Lisbon

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.2
-- | --
update_channel | stable
supervisor_version | supervisor-2022.05.3
agent_version | 1.2.1
docker_version | 20.10.14
disk_total | 116.7 GB
disk_used | 2.0 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.4.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 21 June 2022, 15:02
-- | --
current_recorder_run | 21 June 2022, 16:02
estimated_db_size | 0.19 MiB
database_engine | sqlite
database_version | 3.34.1

</details>

agners, 2022-06-25:
I am assuming you are using the Terminal add-on to run the ping command? DNS resolution inside the add-on is provided by the DNS plug-in. Please check if there is a similar issue in the [plugin-dns](https://github.com/home-assistant/plugin-dns) repository or open one otherwise.

closed 2022-06-25 by github-actions[bot]