home-assistant/operating-system #1982: Enable HDMI-CEC driver in rasberry pi kernel build

s-knibbs, 2022-06-22:
### Describe the issue you are experiencing

I'm currently unable to use the CEC Scanner addon and/or the hdmi_cec integration despite updating to the latest OS version.

Although the kernel has now been updated to 5.15 which includes the HDMI CEC driver, it seems that this is not currently enabled. See https://github.com/raspberrypi/linux/blob/rpi-5.15.y/drivers/gpu/drm/vc4/Kconfig#L26

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

8.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Start the CEC Scanner addon
2. The following error is reported in the container logs:
```
[services.d] starting services
[services.d] done.
[09:47:22] INFO: Starting CEC client scan...
autodetect FAILED
```

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

pvizeli, 2022-06-24:
It is enabled: https://github.com/raspberrypi/linux/blob/rpi-5.15.y/arch/arm64/configs/bcm2711_defconfig but maybe it need this `avoid_warnings=2`

closed 2022-09-30 by github-actions[bot]

akhodyko, 2022-12-11:
I think this is actual issue. CEC Scanner results with same error on RPi 4 64, HA OS 8.5, HA 2022.12.1. How i can get it to work?