home-assistant/operating-system #1994: No HDMI display on Khadas VIM3

cserem, 2022-06-28:
### Describe the issue you are experiencing

When booting HAOS, there is no display on the HDMI port of the Khadas VIM3

### What operating system image do you use?

khadas-vim3 (Khadas VIM3)

### What version of Home Assistant Operating System is installed?

8.2

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

Boot HAOS 8.2 on Khadas VIM3

### Anything in the Supervisor logs that might be useful for us?

```txt
n/a
```

### Anything in the Host logs that might be useful for us?

```txt
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.15.45 (builder@0e93cb2add5f) (aarch64-buildroot-linux-gnu-gcc.br_real (Buildroot -g799554151) 9.4.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP PREEMPT Wed Jun 8 22:49:54 UTC 2022
[    0.000000] Machine model: Khadas VIM3
[    0.000000] efi: UEFI not found.
[    0.000000] OF: fdt: Reserved memory: failed to reserve memory for node 'secmon@5000000': base 0x0000000005000000, size 3 MiB
[    0.000000] Reserved memory: created CMA memory pool at 0x00000000e4c00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000f4e5afff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x00000000052fffff]
[    0.000000]   node   0: [mem 0x0000000005300000-0x00000000072fffff]
[    0.000000]   node   0: [mem 0x0000000007300000-0x00000000f4e5afff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000f4e5afff]
[    0.000000] On node 0, zone DMA: 12709 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.1
[    0.000000] percpu: Embedded 20 pages/cpu s41880 r8192 d31848 u81920
[    0.000000] pcpu-alloc: s41880 r8192 d31848 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 987425
[    0.000000] Kernel command line: zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor systemd.machine_id=6f61b1a92e4f469b8816f131cbe9dfb7 cgroup_enable=memory fsck.repair=yes root=PARTUUID=48617373-06 rootfstype=squashfs ro rootwait rauc.slot=A console=tty0 console=ttyAML0,115200n8
[    0.000000] Unknown kernel command line parameters "cgroup_enable=memory", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 3615276K/4012396K available (12672K kernel code, 1352K rwdata, 3824K rodata, 2240K init, 423K bss, 134976K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=6, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=6.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 100 jiffies.
[... 493 lines cut ...]
[   98.297246] audit: type=1300 audit(1656420404.242:129): arch=c00000b7 syscall=208 success=yes exit=0 a0=4 a1=0 a2=40 a3=aaaaca371770 items=0 ppid=454 pid=3855 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/sbin/xtables-legacy-multi" subj==unconfined key=(null)
[   98.297254] audit: type=1327 audit(1656420404.242:129): proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D740066696C746572002D4100444F434B45520000002D690068617373696F002D6F0068617373696F002D7000756470002D64003137322E33302E33332E30002D2D64706F7274003531383230002D6A00414343455054
[   98.304095] audit: type=1325 audit(1656420404.249:130): table=nat family=2 entries=16 op=xt_replace pid=3857 subj==unconfined comm="iptables"
[   98.688685] eth0: renamed from vethca0ce20
[   98.697082] IPv6: ADDRCONF(NETDEV_CHANGE): vethd39eb33: link becomes ready
[   98.697165] hassio: port 6(vethd39eb33) entered blocking state
[   98.697172] hassio: port 6(vethd39eb33) entered forwarding state
[   99.930901] hassio: port 7(vethfde53f2) entered blocking state
[   99.930916] hassio: port 7(vethfde53f2) entered disabled state
[   99.931048] device vethfde53f2 entered promiscuous mode
[  100.500104] eth0: renamed from veth45f25e6
[  100.509368] IPv6: ADDRCONF(NETDEV_CHANGE): vethfde53f2: link becomes ready
[  100.509629] hassio: port 7(vethfde53f2) entered blocking state
[  100.509652] hassio: port 7(vethfde53f2) entered forwarding state
[  100.928832] hassio: port 8(vethe44667d) entered blocking state
[  100.928847] hassio: port 8(vethe44667d) entered disabled state
[  100.929069] device vethe44667d entered promiscuous mode
[  100.930075] hassio: port 8(vethe44667d) entered blocking state
[  100.930090] hassio: port 8(vethe44667d) entered forwarding state
[  100.933618] hassio: port 8(vethe44667d) entered disabled state
```

### Additional information

Since the device is new, I don't know if it ever worked with HAOS, but out of curiosity I have flashed an older 7.6? version of HAOS. With it there is display for 1-2 seconds, I can see the usual kernel initialization / dmesg lines, but after that the screen goes blank, and the monitor says there is no signal.

agners, 2022-08-16:
It sounds like the output gets lost when the kernel loads the actual DRM driver for this IP. I don't remember if I tested it with previous OS versions :cold_sweat: I'll have to check this on my end.

spikerguy, 2022-12-18:
Hi @agners 
Just read this. 

I am just a tag away to test hassos on arm devices which I have. 

I will test it on vim3 and share the feedback in coming days.

closed 2023-03-25 by github-actions[bot]