home-assistant/operating-system #1996: Unable to take application consistent production checkpoint in HyperV - HAOS 8.2

Locknes, 2022-06-28:
### Describe the issue you are experiencing

I am not able to take production checkpoints of my HAOS VM, with the operation failing with the error in the image below.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 8.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Enable Production Checkpoint in hyperV VM configuration
2. Disable "Create Standard checkpoint if the guest does not support creation of production checkpoints" box
3. Try to take checkpoint
...
[image]
[image]

### Anything in the Supervisor logs that might be useful for us?

```txt
22-06-28 14:03:59 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
22-06-28 14:03:59 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-06-28 14:03:59 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
22-06-28 14:03:59 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
22-06-28 14:05:12 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
22-06-28 14:05:40 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/blakeblackshear/frigate-hass-addons repository
22-06-28 14:05:40 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository
22-06-28 14:05:40 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
22-06-28 14:05:40 WARNING (SyncWorker_0) [supervisor.addons.validate] Add-on have full device access, and selective device access in the configuration. Please report this to the maintainer of Frigate NVR (Full Access)
22-06-28 14:05:40 WARNING (SyncWorker_5) [supervisor.addons.validate] Add-on have full device access, and selective device access in the configuration. Please report this to the maintainer of Frigate NVR (Full Access)
22-06-28 14:05:40 INFO (MainThread) [supervisor.store] Loading add-ons from store: 67 all - 0 new - 0 remove
22-06-28 14:05:40 INFO (MainThread) [supervisor.store] Loading add-ons from store: 67 all - 0 new - 0 remove
22-06-28 14:16:24 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
22-06-28 14:19:00 INFO (MainThread) [supervisor.host.info] Updating local host information
22-06-28 14:19:01 INFO (MainThread) [supervisor.host.services] Updating service information
22-06-28 14:19:01 INFO (MainThread) [supervisor.host.network] Updating local network information
22-06-28 14:19:01 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
22-06-28 14:19:01 INFO (MainThread) [supervisor.host.manager] Host information reload completed
22-06-28 14:46:25 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
22-06-28 15:03:59 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
22-06-28 15:03:59 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
22-06-28 15:03:59 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
22-06-28 15:03:59 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
22-06-28 15:03:59 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_IPV6_ERROR/ContextType.DNS_SERVER
22-06-28 15:03:59 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_FAILED/ContextType.DNS_SERVER
22-06-28 15:03:59 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR
22-06-28 15:03:59 INFO (MainThread) [supervisor.resolution.check] System checks complete
22-06-28 15:03:59 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
22-06-28 15:03:59 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-06-28 15:03:59 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
22-06-28 15:03:59 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
22-06-28 15:16:25 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
22-06-28 15:46:25 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
22-06-28 15:50:55 INFO (MainThread) [supervisor.backups.manager] Creating new partial backup with slug f23abb13
22-06-28 15:50:55 INFO (MainThread) [supervisor.homeassistant.module] Backing up Home Assistant Core config folder
22-06-28 15:54:48 INFO (MainThread) [supervisor.homeassistant.module] Backup Home Assistant Core config folder done
22-06-28 15:55:18 INFO (MainThread) [supervisor.backups.manager] Creating partial backup with slug f23abb13 completed
22-06-28 15:55:18 INFO (MainThread) [supervisor.homeassistant.core] Updating Home Assistant to version 2022.6.7
22-06-28 15:55:18 INFO (SyncWorker_0) [supervisor.docker.interface] Updating image ghcr.io/home-assistant/qemux86-64-homeassistant:2022.6.6 to ghcr.io/home-assistant/qemux86-64-homeassistant:2022.6.7
22-06-28 15:55:18 INFO (SyncWorker_0) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/qemux86-64-homeassistant with tag 2022.6.7.
22-06-28 15:56:05 INFO (SyncWorker_0) [supervisor.docker.interface] Stopping homeassistant application
22-06-28 15:56:16 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning homeassistant application
[... 112 lines cut ...]

oldest_recorder_run | June 20, 2022, 4:31 PM
-- | --
current_recorder_run | June 28, 2022, 3:56 PM
estimated_db_size | 1749.11 MiB
database_engine | sqlite
database_version | 3.34.1

</details>

<details><summary>Sonoff</summary>

version | 3.0.5 (200f243)
-- | --
cloud_online | 2 / 2
local_online | 2 / 2

</details>

### Additional information

Please let me know if this would be better raised in the discord.
I was going to post there, but I didn't see a channel that looked correct.

Thank you.

agners, 2022-07-13:
I think this is better reported to the Hyper-V team. The error message has absolutely no hint what could be wrong in our image. Without additional information of *why* checkpoint failed, there is no way I can debug this.

scyto, 2023-03-01:
I have hit this .I am prepared to create a new issue but wanted to check first what version of the hyper-v integration services are installed / is there anyway for me to update them in the production VM?

I believe the following services are needed inside the VM, i can find no evidence they are running (but i may just be stupid)

The services required for checkpoint creation:

- hv_vss_daemon
- hypervvssd

Generally on generic linux they are installed as follows:
- Ubuntu: apt-get install "linux-cloud-tools-$(uname -r)"
- CentOS: yum install hyperv-daemons
- Debian: apt-get install hyperv-daemons