home-assistant/operating-system #1997: No haos notification update. Stuck on 7.6

rocheuf, 2022-06-30:
### Describe the issue you are experiencing

No haos update available.
Update selection remains empty after manual refresh in the update section of the settings.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

7.6

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Update request 

### Anything in the Supervisor logs that might be useful for us?

```txt
Supervisor log after update request :

22-06-30 20:22:28 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
22-06-30 20:22:28 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/sabeechen/hassio-google-drive-backup repository
22-06-30 20:22:28 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
22-06-30 20:22:28 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/zigbee2mqtt/hassio-zigbee2mqtt repository
22-06-30 20:22:28 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository
22-06-30 20:22:28 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/home-assistant-addon repository
22-06-30 20:22:31 INFO (MainThread) [supervisor.store] Loading add-ons from store: 68 all - 0 new - 0 remove
22-06-30 20:22:31 INFO (MainThread) [supervisor.store] Loading add-ons from store: 68 all - 0 new - 0 remove
```

### Anything in the Host logs that might be useful for us?

```txt
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.10.103-v8 (builder@f0f12117096a) (aarch64-buildroot-linux-gnu-gcc.br_real (Buildroot -g8d88bd9a3) 9.4.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP PREEMPT Mon Mar 28 11:44:31 UTC 2022
[    0.000000] random: fast init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b2fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] On node 0 totalpages: 1012480
[    0.000000]   DMA zone: 3788 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242432 pages, LIFO batch:63
[    0.000000]   DMA32 zone: 12032 pages used for memmap
[    0.000000]   DMA32 zone: 770048 pages, LIFO batch:63
[    0.000000] On node 0, zone DMA32: 256 pages in unavailable ranges
[    0.000000] percpu: Embedded 32 pages/cpu s91544 r8192 d31336 u131072
[    0.000000] pcpu-alloc: s91544 r8192 d31336 u131072 alloc=32*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 996660
[... 650 lines cut ...]
version_api | ok
installed_addons | Samba share (10.0.0), Duck DNS (1.15.0), File editor (5.3.3), Home Assistant Google Drive Backup (0.108.2), Z-Wave JS to MQTT (0.44.0), Zigbee2mqtt (1.25.2-1), Mosquitto broker (6.1.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 12
views | 10
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 27 juin 2022 à 08:56
-- | --
current_recorder_run | 29 juin 2022 à 17:50
estimated_db_size | 23.48 MiB
database_engine | sqlite
database_version | 3.34.1

</details>

agners, 2022-07-08:
Can you open the Terminal and type:

```
ha os info
```

Maybe a browser caching issue? What browser/versions are you using?

paavo, 2022-07-08:
I just saw a new core update is available `2022.7.1`
maybe this works 🤔

rocheuf, 2022-07-08:
[image]

Latest iOS app and latest chrome OS X

Still the same issue after updating to core 2022.7.1

paavo, 2022-07-09:
> I just saw a new core update is available `2022.7.1` maybe this works 🤔

nope 😆

agners, 2022-07-13:
You can update the OS using `ha os update`. However, it still begs the question why the update notification did not make it through to the frontend.

I move this to the frontend as this is not directly related to the OS and the Supervisor seems to correctly detect the new version.