home-assistant/operating-system #1998: HAOS server has no IP address when DHCP server not available on boot

gwww, 2022-07-01:
### Describe the issue you are experiencing

When booting HAOS and a DHCP (v4 in my case) server is not available then the Ethernet interface never gets an IP address without manual intervention (see steps below).

The scenario above happened during a whole house power failure. What happens is that on power restore the HAOS box comes up before the DHCP server (on an OPNSense router) is available. Hence the HAOS box does not get an IP address.

This happened to me about 3 weeks ago and I wrote it off as a glitch. Then on Discord someone else started discussing a similar problem ([here](https://discord.com/channels/330944238910963714/991688051074015332)). That caused me to try and reproduce this and hence this problem report.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

8.2

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Make sure that a DHCP server is not running on the network. In my case I use OPNSense router and turn DHCPv4 off.
2. Reboot HAOS.
3. System boots and Ethernet interface does not have an address (as expected).
4. Turn on DHCP server.
5. HAOS does not obtain an address.

Since I am in "lab mode" with this new x86 box, it is connected to a keyboard and monitor. Issuing the command `net update enp1s0 --ipv4-mode auto` causes an address to be assigned to the interface.

When in "production mode" access to the box may not be possible (e.g.: I'm not at home).

### Anything in the Supervisor logs that might be useful for us?

```txt
As expected, there are multiple flavours of this:

22-06-30 21:46:40 WARNING (MainThread) [supervisor.jobs] 'Updater.fetch_data' blocked from execution, no supervisor internet connection
```
```

### Anything in the Host logs that might be useful for us?

```txt
I can't get to the host log when no DHCP server is available since I cannot access the box. I don't expect anything in these logs as reproducing this problem is very reproducible.
```

carsten-h, 2022-07-31:
You can set a static IP inside the HA OS itself! That is what I am doing for all devices which should get a static IP.
I don‘t know the english names of the whole menu, it must be Settings/System/Network/IPv4 and there choose static.

gwww, 2022-07-31:
Thx @carsten-h. That would stop the problem from happening. The bug still remains that a DHCP served HAOS system does not get a DHCP address on certain conditions. 

One addition to my bug report is a possible solution. When the system does not get a DHCP address, retrying every minute (or so) to get one would be an acceptable solution to this problem. Or slightly more complex is retrying with an exponential backup with a max retry time.

closed 2022-11-05 by github-actions[bot]