home-assistant/operating-system #1999: Is Raspberry Pi 3B + still viable? I'm crashing almost every day.

magic7s, 2022-07-04:
### Describe the issue you are experiencing

Memory is bouncing off 100%, normally over 90% and hits 95%+ a lot (I setup an alert). Power cycle is required. I can ping the host but SSH does not connect nor does UI. 

How can I troubleshoot better?

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

8.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Launching add-ons like VS Code or ESP Home crash the OS. 
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Just the current log (after power cycle)
```

### Anything in the Host logs that might be useful for us?

```txt
Just the current boot log (after power cycle)
```

### System Health information

## System Health

version | core-2022.6.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.12
os_name | Linux
os_version | 5.15.32-v8
arch | aarch64
timezone | America/Los_Angeles

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4896
Installed Version | 1.25.5
Stage | running
Available Repositories | 1071
Downloaded Repositories | 7

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | July 23, 2022 at 5:00 PM
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | false
remote_server | us-west-2-0.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.2
-- | --
update_channel | stable
supervisor_version | supervisor-2022.05.3
agent_version | 1.2.1
docker_version | 20.10.14
disk_total | 58.7 GB
disk_used | 7.0 GB
healthy | true
supported | true
board | rpi3-64
supervisor_api | ok
version_api | ok
installed_addons | Mosquitto broker (6.1.2), Zigbee2MQTT (1.25.2-1), Studio Code Server (5.1.1), Eufy Security Add-on (0.9.2), Terminal & SSH (9.4.0), Home Assistant Google Drive Backup (0.108.2), File editor (5.3.3), ESPHome (2022.6.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 1
views | 4
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | June 24, 2022 at 11:15 AM
-- | --
current_recorder_run | July 3, 2022 at 5:53 PM
estimated_db_size | 325.90 MiB
database_engine | sqlite
database_version | 3.34.1

</details>

magic7s, 2022-07-04:
I was using the 64 bit HAOS and switched to 32 bit and it is much snappier and lower memory utilization. 

[image]

groenmarsmannetje, 2022-08-08:
I also have the feeling that memory consumption of Home Assistant has become too high to properly run on a Raspberry PI 3+. It is really pushing between 80%-90% now. And had to stop almost all addons, besides the ones that I really depend on (MQTT, Zigbee2Mqtt, SSH and SAMBA). But as soon as the systeem needs to do some "extra work" it will become unresponsive until it will restart because Watchdog interfenes with a kill process.

HarvsG, 2022-09-28:
I am using the 3B+ 64-bit with:
Sensors: 122
Lights: 26 - mostly zigbee
Switches: 18
Binary sensors: 33
Automations: 51
and a few add-ons 
I have quite high memory use: 
[image]

groenmarsmannetje, 2022-09-28:
After reading about 32-bit vs. 64-bit on Raspberry Pi 3+. I moved back to 32-bit. That uses approx. 20% less memory and runs smooth.

kongo09, 2022-12-27:
I have the same challenge in a rather tiny setup. Glad that I found this issue so I know that it's not just me, but there seems to be a general problem. Unfortunately, the 32bit version of HAOS doesn't support some important add-ons like the unifi console or the visual studio code server.

It would be great to read some advice on usage of 32 vs. 64 bit in the official installation documentation.

closed 2023-04-03 by github-actions[bot]