home-assistant/operating-system #2000: USB stops working after some time on XU4

RikVerstijnen, 2022-07-04:
### Describe the issue you are experiencing

After a reboot of the host, all USB ports run fine. But after a few hours, one of the USB ports stops responding and this appears in the host log:

`[302902.461914] xhci-hcd xhci-hcd.8.auto: Ring expansion failed
[302902.466501] ch341-uart ttyUSB1: usb_serial_generic_submit_read_urb - usb_submit_urb failed: -12`

Because of that, the add-on using the USB device throws the error:
`ERROR    mixins       run_once                        103 | dsmr_datalogger.management.commands.dsmr_datalogger: [!] Exception raised. Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/serial/serialposix.py", line 322, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
OSError: [Errno 12] Out of memory: '/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A13LNT9A-if00-port0'
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/app/dsmr_datalogger/scripts/dsmr_datalogger_api_client.py", line 37, in read_telegram
    serial_handle = serial.serial_for_url(url=url_or_port, timeout=MAX_READ_TIMEOUT, **serial_kwargs)
  File "/usr/local/lib/python3.10/site-packages/serial/__init__.py", line 90, in serial_for_url
    instance.open()
  File "/usr/local/lib/python3.10/site-packages/serial/serialposix.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 12] could not open port /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A13LNT9A-if00-port0: [Errno 12] Out of memory: '/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A13LNT9A-if00-port0'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/app/dsmr_backend/mixins.py", line 92, in run_once
    self.run(data=self.data, **options)
  File "/app/dsmr_datalogger/management/commands/dsmr_datalogger.py", line 29, in run
    telegram = next(self.telegram_generator)
  File "/app/dsmr_datalogger/scripts/dsmr_datalogger_api_client.py", line 39, in read_telegram
    raise RuntimeError('Failed to connect: {}', error) from error
RuntimeError: ('Failed to connect: {}', SerialException(12, "could not open port /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A13LNT9A-if00-port0: [Errno 12] Out of memory: '/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A13LNT9A-if00-port0'"))
127.0.0.1 - - [04/Jul/2022:10:20:48 +0200] "GET /about HTTP/1.1" 200 19326 "-" "curl/7.83.1" "-"
Starting DSMR Reader - datalogger...
127.0.0.1 - - [04/Jul/2022:10:20:54 +0200] "GET /about HTTP/1.1" 200 19326 "-" "curl/7.83.1" "-"
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html`

### What operating system image do you use?

odroid-xu4 (Hardkernel ODROID-XU4)

### What version of Home Assistant Operating System is installed?

8.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Reboot the host with USB devices plugged in
2. Wait for a few hours
3. USB stops working and errors appear

### Anything in the Supervisor logs that might be useful for us?

```txt
Nope
```

### Anything in the Host logs that might be useful for us?

```txt
[302902.461914] xhci-hcd xhci-hcd.8.auto: Ring expansion failed
[302902.466501] ch341-uart ttyUSB1: usb_serial_generic_submit_read_urb - usb_submit_urb failed: -12
```

### System Health information

## System Health

version | core-2022.6.7
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.12
os_name | Linux
os_version | 5.15.45
arch | armv7l
timezone | UTC

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4908
Installed Version | 1.25.5
Stage | running
Available Repositories | 1076
Downloaded Repositories | 14

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.2
-- | --
update_channel | stable
supervisor_version | supervisor-2022.05.3
agent_version | 1.2.1
docker_version | 20.10.14
disk_total | 28.0 GB
disk_used | 15.1 GB
healthy | true
supported | true
board | odroid-xu4
supervisor_api | ok
version_api | ok
installed_addons | SSH & Web Terminal (11.0.1), Terminal & SSH (9.4.0), Samba share (10.0.0), Duck DNS (1.15.0), Grafana (7.6.0), InfluxDB (4.5.0), Home Assistant Google Drive Backup (0.108.2), File editor (5.3.3), Node-RED (12.0.2), TimescaleDB (2.0.1), Mosquitto broker (6.1.2), DSMR Reader (1.4.0), NGINX Home Assistant SSL proxy (3.1.5)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 11
views | 6
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | June 27, 2022 at 21:46
-- | --
current_recorder_run | June 30, 2022 at 21:30
estimated_db_size | 546.95 MiB
database_engine | sqlite
database_version | 3.34.1

</details>

RikVerstijnen, 2022-07-11:
> Hm, -12 means Out of memory, at the error just before that reports that expansion failed.
> 
> How is the memory usage of your system?

Pretty OK I guess:
[image]

I am also not experiencing any issues besides the USB problem.

agners, 2022-07-13:
Can you setup SSH access to the underlying OS (see [SSH access to the host](https://developers.home-assistant.io/docs/operating-system/debugging#ssh-access-to-the-host)) and execute this command in the shell?

```
sysctl -w vm.min_free_kbytes=16384
```

RikVerstijnen, 2022-08-01:
> Can you setup SSH access to the underlying OS (see [SSH access to the host](https://developers.home-assistant.io/docs/operating-system/debugging#ssh-access-to-the-host)) and execute this command in the shell?
> 
> ```
> sysctl -w vm.min_free_kbytes=16384
> ```

I managed to make the change, but the error appeared again some time after an OS reboot. Should the setting "survive" an OS reboot? In that case, the solution unfortunately does not work. 

I use the USB ports mainly for P1/modbus polling. I will increase the polling intervals, maybe that will help.

closed 2022-11-06 by github-actions[bot]