home-assistant/operating-system #2003: HA OS boot issue on NUC model NUC7CJYH

gabriel600r, 2022-07-05:
### Describe the issue you are experiencing

Install Homeassistant Generic x86-64 on SSD with Balena, at first boot I have no problem, but then I simulate a power outage and it gets stuck at boot (attached image).

The NUC has the updated BIOS, I tried removing the USBs, formatting again and the same thing continues.

[image]
[image]

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

8.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. cut power to the NUC
2. turn on NUC
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Boot problem
```

### Anything in the Host logs that might be useful for us?

```txt
Boot Problem
```

### System Health information

hardware data
Celero J4005
8GB RAM
120GB SSD

agners, 2022-07-05:
It seems like the kernel triggers a page fault on this machine every time. It feels like a kernel bug. Can you try this latest development build?

https://os-builds.home-assistant.io/9.0.dev20220704/haos_generic-x86-64-9.0.dev20220704.img.xz

gabriel600r, 2022-07-05:
> It seems like the kernel triggers a page fault on this machine every time. It feels like a kernel bug. Can you try this latest development build?
> 
> https://os-builds.home-assistant.io/9.0.dev20220704/haos_generic-x86-64-9.0.dev20220704.img.xz

Thank you very much for your answer and your time. Now I try it and I'll pass you the news!,

Thank you!

gabriel600r, 2022-07-05:
@agners I did the tests with the image you gave me and it doesn't fail anymore at boot. Now I ask you a question, can I use this version and when the final release comes out, update from hassio? Or do I have to wait for the final release and only then do the installation? I hope you understand my query,

Thanks for your help!

agners, 2022-07-06:
You can safely use that version, and downgrade to the 8.3 release once it is out (using `ha os update --version 8.3`).

PremyslLeszko, 2022-09-11:
Hi I am having similar issue. I am running 8.5 version alredy... 
When I tur off HA or the power is cut to NUC and restarted again the HA OS desnt boot. I have to cut power (reboot NUC) several times until randomly HA OS boot properly. Please any suugestion or help? 
Thank you

NUC7CJYSN 4GB/64GB Internal disk. HA v8.5

agners, 2022-09-12:
@PremyslLeszko  use the latest 9.0 release candidate, it seems to work for the OP: https://github.com/home-assistant/operating-system/releases/tag/9.0.rc2

agners, 2022-09-12:
I am closing the bug as this will be fixed in 9.0.

closed 2022-09-12 by agners

PremyslLeszko, 2022-12-10:
Yes, I agree the issue persisit. I have just experienced this a week ago with current configuration: NUC7CJYSN 4GB/64GB Internal disk. HA v9.3

andruance, 2023-10-19:
For those coming back this thread, I just flashed 9.0 and can confirm its working. This was after trying several of the more recent developer versions, including  11.1

agners, 2023-10-19:
@andruance this shouuld not be a problem in newer versions. If you experience this in the most recent release (e.g. 11.0), please open a new issue along with the details (what exactly happens).