home-assistant/operating-system #2006: Update to 8.3: 'OSManager.update' blocked from execution, system is not running

RubenKelevra, 2022-07-08:
### Describe the issue you are experiencing

I saw an update for Home Assistant OS 8.3 but could not initiate the update.

There was a toast-popup:

[image]

Updating the MariaDB server on the other hand wasn't an issue. 

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

8.2

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Go to settings
2. Click on the Home Assistant OS update
3. Click on install

### Anything in the Supervisor logs that might be useful for us?

```txt
22-07-08 05:25:37 INFO (SyncWorker_5) [supervisor.docker.interface] Cleaning addon_db21ed7f_wger application
22-07-08 05:25:50 INFO (SyncWorker_5) [supervisor.docker.addon] Starting Docker add-on db21ed7f/aarch64-addon-wger with version 1.9-2
22-07-08 05:26:21 WARNING (MainThread) [supervisor.misc.tasks] Watchdog found a problem with db21ed7f_wger!
22-07-08 05:26:21 INFO (SyncWorker_4) [supervisor.docker.interface] Cleaning addon_db21ed7f_wger application
22-07-08 05:26:22 INFO (SyncWorker_4) [supervisor.docker.addon] Starting Docker add-on db21ed7f/aarch64-addon-wger with version 1.9-2
22-07-08 05:26:52 WARNING (MainThread) [supervisor.misc.tasks] Watchdog found a problem with db21ed7f_wger!
22-07-08 05:26:52 INFO (SyncWorker_4) [supervisor.docker.interface] Cleaning addon_db21ed7f_wger application
22-07-08 05:26:54 INFO (SyncWorker_4) [supervisor.docker.addon] Starting Docker add-on db21ed7f/aarch64-addon-wger with version 1.9-2
22-07-08 05:27:24 WARNING (MainThread) [supervisor.misc.tasks] Watchdog found a problem with db21ed7f_wger!
22-07-08 05:27:24 INFO (SyncWorker_5) [supervisor.docker.interface] Cleaning addon_db21ed7f_wger application
22-07-08 05:27:26 INFO (SyncWorker_5) [supervisor.docker.addon] Starting Docker add-on db21ed7f/aarch64-addon-wger with version 1.9-2
22-07-08 05:27:56 WARNING (MainThread) [supervisor.misc.tasks] Watchdog found a problem with db21ed7f_wger!
22-07-08 05:27:56 INFO (SyncWorker_7) [supervisor.docker.interface] Cleaning addon_db21ed7f_wger application
22-07-08 05:27:57 INFO (SyncWorker_7) [supervisor.docker.addon] Starting Docker add-on db21ed7f/aarch64-addon-wger with version 1.9-2
22-07-08 05:28:29 ERROR (SyncWorker_7) [supervisor.docker.interface] Container addon_db21ed7f_wger is not running
22-07-08 05:28:29 WARNING (MainThread) [supervisor.misc.tasks] Watchdog found a problem with db21ed7f_wger!
22-07-08 05:28:29 INFO (SyncWorker_7) [supervisor.docker.interface] Cleaning addon_db21ed7f_wger application
22-07-08 05:28:31 INFO (SyncWorker_7) [supervisor.docker.addon] Starting Docker add-on db21ed7f/aarch64-addon-wger with version 1.9-2
22-07-08 05:29:01 WARNING (MainThread) [supervisor.misc.tasks] Watchdog found a problem with db21ed7f_wger!
22-07-08 05:29:01 INFO (SyncWorker_1) [supervisor.docker.interface] Cleaning addon_db21ed7f_wger application
22-07-08 05:29:02 INFO (SyncWorker_1) [supervisor.docker.addon] Starting Docker add-on db21ed7f/aarch64-addon-wger with version 1.9-2
22-07-08 05:29:32 WARNING (MainThread) [supervisor.misc.tasks] Watchdog found a problem with db21ed7f_wger!
22-07-08 05:29:33 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning addon_db21ed7f_wger application
22-07-08 05:29:34 INFO (SyncWorker_0) [supervisor.docker.addon] Starting Docker add-on db21ed7f/aarch64-addon-wger with version 1.9-2
22-07-08 05:30:00 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup
22-07-08 05:30:00 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup
22-07-08 05:30:01 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from cebe7a76_hassio_google_drive_backup
22-07-08 05:30:01 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup
22-07-08 05:30:01 INFO (MainThread) [supervisor.api.middleware.security] /addons/a0d7b954_grocy/info access from cebe7a76_hassio_google_drive_backup
22-07-08 05:30:01 INFO (MainThread) [supervisor.api.middleware.security] /addons/a0d7b954_grocy/options access from cebe7a76_hassio_google_drive_backup
22-07-08 05:30:01 INFO (MainThread) [supervisor.api.middleware.security] /addons/a0d7b954_grocy/stop access from cebe7a76_hassio_google_drive_backup
22-07-08 05:30:01 INFO (SyncWorker_5) [supervisor.docker.interface] Stopping addon_a0d7b954_grocy application
22-07-08 05:30:04 WARNING (MainThread) [supervisor.misc.tasks] Watchdog found a problem with db21ed7f_wger!
22-07-08 05:30:04 INFO (SyncWorker_3) [supervisor.docker.interface] Cleaning addon_db21ed7f_wger application
22-07-08 05:30:06 INFO (SyncWorker_3) [supervisor.docker.addon] Starting Docker add-on db21ed7f/aarch64-addon-wger with version 1.9-2
22-07-08 05:30:06 INFO (SyncWorker_5) [supervisor.docker.interface] Cleaning addon_a0d7b954_grocy application
22-07-08 05:30:06 INFO (MainThread) [supervisor.api.middleware.security] /addons/core_mariadb/info access from cebe7a76_hassio_google_drive_backup
22-07-08 05:30:06 INFO (MainThread) [supervisor.api.middleware.security] /addons/core_mariadb/options access from cebe7a76_hassio_google_drive_backup
22-07-08 05:30:07 INFO (MainThread) [supervisor.api.middleware.security] /addons/core_mariadb/stop access from cebe7a76_hassio_google_drive_backup
[... 174 lines cut ...]
The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 193, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1713, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1750, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 680, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 930, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 717, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 152, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 411, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 214, in async_install
    raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Error updating Home Assistant Operating System: 'OSManager.update' blocked from execution, system is not running - CoreState.FREEZE
```

RubenKelevra, 2022-07-08:
Here's the host log (pasting everything into the ticket lead to a "too long" error in github).

<details>

```
[486147.655721] audit: type=1325 audit(1657250253.785:30585): table=filter family=2 entries=44 op=xt_replace pid=423355 subj==unconfined comm="iptables"
[486147.655754] audit: type=1300 audit(1657250253.785:30585): arch=c00000b7 syscall=208 success=yes exit=0 a0=4 a1=0 a2=40 a3=5596e7c9c0 items=0 ppid=496 pid=423355 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/sbin/xtables-legacy-multi" subj==unconfined key=(null)
[486147.655774] audit: type=1327 audit(1657250253.785:30585): proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D740066696C746572002D4100444F434B45520000002D690068617373696F002D6F0068617373696F002D7000746370002D64003137322E33302E33332E3130002D2D64706F72740038303030002D6A00414343455054
[486147.666049] audit: type=1325 audit(1657250253.797:30586): table=nat family=2 entries=44 op=xt_replace pid=423357 subj==unconfined comm="iptables"
[486148.344773] eth0: renamed from veth351aaf2
[486148.367289] IPv6: ADDRCONF(NETDEV_CHANGE): veth8ea76ea: link becomes ready
[486148.367522] hassio: port 16(veth8ea76ea) entered blocking state
[486148.367537] hassio: port 16(veth8ea76ea) entered forwarding state
[486152.551062] hassio: port 16(veth8ea76ea) entered disabled state
[486152.551444] veth351aaf2: renamed from eth0
[486152.629396] hassio: port 16(veth8ea76ea) entered disabled state
[486152.631903] device veth8ea76ea left promiscuous mode
[486152.631945] hassio: port 16(veth8ea76ea) entered disabled state
[486152.632000] kauditd_printk_skb: 41 callbacks suppressed
[486152.632007] audit: type=1700 audit(1657250258.753:30600): dev=veth8ea76ea prom=0 old_prom=256 auid=4294967295 uid=0 gid=0 ses=4294967295
[486152.694136] audit: type=1300 audit(1657250258.753:30600): arch=c00000b7 syscall=206 success=yes exit=32 a0=f a1=40009dc7e0 a2=20 a3=0 items=0 ppid=1 pid=496 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="dockerd" exe="/usr/bin/dockerd" subj==unconfined key=(null)
[486152.694165] audit: type=1327 audit(1657250258.753:30600): proctitle=2F7573722F62696E2F646F636B657264002D480066643A2F2F002D2D6370752D72742D72756E74696D653D393530303030002D2D73746F726167652D6472697665723D6F7665726C617932002D2D6C6F672D6472697665723D6A6F75726E616C64002D2D6C6F672D6F7074007461673D7B7B2E496D6167654E616D657D7D2F7B
[486152.787426] audit: type=1325 audit(1657250258.917:30601): table=nat family=2 entries=15 op=xt_unregister pid=402974 subj==unconfined comm="kworker/u8:4"
[486152.787451] audit: type=1325 audit(1657250258.917:30602): table=filter family=2 entries=4 op=xt_unregister pid=402974 subj==unconfined comm="kworker/u8:4"
[486179.382610] hassio: port 16(veth4c99bc6) entered blocking state
[486179.382631] hassio: port 16(veth4c99bc6) entered disabled state
[486179.387126] device veth4c99bc6 entered promiscuous mode
[486179.387196] audit: type=1700 audit(1657250285.512:30603): dev=veth4c99bc6 prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
[486179.387560] audit: type=1300 audit(1657250285.512:30603): arch=c00000b7 syscall=206 success=yes exit=40 a0=f a1=4003c49710 a2=28 a3=0 items=0 ppid=1 pid=496 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="dockerd" exe="/usr/bin/dockerd" subj==unconfined key=(null)
[486179.387606] audit: type=1327 audit(1657250285.512:30603): proctitle=2F7573722F62696E2F646F636B657264002D480066643A2F2F002D2D6370752D72742D72756E74696D653D393530303030002D2D73746F726167652D6472697665723D6F7665726C617932002D2D6C6F672D6472697665723D6A6F75726E616C64002D2D6C6F672D6F7074007461673D7B7B2E496D6167654E616D657D7D2F7B
[486179.447691] audit: type=1325 audit(1657250285.576:30604): table=nat family=2 entries=43 op=xt_replace pid=423529 subj==unconfined comm="iptables"
[486179.447724] audit: type=1300 audit(1657250285.576:30604): arch=c00000b7 syscall=208 success=yes exit=0 a0=4 a1=0 a2=40 a3=55828ee6a0 items=0 ppid=496 pid=423529 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/sbin/xtables-legacy-multi" subj==unconfined key=(null)
[486179.447743] audit: type=1327 audit(1657250285.576:30604): proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D74006E6174002D4100444F434B4552002D7000746370002D6400302F30002D2D64706F72740039393237002D6A00444E4154002D2D746F2D64657374696E6174696F6E003137322E33302E33332E31303A383030300000002D690068617373696F
[486179.456948] audit: type=1325 audit(1657250285.584:30605): table=filter family=2 entries=44 op=xt_replace pid=423531 subj==unconfined comm="iptables"
[486179.457548] audit: type=1300 audit(1657250285.584:30605): arch=c00000b7 syscall=208 success=yes exit=0 a0=4 a1=0 a2=40 a3=55a82569c0 items=0 ppid=496 pid=423531 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/sbin/xtables-legacy-multi" subj==unconfined key=(null)
[486179.457948] audit: type=1327 audit(1657250285.584:30605): proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D740066696C746572002D4100444F434B45520000002D690068617373696F002D6F0068617373696F002D7000746370002D64003137322E33302E33332E3130002D2D64706F72740038303030002D6A00414343455054
[... cut ...]
[486905.088963] hassio: port 16(veth13eb23c) entered disabled state
[486905.089004] audit: type=1700 audit(1657251011.200:31066): dev=veth13eb23c prom=0 old_prom=256 auid=4294967295 uid=0 gid=0 ses=4294967295
[486908.174979] vethfd84117: renamed from eth0
[486908.195215] hassio: port 11(vethf828650) entered disabled state
[486908.237052] hassio: port 11(vethf828650) entered disabled state
[486908.238962] device vethf828650 left promiscuous mode
[486908.239000] hassio: port 11(vethf828650) entered disabled state
```

</details>

RubenKelevra, 2022-07-08:
After the MariaDB update was completed, the system health:

## System Health

version | core-2022.7.0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.32-v8
arch | aarch64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | failed to load: timeout
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.25.5
Stage | running
Available Repositories | 1073
Downloaded Repositories | 13

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | July 28, 2022 at 02:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
remote_server | eu-west-2-2.ui.nabu.casa
can_reach_cert_server | failed to load: timeout
can_reach_cloud_auth | failed to load: timeout
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.2
-- | --
update_channel | beta
supervisor_version | supervisor-2022.07.0
agent_version | 1.2.1
docker_version | 20.10.14
disk_total | 58.7 GB
disk_used | 37.9 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | failed to load: timeout
installed_addons | Mosquitto broker (6.1.2), Grocy (0.18.2), Portainer (2.0.0), MariaDB (2.5.0), Home Assistant Google Drive Backup (0.108.2), Zigbee2MQTT (1.26.0-1), Tailscale (0.7.0), SSH & Web Terminal (11.0.1), Log Viewer (0.14.0), Dnsmasq (1.4.4), Tor (4.0.0), phpMyAdmin (0.8.0), Home Panel (2.5.0), Studio Code Server (5.1.2), WireGuard (0.7.0), ESPHome (2022.6.2), Git pull (7.13.1), Photoprism (220629-jammy), Seafile (v9.0.6-pro), Papermerge (v2.0.1-ls52), Piwigo (12.3.0), Nextcloud (24.0.2), Wger (1.9-2), Webtop Alpine KDE (4.16-r0-ls62), Zoneminder (1.36.19)

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 1
views | 18
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | June 14, 2022 at 10:48
-- | --
current_recorder_run | July 7, 2022 at 16:12
estimated_db_size | 5087.95 MiB
database_engine | mysql
database_version | 10.4.19

</details>

agners, 2022-07-08:
Something put the Supervisor in `CoreState.FREEZE`, which prevents any execution. From the snapshot its not clear what that exactly was. I'd guess that a reboot will help resolve this. If it continues to persist, try to capture the logs with the information what put the Supervisor into freeze state and file an issue with the Supervisor.

closed 2022-07-08 by github-actions[bot]

RubenKelevra, 2022-07-08:
> Something put the Supervisor in `CoreState.FREEZE`, which prevents any execution. From the snapshot its not clear what that exactly was. I'd guess that a reboot will help resolve this. If it [...]

I've installed the 2022.7.1 version of HA which was released in the meantime and (without a host reboot) the issue went away.

Can I somehow check the state without running a Home Assistant OS update, to see if it may reoccur?

RubenKelevra, 2022-07-08:
> Something put the Supervisor in CoreState.FREEZE, which prevents any execution.

I feel like the UX isn't great for this message, is I don't have any idea what it means or how to solve it.

Maybe it can be dealed with by informing that there's something already running and he/she should retry later or something along this lines? 🤔

agners, 2022-07-08:
Agreed, something along these lines would probably help people better understand.

But this is the wrong repository as this is not related to OS directly. It belongs to the frontend or Supervisor repository.