home-assistant/operating-system #2017: Rpi 3+ Rev. B (64-bit)

erdnuesse, 2022-07-12:
### Describe the issue you are experiencing

Web interface does not come up after fresh install

### What operating system image do you use?

rpi3 (Raspberry Pi 3 32-bit OS)

### What version of Home Assistant Operating System is installed?

haos_rpi3-8.2

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Image push to SD
2. Boot Rpi with SD
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
no idea. didn't know what to look for.
It was right after installation / image pushing, first boots, so I had to assume a not working image tbh.
```

### Anything in the Host logs that might be useful for us?

```txt
no idea. didn't know what to look for.
It was right after installation / image pushing, first boots, so I had to assume a not working image tbh.
```

### System Health information

N/A

### Additional information

I have a Homelab, and I block outbound DNS and NTP from any devices that are not my internal DNS or NTP servers.

frenck, 2022-07-12:
duplicate of #2016

erdnuesse, 2022-07-12:
> duplicate of #2016

nope. 
2 different SD-card images. Please read first. 
the 64-bit image has a different symptom. than the 32-bit one.

erdnuesse, 2022-07-12:
I described: 64 bit: web interface does say: "this can take up to 20 minutes" 
and after fixing with aforementioned workflow - THEN the web iface does not even load. 

32bit: the web iface does not even come up at first boot. 

if these are the same, even with different symptoms, then sure, it's a duplicate. 

if a webinterface with *please wait* and no webinterface at all sound different to you, then no, it's most likely something else.