home-assistant/operating-system #2021: generic-aarch64 does not have all 32bit ARM extensions available to run 32bit binaries

jens-maus, 2022-07-15:
### Describe the issue you are experiencing

After some user reported issues to me (cf. https://homematic-forum.de/forum/viewtopic.php?f=65&t=74823) regarding using my "RaspberryMatic CCU" HA add-on in his `generic-aarch64` installation under macOS (UTM virtualization) I actually investigated the reported issue myself by setting up a virtual `generic-aarch64` system myself under macOS UTM using an Apple Silicon M1 system.

Net result of the investigation was/is, that due to development restrictions, the "RaspberryMatic CCU" HA add-on is not only using a 64bit environment (e.g. for aarch64), but always has to ship also a 32bit multi-arch environment with it so that 32bit only (armhf) binaries can be executed in such a aarch64 system as well. This perfectly works with a RaspberryPi3 running in aarch64, but also with the `generic-x86_64` HAos platform this isn't an issue.

However, as it seems the relatively new `generic-aarch64` platform might come with a Linux kernel that has all the necessary 32bit extensions disabled to allow to deal with such multi-arch environments like the "RaspberryMatic CCU" HA add-on is relying on.

Thus, it seems that the aarch64 kernel defconfig for the `generic-aarch64` requires some more attention and additional `CONFIG_XXXX` settings here so that the kernel will ship the necessary extensions for such multi-arch environments to be usable like this is currently the case for the `generic-x86_64` or even the `rpi3_64` target platforms.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

8.3

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Install a `generic-aarch64` HomeAssistantOS on e.g. macOS using UTM (https://mac.getutm.app/)
2. After HA setup, try to install the "RaspberryMatic CCU" add-on.
3. Start the add-on and not the ERROR output in the startup log when it tries to start "ReGaHss" which actually is a 32bit binary.

### Anything in the Supervisor logs that might be useful for us?

```txt
n/a
```

### Anything in the Host logs that might be useful for us?

```txt
n/a
```

### System Health information

n/a

### Additional information

See here an output/image of an interactive dive-in session into the RaspberryMatic CCU docker container demonstrating the issue with the ReGaHss 32bit binary:

[image]

Please note that the same applies for all other 32bit binaries in place within the RaspberryMatic CCU Add-on environment.

P.S.:
I had to select `generic-x86_64` here in the ticket template because `generic-aarch64` hasn't been added yet to the template.

agners, 2022-07-18:
From what I know allowing to run 32-bit Arm binaries is supported through `CONFIG_COMPAT` (see https://github.com/torvalds/linux/blob/master/arch/arm64/Kconfig#L1442). However, this option seems to be enabled for the `generic-aarch64` machine.

However, running 32-bit Arm on aarch64 also needs hardware support. And it seems that this is missing for Apple's M1 (see https://news.ycombinator.com/item?id=27277351). Unfortunately, I don't think anything can be done here from the operating system side.

jens-maus, 2022-07-18:
Oh, you are of course damn right here :) I completely missed that point, or better, did not expect Apple having wiped 32bit compatibility from their Apple Silicon chips altogether.  While this of course explains the issue, it does not solve the issues on my "RaspberryMatic CCU" Add-on end because I cannot unfortunately make it 64bit only in the near future. So thanks for point your finger into the right direction even thought I would have wished to get a more satisfacting answer to my issue here.

closed 2022-07-29 by agners