home-assistant/operating-system #2024: Cannot update Home Assistant OS

czrogers13, 2022-07-19:
### Describe the issue you are experiencing

I’m trying to update the OS, but every time I attempt to the process doesn’t work. I tell it to update, it runs, reboots, and comes back online, but doesn’t change to the newer version. The only way I’ve found to get to newer versions is to re-flash my SD card entirely and start over from scratch. This issue has been ongoing for about a year.

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

7.6

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Open Settings
2. Select Home Assistant Operating System under Updates Available
3. Select Install
4. Wait for system to reboot
5. Check System info and find that nothing changed and the same update is listed as available again

### Anything in the Supervisor logs that might be useful for us?

```txt
Not sure
```

### Anything in the Host logs that might be useful for us?

```txt
Not sure
```

### System Health information

## System Health

version | core-2022.7.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.10.103-v8
arch | aarch64
timezone | America/Chicago
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | July 26, 2022, 7:00 PM
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
remote_server | us-east-1-0.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.6
-- | --
update_channel | stable
supervisor_version | supervisor-2022.07.0
agent_version | 1.2.1
docker_version | 20.10.9
disk_total | 58.7 GB
disk_used | 5.4 GB
healthy | true
supported | true
board | rpi3-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.3)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
views | 2
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | July 11, 2022, 3:16 PM
-- | --
current_recorder_run | July 14, 2022, 10:05 AM
estimated_db_size | 33.73 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

agners, 2022-07-19:
This is likely a duplicate of #1978.

> This issue has been ongoing for about a year.

You claim to use OS 7.6, which got released on Mar 28, 2022. OS 8.0 got only released in May, so this can be only 2-3 months.

Unless you had the same problems with previous updates?

czrogers13, 2022-07-19:
Closing this, as @agners points out it appears to be a similar issue to #1978. Will reopen if they diverge.

closed 2022-07-19 by czrogers13