home-assistant/operating-system #2026: Intel AX210NGW Wifi adapter Is not detected in the system

snow-mix, 2022-07-20:
### Describe the issue you are experiencing

PC: beelink u59 + m2 module ax210ngw
After connecting the m2 module wi-fi 6 ax210, WiFi is not detected in the system. Attached the dmesg log.
[dmesg_wifi_ax210_error.log](https://github.com/home-assistant/operating-system/files/9153049/dmesg_wifi_ax210_error.log)

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

8.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.Install system
2.learn dmesg

### Anything in the Supervisor logs that might be useful for us?

```txt
No errors have been reported
```

### Anything in the Host logs that might be useful for us?

```txt
[    1.159415] Intel(R) Wireless WiFi driver for Linux
[    1.159795] iwlwifi 0000:01:00.0: enabling device (0000 -> 0002)
[    1.190470] iwlwifi 0000:01:00.0: api flags index 2 larger than supported by driver
[    1.190498] iwlwifi 0000:01:00.0: TLV_FW_FSEQ_VERSION: FSEQ Version: 0.63.2.1
[    1.191004] iwlwifi 0000:01:00.0: loaded firmware version 66.55c64978.0 ty-a0-gf-a0-66.ucode op_mode iwlmvm
[    1.212553] iwlwifi 0000:01:00.0: Detected Intel(R) Wi-Fi 6 AX210 160MHz, REV=0x420
[    1.381070] iwlwifi 0000:01:00.0: Detected RF GF, rfid=0x10d000
[    1.382025] iwlwifi 0000:01:00.0: Microcode SW error detected. Restarting 0x0.
[    1.382513] iwlwifi 0000:01:00.0: Start IWL Error Log Dump:
[    1.382515] iwlwifi 0000:01:00.0: Transport status: 0x0000004A, valid: 6
[    1.382518] iwlwifi 0000:01:00.0: Loaded firmware version: 66.55c64978.0 ty-a0-gf-a0-66.ucode
[    1.382521] iwlwifi 0000:01:00.0: 0x00000071 | NMI_INTERRUPT_UMAC_FATAL    
[    1.382524] iwlwifi 0000:01:00.0: 0x002002F0 | trm_hw_status0
[    1.382526] iwlwifi 0000:01:00.0: 0x00000000 | trm_hw_status1
[    1.382528] iwlwifi 0000:01:00.0: 0x004DA1A2 | branchlink2
[    1.382530] iwlwifi 0000:01:00.0: 0x004D0766 | interruptlink1
[    1.382532] iwlwifi 0000:01:00.0: 0x004D0766 | interruptlink2
[    1.382534] iwlwifi 0000:01:00.0: 0x004D8F5A | data1
[    1.382536] iwlwifi 0000:01:00.0: 0x00000010 | data2
[    1.382538] iwlwifi 0000:01:00.0: 0x00000000 | data3
[    1.382540] iwlwifi 0000:01:00.0: 0x00000000 | beacon time
[    1.382542] iwlwifi 0000:01:00.0: 0x00013A61 | tsf low
[    1.382544] iwlwifi 0000:01:00.0: 0x00000000 | tsf hi
[    1.382546] iwlwifi 0000:01:00.0: 0x00000000 | time gp1
[    1.382548] iwlwifi 0000:01:00.0: 0x00027C1A | time gp2
[    1.382550] iwlwifi 0000:01:00.0: 0x00000001 | uCode revision type
[    1.382552] iwlwifi 0000:01:00.0: 0x00000042 | uCode version major
[    1.382554] iwlwifi 0000:01:00.0: 0x55C64978 | uCode version minor
[    1.382556] iwlwifi 0000:01:00.0: 0x00000420 | hw version
[    1.382558] iwlwifi 0000:01:00.0: 0x00C89002 | board version
[    1.382560] iwlwifi 0000:01:00.0: 0x8008FF05 | hcmd
[    1.382562] iwlwifi 0000:01:00.0: 0x00020000 | isr0
[    1.382564] iwlwifi 0000:01:00.0: 0x60000000 | isr1
[    1.382565] iwlwifi 0000:01:00.0: 0x48F00002 | isr2
[    1.382567] iwlwifi 0000:01:00.0: 0x00C0000C | isr3
[    1.382569] iwlwifi 0000:01:00.0: 0x00000000 | isr4
[    1.382571] iwlwifi 0000:01:00.0: 0x00000000 | last cmd Id
[    1.382573] iwlwifi 0000:01:00.0: 0x004D8F5A | wait_event
[... 237 lines cut ...]
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.3), Terminal & SSH (9.6.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | July 10, 2022 at 2:38 PM
-- | --
current_recorder_run | July 19, 2022 at 10:17 PM
estimated_db_size | 3.78 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

agners, 2022-07-27:
This looks very much like a kernel bug. Can you try OS 8.4, as it has the latest stable kernel?

If that does not help, bumping to the latest firmware might help, but this will need a OS change.

What ax210ngw M.2 model are you using exactly? I see there are various variants.

WaszaMac, 2022-08-09:
i shall connect with this thread... Home Assistant OC on KVM VM do not work with Intel AX200NGW Wi-Fi m.2 card  (passthrou from host) it does not see the card with "nmcli devices" (but nmcli radio suggests everything is good). I am not a log specialist, so I only speculate that it is a driver problem. I try few Live Linux distrios and all of them fount this card.
The question is how to update drivers in cli/ terminal? as all known to me (I am a noob) commands/ apps/ tools (whatever we call them) from BIG Linux distros do not work in HA OS :(