home-assistant/operating-system #2031: Can’t update from 8.2

astirling01, 2022-07-26:
### Describe the issue you are experiencing

I haven’t been able to update HAOS since updating to 8.1. Installation appears to run but then stops without any obvious error message. Couldn’t install the last update and 8.4 is still not working 

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

8.1

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Update fails every time for me 
2.
3.

### Anything in the Supervisor logs that might be useful for us?

```txt
Only obvious error I can see is..

ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Failed marking slot kernel.1 non-bootable: Failed marking 'kernel.1' as bad: Failed to run grub-editenv: Child process exited with code 1
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing of note
```

### System Health information

## System Health

version | core-2022.7.7
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.41
arch | x86_64
timezone | Europe/London
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.26.2
Stage | running
Available Repositories | 1138
Downloaded Repositories | 8

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.1
-- | --
update_channel | stable
supervisor_version | supervisor-2022.07.0
agent_version | 1.2.1
docker_version | 20.10.14
disk_total | 116.7 GB
disk_used | 5.4 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.6.0), Home Assistant Google Drive Backup (0.108.2), Duck DNS (1.15.0), Studio Code Server (5.2.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 3
views | 8
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 16 July 2022, 09:12
-- | --
current_recorder_run | 26 July 2022, 18:17
estimated_db_size | 189.31 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

### Additional information

N/a

agners, 2022-07-27:
Likely the rauc logs will have more detailed information.

If you have [SSH access to the underlying OS](https://developers.home-assistant.io/docs/operating-system/debugging#ssh-access-to-the-host), please try to get them with the following commands:

```
journalctl -u rauc.service
```

balilu91, 2022-07-27:
Having same issue. Coudn't update from 8.1 to 8.2 and now I have the same issue to update from 8.1 (still) to 8.4

`Home Assistant Operating System update failed with: Installation error: Failed marking slot kernel.1 non-bootable: Failed marking 'kernel.1' as bad: Failed to run grub-editenv: Child process exited with code 1`

Running on NUC 8iBEH

This might be a duplicate of #1975 

Will try to set up ssh access to the OS

## System Health

version | core-2022.7.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.41
arch | x86_64
timezone | Europe/Malta
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4608
Installed Version | 1.26.2
Stage | running
Available Repositories | 1152
Downloaded Repositories | 30

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.1
-- | --
update_channel | stable
supervisor_version | supervisor-2022.07.0
agent_version | 1.2.1
docker_version | 20.10.14
disk_total | 109.3 GB
disk_used | 69.8 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Node-RED (13.2.0), InfluxDB (4.5.0), Studio Code Server (5.2.0), Grafana (7.6.0), Duck DNS (1.15.0), MariaDB (2.5.0), Mosquitto broker (6.1.2), Home Assistant Google Drive Backup (0.108.2), SSH & Web Terminal (11.0.2), Samba share (10.0.0), FTP (4.6.0), phpMyAdmin (0.8.1), Vaultwarden (Bitwarden) (0.17.1), Nginx Proxy Manager (0.12.1), Matrix (0.10.0), Glances (0.16.0), Frigate NVR (Full Access) (3.1), Promtail (2.1.0), Loki (1.10.2), Uptime Kuma (0.2.0), chrony (2.4.0), ESPHome (2022.6.2), AdGuard Home (4.6.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 19
views | 8
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | July 12, 2022 at 12:48 PM
-- | --
current_recorder_run | July 27, 2022 at 10:08 PM
estimated_db_size | 2045.61 MiB
database_engine | mysql
database_version | 10.6.8

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

balilu91, 2022-07-28:
Ran 
`journalctl -u rauc.service`

`Jul 28 18:36:56 homeassistant rauc[362]: Using deprecated 'Install' D-Bus Method (replaced by 'InstallBundle')
Jul 28 18:36:56 homeassistant rauc[362]: Active slot bootname: A
Jul 28 18:36:56 homeassistant rauc[362]: input bundle: /mnt/data/supervisor/tmp/hassos-8.4.raucb
Jul 28 18:36:56 homeassistant rauc[362]: installing /mnt/data/supervisor/tmp/hassos-8.4.raucb: started
Jul 28 18:36:56 homeassistant rauc[362]: installing /mnt/data/supervisor/tmp/hassos-8.4.raucb: Checking and mounting bundle...
Jul 28 18:36:56 homeassistant rauc[362]: Reading bundle: /mnt/data/supervisor/tmp/hassos-8.4.raucb
Jul 28 18:36:56 homeassistant rauc[362]: Detected CRL but CRL checking is disabled!
Jul 28 18:36:56 homeassistant rauc[362]: Verifying bundle signature...
Jul 28 18:36:57 homeassistant rauc[362]: Verified detached signature by 'O = HassOS, CN = HassOS GitHub-1'
Jul 28 18:36:57 homeassistant rauc[362]: Mounting bundle '/mnt/data/supervisor/tmp/hassos-8.4.raucb' to '/run/rauc/bundle'
Jul 28 18:36:57 homeassistant rauc[362]: Configured loop device '/dev/loop0' for 229822464 bytes
Jul 28 18:36:57 homeassistant rauc[362]: Running bundle hook install-check
Jul 28 18:36:57 homeassistant rauc[362]: Marking target slot kernel.1 as non-bootable...
Jul 28 18:36:57 homeassistant rauc[131137]: grub-editenv: error: invalid environment block.
Jul 28 18:36:57 homeassistant rauc[362]: Installation error: Failed marking slot kernel.1 non-bootable: Failed marking 'kernel.1' as bad: Failed to run grub-editenv: Child process exited with code 1
Jul 28 18:36:57 homeassistant rauc[362]: installing /mnt/data/supervisor/tmp/hassos-8.4.raucb: Installation error: Failed marking slot kernel.1 non-bootable: Failed marking 'kernel.1' as bad: Failed to run grub-editenv: Child process exited with code 1
Jul 28 18:36:57 homeassistant rauc[362]: installing /mnt/data/supervisor/tmp/hassos-8.4.raucb: finished
Jul 28 18:36:57 homeassistant rauc[362]: installing `/mnt/data/supervisor/tmp/hassos-8.4.raucb` failed: 1
`

agners, 2022-07-29:
```
Jul 28 18:36:57 homeassistant rauc[131137]: grub-editenv: error: invalid environment block.
```

This seems to be the culprit. Can you check the size and the content of the GRUB environment using:

```
ls -la /mnt/boot/EFI/BOOT/grubenv
cat /mnt/boot/EFI/BOOT/grubenv
```

agners, 2022-07-29:
It does look like file system corruption, and we used to have such reports in the past. Unfortunately, I have no idea where they are coming from. When properly shutdown, HAOS should unmount the file system.

Can you check if there were file system checks necessary recently:

```
journalctl -u mnt-boot.mount
journalctl  -u systemd-fsck@dev-disk-by\\x2dlabel-hassos\\x2dboot.service
```

balilu91, 2022-07-29:
@agners 

grubenv
-rwxr-xr-x    1 root     root          1024 May 23 21:42 grubenv

it's 1024 .. however using cat nothing returns nothing like its empty 

using vi I get 

> ^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^

`journalctl -u mnt-boot.mount`

> Jul 27 19:40:03 homeassistant systemd[1]: Mounting HassOS boot partition...
> Jul 27 19:40:03 homeassistant systemd[1]: Mounted HassOS boot partition.

`journalctl  -u systemd-fsck@dev-disk-by\\x2dlabel-hassos\\x2dboot.service`

> Jul 27 19:40:03 homeassistant systemd[1]: Starting File System Check on /dev/disk/by-label/hassos-boot...
> Jul 27 19:40:03 homeassistant systemd-fsck[224]: Cannot initialize conversion from codepage 850 to ANSI_X3.4-1968: Invalid argument
> Jul 27 19:40:03 homeassistant systemd-fsck[224]: Cannot initialize conversion from ANSI_X3.4-1968 to codepage 850: Invalid argument
> Jul 27 19:40:03 homeassistant systemd-fsck[224]: Using internal CP850 conversion table
> Jul 27 19:40:03 homeassistant systemd-fsck[224]: fsck.fat 4.2 (2021-01-31)
> Jul 27 19:40:03 homeassistant systemd-fsck[224]: /dev/sda1: 14 files, 537/16343 clusters
> Jul 27 19:40:03 homeassistant systemd[1]: Finished File System Check on /dev/disk/by-label/hassos-boot.

let me know what else you need

balilu91, 2022-07-29:
Noticed the commit you did [85876c5](https://github.com/agners/operating-system/commit/85876c5e2a067dca61d7ea85dc1d40373effa8ff) 

Ran `grub-editenv /mnt/boot/EFI/BOOT/grubenv create`

Later was able to update successfully 

> Jul 29 18:56:06 homeassistant rauc[362]: Using deprecated 'Install' D-Bus Method (replaced by 'InstallBundle')
> Jul 29 18:56:06 homeassistant rauc[362]: Active slot bootname: A
> Jul 29 18:56:06 homeassistant rauc[362]: input bundle: /mnt/data/supervisor/tmp/hassos-8.4.raucb
> Jul 29 18:56:06 homeassistant rauc[362]: installing /mnt/data/supervisor/tmp/hassos-8.4.raucb: started
> Jul 29 18:56:06 homeassistant rauc[362]: installing /mnt/data/supervisor/tmp/hassos-8.4.raucb: Checking and mounting bundle...
> Jul 29 18:56:06 homeassistant rauc[362]: Reading bundle: /mnt/data/supervisor/tmp/hassos-8.4.raucb
> Jul 29 18:56:06 homeassistant rauc[362]: Detected CRL but CRL checking is disabled!
> Jul 29 18:56:06 homeassistant rauc[362]: Verifying bundle signature...
> Jul 29 18:56:06 homeassistant rauc[362]: Verified detached signature by 'O = HassOS, CN = HassOS GitHub-1'
> Jul 29 18:56:06 homeassistant rauc[362]: Mounting bundle '/mnt/data/supervisor/tmp/hassos-8.4.raucb' to '/run/rauc/bundle'
> Jul 29 18:56:06 homeassistant rauc[362]: Configured loop device '/dev/loop0' for 229822464 bytes
> Jul 29 18:56:06 homeassistant rauc[362]: Running bundle hook install-check
> Jul 29 18:56:06 homeassistant rauc[362]: Marking target slot kernel.1 as non-bootable...
> Jul 29 18:56:06 homeassistant rauc[362]: installing /mnt/data/supervisor/tmp/hassos-8.4.raucb: Updating slots...
> Jul 29 18:56:06 homeassistant rauc[362]: installing /mnt/data/supervisor/tmp/hassos-8.4.raucb: Checking slot boot.0
> Jul 29 18:56:07 homeassistant rauc[362]: Updating /dev/disk/by-partlabel/hassos-boot with 'install' slot hook
> Jul 29 18:56:07 homeassistant rauc[362]: installing /mnt/data/supervisor/tmp/hassos-8.4.raucb: Updating slot boot.0
> Jul 29 18:56:07 homeassistant rauc[362]: Running slot hook 'slot-install' for boot.0
> Jul 29 18:56:07 homeassistant rauc[362]: installing /mnt/data/supervisor/tmp/hassos-8.4.raucb: Updating slot boot.0 status
> Jul 29 18:56:07 homeassistant rauc[362]: Checking image type for slot type: raw
> Jul 29 18:56:07 homeassistant rauc[362]: Image detected as type: *.img
> Jul 29 18:56:07 homeassistant rauc[362]: installing /mnt/data/supervisor/tmp/hassos-8.4.raucb: Updating slot boot.0 done
> Jul 29 18:56:07 homeassistant rauc[362]: installing /mnt/data/supervisor/tmp/hassos-8.4.raucb: Checking slot kernel.1
> Jul 29 18:56:07 homeassistant rauc[362]: Updating /dev/disk/by-partlabel/hassos-kernel1 with /run/rauc/bundle/kernel.img
> Jul 29 18:56:07 homeassistant rauc[362]: installing /mnt/data/supervisor/tmp/hassos-8.4.raucb: Updating slot kernel.1
> Jul 29 18:56:07 homeassistant rauc[362]: opening slot device /dev/disk/by-partlabel/hassos-kernel1
> Jul 29 18:56:07 homeassistant rauc[362]: writing data to device /dev/disk/by-partlabel/hassos-kernel1
> Jul 29 18:56:07 homeassistant rauc[362]: Running slot hook 'slot-post-install' for kernel.1
> Jul 29 18:56:07 homeassistant rauc[362]: installing /mnt/data/supervisor/tmp/hassos-8.4.raucb: Updating slot kernel.1 status
> Jul 29 18:56:07 homeassistant rauc[362]: Checking image type for slot type: raw
> Jul 29 18:56:07 homeassistant rauc[362]: installing /mnt/data/supervisor/tmp/hassos-8.4.raucb: Updating slot kernel.1 done
> Jul 29 18:56:07 homeassistant rauc[362]: Image detected as type: *.img
> Jul 29 18:56:07 homeassistant rauc[362]: installing /mnt/data/supervisor/tmp/hassos-8.4.raucb: Checking slot rootfs.1
> Jul 29 18:56:07 homeassistant rauc[362]: Updating /dev/disk/by-partlabel/hassos-system1 with /run/rauc/bundle/rootfs.img
> Jul 29 18:56:07 homeassistant rauc[362]: installing /mnt/data/supervisor/tmp/hassos-8.4.raucb: Updating slot rootfs.1
> Jul 29 18:56:07 homeassistant rauc[362]: opening slot device /dev/disk/by-partlabel/hassos-system1
> Jul 29 18:56:07 homeassistant rauc[362]: writing data to device /dev/disk/by-partlabel/hassos-system1
> Jul 29 18:56:10 homeassistant rauc[362]: installing /mnt/data/supervisor/tmp/hassos-8.4.raucb: Updating slot rootfs.1 status
> Jul 29 18:56:10 homeassistant rauc[362]: Marking target slot kernel.1 as bootable...
> Jul 29 18:56:10 homeassistant rauc[362]: installing /mnt/data/supervisor/tmp/hassos-8.4.raucb: Updating slot rootfs.1 done
> Jul 29 18:56:10 homeassistant rauc[362]: installing /mnt/data/supervisor/tmp/hassos-8.4.raucb: All slots updated
> Jul 29 18:56:10 homeassistant rauc[362]: installing /mnt/data/supervisor/tmp/hassos-8.4.raucb: finished
> Jul 29 18:56:10 homeassistant rauc[362]: installing `/mnt/data/supervisor/tmp/hassos-8.4.raucb` succeeded

astirling01, 2022-07-29:
Good to hear. Do I need to SSH into the underlying OS to fix?

balilu91, 2022-07-29:
It seems that command only works from the underlying OS.

Not an expert here so not sure if it's possible to run it in some other way

astirling01, 2022-07-29:
Oh ok.. will give it a go over the weekend

agners, 2022-08-04:
@balilu91 thanks for confirming that this fixes it!

@astirling01 now that the work around is proven to be helpful, I'll merge and backport it to release 8. So once the next release 8 is out, update should work for you.

Thanks for your help to get to the bottom of this.

closed 2022-08-04 by agners