home-assistant/operating-system #2032: Upgrade from 8.2 to 8.4 failed and made system inaccessible.

bfellner, 2022-07-27:
### Describe the issue you are experiencing

Mine didn't work at all after upgrading 8.2 to 8.4.  I did the upgrade and then I could no longer access the machine.  It wasn't even pulling the static DHCP IP address (not responding to ping).  I also had nothing showing when I plugged in a monitor.  I took the drive out and proceeded to reinstall clean with 8.4.  Same thing.  I'm going to revert back to 8.2. 

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

8.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. used the UI to upgrade from OS from 8.2 to 8.4
2. After reboot the system was inaccessible.  
3. Tried pinging IP Address (Static assignment from DHCP) and it was unresponsive.
4. Checked DHCP Server for lease and it was not active
5. Plugged in a monitor, nothing displaying.
6. Waited 2 hours and rebooted machine
7. Nothing displaying on monitor
8. Same results.
9. Removed drive from machine
10. Attached drive to desktop computer
11. Flashed 8.4 image with Balena
12. Reinstalled drive
13. Same results (no display on monitory and inaccessible)
14. Will revert to 8.2 reflashing with Balena to get back up and running
...

### Anything in the Supervisor logs that might be useful for us?

```txt
cannot access
```

### Anything in the Host logs that might be useful for us?

```txt
cannot access
```

gferen, 2022-07-27:
exactly the same here. upon checking SD card - boot partition ok, data/os gone

agners, 2022-07-27:
> 7. Nothing displaying on monitor

As this is a x86-64, I'd expect that at the very least the BIOS should show, then the boot loader, and at least part of kernel booting.

If the GRUB boot loader appears, pulling the plug three times (after when GRUB starts), should make GRUB to revert to the old OS installation.

As almost nothing else changed, this seems to be a stable kernel breakage. Can you try to update to the latest dev, it uses 5.15.57 already:

```
ha su options --channel=dev
ha su reload
ha os update
```

Switch back to stable channel using:
```
ha su options --channel=stable
ha su reload
```

ve6rah, 2022-07-27:
I seem to have run across the same issue. Console showed "Waiting for the Home Assistant CLI to be ready..." and then showed  an error message about the CLI failing to load.
I loaded the backup slot on boot, and that brought me back up on 8.2, I then ran the core update from 2022.7.6 to 2022.7.7, then re-ran the OS update to 8.4 and everything works fine now.

Not sure if the issue was related to not doing the core update first? or just trying again got it, but something wasn't quite smooth here this time!

agners, 2022-07-28:
> Console showed "Waiting for the Home Assistant CLI to be ready..." and then showed an error message about the CLI failing to load.

The script showing the console only waits a specific time, for the CLI plugin to come up. This could be delayed due to Supervisor or plug-in updates.

Rubquila, 2022-08-05:
> I seem to have run across the same issue. Console showed "Waiting for the Home Assistant CLI to be ready..." and then showed an error message about the CLI failing to load. I loaded the backup slot [...]

Same issue, same solution.

caheredia, 2022-08-13:
I'm seeing the same bug. I'm running an HA VM inside of proxmox. Every time I try to update to 8.4 the system reboots and fails to boot up. So I have to restore the backup.

pst11790, 2022-08-18:
Same issue here; home Assistant OS x86-64 on a Dell Optiplex. After update, home Assistant never comes back online. I have to recover it by hard reset to get home Assistant loaded & it is still loading 8.2 after hard reset.

closed 2022-11-24 by github-actions[bot]