home-assistant/operating-system #2033: Upgrade HASS OS from 8.2 to 8.4 failed and made system inaccessible. RPi4

gferen, 2022-07-27:
### Describe the issue you are experiencing

after update hassio to latest frontend, there was os update waiting - 8.4  os update.
OS update seems never completed or it is stocked somewhere. No ping to device, no any response.
upon examination of SD card hassos-boot partition is ok,  the other partition with OS and data - is gone. At least windows sees it as unallocated.
Wait time before checking SD - force restarting PRi4 12 hours of waiting.

### What operating system image do you use?

rpi4  (Raspberry Pi 4/400 32-bit OS)

### What version of Home Assistant Operating System is installed?

8.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. used the UI to upgrade from OS from 8.2 to 8.4
2. waiting for upgrade to complete overnight (10+ hours)
3. After reboot the system was inaccessible.
4. pinging IP Address - no responsive.
5. Removed SD Card from machine
6. Attached drive to desktop computer
7. data partition missing - unallocated as windows says, boot partition ok

### Anything in the Supervisor logs that might be useful for us?

```txt
no access
```

### Anything in the Host logs that might be useful for us?

```txt
no access
```

gferen, 2022-07-27:
i managed to get system.journal  file  will it help

agners, 2022-07-27:
Anything on the screen?

Can you try to pull the plug three times? If U-Boot still works, this should cause the system to boot back into the previous OS installation.

> i managed to get system.journal file will it help

Maybe, it depends if the system booted as far.

gferen, 2022-07-27:
3 times interrupting boot did not help. 
attached are logs.

But now when i already decided to reinstall - at least in windows - no tool was able to format SD card. so i would as well suspect that SD got screwed.
[logs.zip](https://github.com/home-assistant/operating-system/files/9195987/logs.zip)

agners, 2022-07-27:
Yeah the logs don't show much, it seems the update got downloaded, then the logs end. That makes the SD card breakage quite plausible as well. Closing then, thanks for the update/logs.

closed 2022-07-27 by agners

gferen, 2022-07-27:
i managed to dig out logs from one of the partitions.  home-assistant.log