home-assistant/operating-system #2037: zigbee not working with SSD Boot on Pi4

Gargy, 2022-07-30:
### Describe the issue you are experiencing

I have an unusual probelm...

I have RPI4 running from a SD card, with ZWave and ConBeeII USB sticks. The system was running without a hiccup but with time was getting noticeable slower... So I decided to run my PRi from external SSD. So I bought Argon One M.2 case together with a Verbatim 256 SATAIII SSD. Flashed it with Pi Imager, put everything together and boot it up. System booted just fine and I start to config everything. Set the ZWave just fine (detected on USB port and it also find all of my nodes) and it also found a ZHA configuration. ConBeeII was correctly detected and I set up the ZHA Integration. Here the problem(s) started - I doesn't find any Zigbee devices. It just searching and searching but never connects. I the tried with deConz integrations (deleted ZHA and restarted), it find ConBeeII just fine, but again - it doesn't find any devices. So I tried with Sonoff Zigbee 3.0, same thing. I then tried both of the Zigbee sticks on PC with Hassio in VM, same HA version, same integrations (both deConz and ZHA) and they work flawlessly. I also tried different cases if One M.2 has some issues, but without luck. I tried every possible USB port combination.
I then installed fresh HA on SD card, tried with the same HW setup and it works. Both sticks, same RPi4, on both cases.
I did look into logs but I only found some errors regarding zpy and something with serial ports. I sadly didn't saved them...

Did anybody experienced something similar or did I missed something in setting up HA?

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

8.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.  Installed HA on SSD
2. Flashed RPi4 with BOOT version
3. Tried with Sonoff Zigbee 3.0 an ConBeeII USB stick
4. Tried with ZHA and deConz integration
5. It works with SD card

### Anything in the Supervisor logs that might be useful for us?

```txt
Some zpy zigbee lines with serial port exceptions or errors
```

### Anything in the Host logs that might be useful for us?

```txt
Some zpy zigbee lines with serial port exceptions or errors
```

carsten-h, 2022-07-31:
How did you connect the ConBee II to the Pi?
If you read the installation instructions from Dresden Electronic you will find that you have to connect it with an USB extension cable to the Pi! I also would attach it only to an USB 2 port and not to a USB 3 one because USB 3 is also sending in the 2,4 GHz range.

For Zwave I can tell you nothing as I am not using it.

Gargy, 2022-08-01:
ConBeeII is working on RPI4 when it is running from SD card. Same HW layout, only thing changed is running RPi4 from SSD and not SD card. System recognizes ConBeeII, but when I start the pairing process, nothing happens. It is the same with Sonoff Zigbee 3.0. It doesn't matter in which USB port I plug it in. As soon I revert back to SD card, everything is fine. 

Z-Wave is working properly on SD and SSD.

allan-reynolds, 2022-08-04:
Are you connecting your ssd via USB3, this could be as others have mentioned the issue.  USB3 can interfere with 2.4Ghz radios at close range. I know that some of my ESP devices are not always able to communicate over wifi if my USB3 Hub cable is between them and the WAP.  If I move them to the other side of the cable they work just fine. 

When you are using the SD card and the USB3/ssd is not in use then the interference will be less and things my well work.  If the SSD is on the end of a cable, even a short one it may well be acting as an antenna and interfering with zigbee signal if it is too close (ie both plugged directly  into the Pi ports).  This is why there is advice to connect to the USB2 ports using an extension cable.  This reduces the interference seen by the zigbee adapter by keeping it and the 'noise' from USB3 apart .  I have my zigbee adapter on the end of a 1M USB2 extension and it works OK even with several USB3 devices near the Pi.

Gargy, 2022-08-04:
If I use SD card, then it doesn't matter where the Zigbee stick is. It works in USB3 ports just fine - ConBeeII and Sonoff. If I use SSD, then there is only a small USB bridge that connects SSD and RPi4 (on USB3 port). In that case, both Zigbee and ZWave USB sticks are in USB2 ports. I didn't tried with extension cable, will try to do that next. But I did try a USB hub in USB2 port with same results - ZWave works, Zigbee not (even Sonoff 3.0 which has metal housing and external antenna).

carsten-h, 2022-08-04:
Look at the installation instructions of the ConBee II stick: [](https://phoscon.de/en/conbee2/install)

Gargy, 2022-08-14:
Hi all! 

After putting ConBeeII and ZWave on extension cables, everything works. So indeed it was some interference trough USB3 port, where SSD is. Which is weird, because it couldn't pick up a Zigbee device even if it was 1cm away... Anyway, it is OK now. Lesson for next time...

closed 2022-08-14 by Gargy