home-assistant/operating-system #2042: NSURLErrorDomain - 1200 - Cannot connect externally

vandad51, 2022-08-04:
### Describe the issue you are experiencing

So I've tried getting help from reddit and from Discord but no one seemed to have any answers for me.

Recently I changed my wifi channel (my router made it difficult to do so simply) and my devices didn’t like something so I ended up doing a factory reset on my router. Devices all connected back once they realized it was ok to do so.

I still have full internet access at home. HA runs totally fine. I’m running AdGuard with DHCP and that’s running fine, no errors in logs. I just cannot access HA outside anymore. I get a “NSURLerrorDomain 1200” error when accessing from my iPhone externally. When accessing from a desktop, I get ERR_FAILED (this site can’t be reached)

I port forwarded 443 to 8123 to my HA on VirtualBox. Port checkers online say it’s open. I get no access errors on my logs. I reinstalled duckdns and nginx SSL proxy, also no errors in those. I updated virtual box to 6.1.36r to be sure that wasn't the issue. I’m at latest the HAOS version. No addons have any updates available.

I can't seem to find anything on that error code relating to HA. 

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

8.4

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

I simply cannot connect to HA from outside my network.

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing applicable
```

### Anything in the Host logs that might be useful for us?

```txt
Nope.
```

### System Health information

Home Assistant 2022.8.0
Supervisor 2022.07.0
Operating System 8.4
Frontend 20220802.0 - latest

### Additional information

Here's a portion of my log from my iphone when attempting to connect to HA

NSLocalizedDescription=An SSL error has occurred and a secure connection to the server cannot be made., NSErrorFailingURLKey=https://****.duckdns.org/api/webhook/e538bedbd3e371966b565bd007f55ebad8b05bf86ee6e12389cbec5399751463, NSUnderlyingError=0x111e17aa0 {Error Domain=kCFErrorDomainCFNetwork Code=-1200 "(null)" UserInfo={_kCFStreamPropertySSLClientCertificateState=0, _kCFNetworkCFStreamSSLErrorOriginalValue=-9836, _kCFStreamErrorDomainKey=3, _kCFStreamErrorCodeKey=-9836, _NSURLErrorNWPathKey=satisfied (Path is satisfied), viable, interface: en0, ipv4, dns}}, _kCFStreamErrorCodeKey=-9836}
2022-08-03 06:58:27.638 [Info] [main] [Environment.swift:54] init() > PromiseKit:cauterized-error: Error Domain=NSURLErrorDomain Code=-1200 "An SSL error has occurred and a secure connection to the server cannot be made." UserInfo={NSErrorFailingURLStringKey=https://****.duckdns.org/api/webhook/e538bedbd3e371966b565bd007f55ebad8b05bf86ee6e12389cbec5399751463, NSLocalizedRecoverySuggestion=Would you like to connect to the server anyway?, _kCFStreamErrorDomainKey=3, _NSURLErrorFailingURLSessionTaskErrorKey=LocalUploadTask <FF60C3B9-FDBE-4ABA-821D-64D76B5823E1>.<1>, _NSURLErrorRelatedURLSessionTaskErrorKey=(
    "LocalUploadTask <FF60C3B9-FDBE-4ABA-821D-64D76B5823E1>.<1>"

agners, 2022-08-04:
This issue tracker is specifically for Home Assistant OS issues. The web server is part of Home Assistant Core, so in case it is a Core issue, it would belong there.

However, before reporting an issue, make sure it is not a local problem, e.g. by asking in the Community Forum. [This thread](https://community.home-assistant.io/t/iphone-app-error-nsurlerrordomain-1200/323277) seems to describe a similar problem.

closed 2022-08-04 by agners

vandad51, 2022-08-05:
I understand it's closed, but I just wanted to close this by saying I fixed it. I found another issue with similar conditions and read someone did the exact same mistake I did. I forgot my port forwarding settings when I set my router back up again and just forwarded 443 to 8123. But that's wrong. NGINX expects a connection from port 443 which then bumps it over to 8123 for HA. So I forwarded 443 to 443 to HA and it's fixed now.