home-assistant/operating-system #2046: Update to 8.4 not doing anything

mikeplanzer, 2022-08-04:
### Describe the issue you are experiencing

Dear community,

Since a few days I am trying to update home assistant os from 7.6 to 8.4 with no luck.

**Installation via GUI**
If I update via the UI it says “Installing” for a minute, then restarts home assistant and it’s still on the same version (7.6) as before. No records in the logs.

**Installation via Terminal**
If I try to update via Terminal it starts but after 1 second saying “_Error: Error raised from OTA Webserver: 404_”. As seen in the image below.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

7.6

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

**GUI version**
1. Click on insall from the GUI
2. OS restarts after a few minutes

Terminal version
1. Run command _ha os update --version 2022.8.4_
2. See error message _Error: Error raised from OTA Webserver: 404_

### Anything in the Supervisor logs that might be useful for us?

```txt
Only "INFO" messages apart from one warning:

22-08-04 15:42:02 WARNING (MainThread) [supervisor.host.network] Requested to update interface eth0 which does not exist or is disabled.
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing
```

### System Health information

## System Information

version | core-2022.8.0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.10.103-v8
arch | aarch64
timezone | Europe/Zurich
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 4 July 2023 at 02:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-west-2-3.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.6
-- | --
update_channel | stable
supervisor_version | supervisor-2022.07.0
agent_version | 1.2.1
docker_version | 20.10.9
disk_total | 28.6 GB
disk_used | 4.3 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (10.0.0), deCONZ (6.15.0), File editor (5.3.3), Terminal & SSH (9.6.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 4
views | 10
mode | yaml

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 25 July 2022 at 09:35
-- | --
current_recorder_run | 4 August 2022 at 15:42
estimated_db_size | 95.48 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

saeedesmaili, 2022-08-04:
I have the same issue. I keep getting the notification to update from 8.2 to 8.4, then I hit "Install" it takes a couple of minutes, the os restarts, and then in the About section, I see that the current OS version is 8.4. A few hours later (I guess after turning off/on the raspberry pi for any reason), I get the same notification to update from 8.2 to 8.4, and the About section shows 8.2 as the OS's current version! This has happened frequently in the past few weeks.

agners, 2022-08-05:
@mikeplanzer the Operating System version is not using calendar versioning currently. Try:

```
ha os update --version 8.4
```

agners, 2022-08-05:
@saeedesmaili since in your case the OS restarts, it is a different issue. Your issue looks similar to #1978. Can you report the logs requested in https://github.com/home-assistant/operating-system/issues/1978#issuecomment-1189158058 and report it in that issue?

mikeplanzer, 2022-08-05:
> @mikeplanzer the Operating System version is not using calendar versioning currently. Try:
> 
> ```
> ha os update --version 8.4
> ```

Thanks for your comment - I tried mentioned command and had high hops as it was going into "progressing..." for a few minutes and then to "command completed" and the os restarted. Unfortunately still on the old version. No entries in the logs.

agners, 2022-08-06:
Likely the installation succeeded, but for some reason the boot loader decided to fall back to the previous version (the OS always maintains two installation, this allows to fallback in case the new version does not boot successfully).

Do you use USB SSD or similar?

mikeplanzer, 2022-08-08:
> Do you use an USB SSD?

I am using an SD card where the version is flashed on to. This SD card is inside a raspberry pi 4.

mikeplanzer, 2022-08-08:
Okay guys, I didn’t solve the issue yet however to make sure It's not the fault of my current configuration/installations, here is what I did.

I flashed a brand new SD card with the latest version from github 8.4 and tried to boot my raspberry pi 4 connected to a screen. It starts booting but after 5 seconds crashes (without visible error) and reboots again _infinite_ times.

I took the same SD card and flashed version 7.6 and it boots up like a charm. Tried newer versions higher than 7.6 without success. Therefore I assume there is an issue with the newer versions not being supported by my hardware?

LuckyPants703, 2022-08-14:
I have this same issue on Intel NUC. Says "Installing..." Then it just returns to normal state without any reboots or anything. I have OS version 8.2

convicte, 2022-08-17:
Just for an additional data point, in my case, the same issue happens when I gracefully reboot/shutdown my VirtualBox VM.
This said, I am on 8.0 and updating to v.8.4.

Every time I need to power down the VM for maintenance, I get the 8.4 update notice.

kayyum-uddin, 2022-08-18:
Same issue happen with me can't able to upgrade 7.6 to 8.4, raspberry pi 4 reboot every time while try to update os.

upspowerhouse, 2022-08-18:
> I have this same issue on Intel NUC. Says "Installing..." Then it just returns to normal state without any reboots or anything. I have OS version 8.2

Same type of error experienced on Generic x86-64 system.  Or when I restart system it reverts to OS 8.2 and the update notice reappears.

convicte, 2022-08-23:
Should we assume this matter to be addresses in 8.5 in any way?

The change log doesn't indicate anything specifically related.

mikeplanzer, 2022-08-23:
> Should we assume this matter to be addresses in 8.5 in any way?

Yes - I've tried with installing 8.5 but still having the same issue.

By the way, there is also an discussion on the home-assistant forum where some others confirm the same issue:
https://community.home-assistant.io/t/unable-to-update-home-assistant-os-stuck-on-os-8-1/444328/15

agners, 2022-09-08:
@mikeplanzer that referenced community thread has a lot of problems: `ha core update --version 2022.8.4` is the wrong command, as this is to update the core. Then 2022.8.4 is *not* a valid OS version. OS versions are 8.4 or 8.5.

From your last description, it seems that you are not having an update problem, but problem with a new installation? This are different things...

In general, HAOS 8.5 has been tested with updating and new installation on a Raspberry Pi 4. So the problem is likely related to your setup, e.g. SD card has not beeing flashed properly or similar. Make sure to follow the installation guide at https://www.home-assistant.io/installation/raspberrypi. If the problem persists, please create a new issue with exact description of your environment, including hardware configuration and the steps you took to flash the SD card.

mikeplanzer, 2022-09-08:
Thanks for the reply @agners,

- the community thread also references the correct command which unfortunately didn't solve the problem.
- As written in this ticket on, it is an update Problem, the thread you are referencing I wrote "_to make sure I’t not my current configuration/installation_". Meaning just out of curiosity I also tried with a new installation with yes, same problem. Of course I can also create a ticket for that if needed. 
- I followed the installation guide as linked and successfully flashed every version starting from 7.0 going upwards to 8.4 but booting only works up to 7.6

ShonP40, 2022-10-10:
Having the same issue on an RPi 4, I’m currently stuck on 8.2
Home Assistant’s settings page only shows that I can update to 8.5 (not even 9), which doesn’t work as described by other users here.

I’ll try to force update it to 9.2 through SSH (`ha os update --version 9.2`) (already tried that for 8.5), but I don’t think that it’ll work.

ShonP40, 2022-10-10:
Also, is there a way to transfer a data disk to a new OS installation without formatting it?

(I currently have an SD Card to boot the OS, and an SSD for all the data)

ShonP40, 2022-10-17:
> I’ll try to force update it to 9.2 through SSH (`ha os update --version 9.2`) (already tried that for 8.5), but I don’t think that it’ll work.

Didn't work, it still reverted me back to 8.2

agners, 2022-10-17:
> Home Assistant’s settings page only shows that I can update to 8.5 (not even 9), which doesn’t work as described by other users here.

This is expected: The Supervisor offers the last version of the current major release before jumping to the next major release. This is the best tested migration path, so we offer that by default.

Most of the time a direct upgrade (as you forced it through console) works too.

I need logs of the system after the failed attempt to better understand the issue. Do you have SSH debug shell access to that system (port 22222)?

ShonP40, 2022-10-17:
> This is expected: The Supervisor offers the last version of the current major release before jumping to the next major release. This is the best tested migration path, so we offer that by default.

Yea realized this later

> I need logs of the system after the failed attempt to better understand the issue. Do you have SSH debug shell access to that system (port 22222)?

Don't have it enabled at the moment
To what directory are those logs being saved to though?

ShonP40, 2022-11-08:
Also, I don't know if its related to this, but every change I'm doing to my `config.txt` file gets reverted as soon as I restart the RPi

ShonP40, 2022-11-08:
I'll try it in a couple days if there really aren't any other ways to make it update

ShonP40, 2023-01-13:
Finally got some time to mess with it, and found out that the SDCard just locked itself into read-only mode
Which is why it wasn't writing any changes/updates.

Ended up just making the RPi boot off the SATA M.2 SSD I had in it (as the datadisk before) instead

closed 2023-04-20 by github-actions[bot]