home-assistant/operating-system #2049: "request canceled while waiting for connection" - probably issue with MTU size

kuki13, 2022-08-08:
### Describe the issue you are experiencing

Hi, I am unable to upgrade the home assistant to the new version (doesn't matter if it is core/superviser or anything else).

Error:
```
22-08-08 20:06:43 ERROR (SyncWorker_4) [supervisor.docker.interface] Can't install ghcr.io/home-assistant/raspberrypi4-64-homeassistant:2022.8.2: 500 Server Error for http+docker://localhost/v1.41/images/create?tag=2022.8.2&fromImage=ghcr.io%2Fhome-assistant%2Fraspberrypi4-64-homeassistant&platform=linux%2Farm64: Internal Server Error ("Get "https://ghcr.io/v2/": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)")
22-08-08 20:06:43 WARNING (MainThread) [supervisor.homeassistant.core] Updating Home Assistant image failed
```

Most likely the issue is that for some reason Home Assistant `MTU` parameter on the network interface is set to 1500. My internet provider uses `PPPoE` and the max allowed MTU size is `1492`. From my computer, I am able to download HA docker image just fine. But HA that is connected to the same network has issues. 

I am struggling with this error for a very long time (2 years probably). To update HA I often switch to another internet access point for a while (my mobile phone).

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

8.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Fresh install of home assistant to rasbery pi and connect to router with `PPPoE` connection

### Anything in the Supervisor logs that might be useful for us?

```txt
22-08-08 20:06:43 ERROR (SyncWorker_4) [supervisor.docker.interface] Can't install ghcr.io/home-assistant/raspberrypi4-64-homeassistant:2022.8.2: 500 Server Error for http+docker://localhost/v1.41/images/create?tag=2022.8.2&fromImage=ghcr.io%2Fhome-assistant%2Fraspberrypi4-64-homeassistant&platform=linux%2Farm64: Internal Server Error ("Get "https://ghcr.io/v2/": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)")
22-08-08 20:06:43 WARNING (MainThread) [supervisor.homeassistant.core] Updating Home Assistant image failed
```

### Anything in the Host logs that might be useful for us?

```txt
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.15.32-v8 (builder@02acbc8bee81) (aarch64-buildroot-linux-gnu-gcc.br_real (Buildroot -g686000eee) 9.4.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP PREEMPT Sat Jul 23 18:06:19 UTC 2022
[    0.000000] random: fast init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.2
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000007fffffff]
[    0.000000] percpu: Embedded 28 pages/cpu s77272 r8192 d29224 u114688
[    0.000000] pcpu-alloc: s77272 r8192 d29224 u114688 alloc=28*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 496944
[    0.000000] Kernel command line: zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor rootwait systemd.machine_id=b86ef5dd3d6b4fe0a5f11eeced9b04dd cgroup_enable=memory fsck.repair=yes coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:C3:71:03 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=a3ec664e-32ce-4665-95ea-7ae90ce9aa20 rootfstype=squashfs ro rauc.slot=B
[    0.000000] cgroup: Enabling memory control group subsystem
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[... 556 lines cut ...]
[  118.169840] audit: type=1334 audit(1659981514.874:181): prog-id=0 op=UNLOAD
[  118.169872] audit: type=1334 audit(1659981514.874:182): prog-id=0 op=UNLOAD
[  138.142420] audit: type=1701 audit(1659981534.846:183): auid=4294967295 uid=0 gid=0 ses=4294967295 subj==docker-default (enforce) pid=4074 comm="ttyd" exe="/usr/bin/ttyd" sig=11 res=1
[  138.167518] audit: type=1334 audit(1659981534.870:184): prog-id=17 op=LOAD
[  138.168016] audit: type=1334 audit(1659981534.870:185): prog-id=18 op=LOAD
[  138.725928] audit: type=1334 audit(1659981535.430:186): prog-id=0 op=UNLOAD
[  138.725954] audit: type=1334 audit(1659981535.430:187): prog-id=0 op=UNLOAD
[  143.754882] audit: type=1334 audit(1659981540.458:188): prog-id=19 op=LOAD
[  143.755484] audit: type=1334 audit(1659981540.458:189): prog-id=20 op=LOAD
[  144.211636] audit: type=1334 audit(1659981540.914:190): prog-id=21 op=LOAD
[  144.211764] audit: type=1334 audit(1659981540.914:191): prog-id=22 op=LOAD
[  174.334918] audit: type=1334 audit(1659981571.037:192): prog-id=0 op=UNLOAD
[  174.334959] audit: type=1334 audit(1659981571.037:193): prog-id=0 op=UNLOAD
[  174.751004] audit: type=1334 audit(1659981571.453:194): prog-id=0 op=UNLOAD
[  174.751035] audit: type=1334 audit(1659981571.453:195): prog-id=0 op=UNLOAD
[  645.031653] audit: type=1701 audit(1659982041.729:196): auid=4294967295 uid=0 gid=0 ses=4294967295 subj==docker-default (enforce) pid=4104 comm="ttyd" exe="/usr/bin/ttyd" sig=11 res=1
[  645.053064] audit: type=1334 audit(1659982041.749:197): prog-id=23 op=LOAD
[  645.053152] audit: type=1334 audit(1659982041.749:198): prog-id=24 op=LOAD
[  645.534995] audit: type=1334 audit(1659982042.233:199): prog-id=0 op=UNLOAD
[  645.535025] audit: type=1334 audit(1659982042.233:200): prog-id=0 op=UNLOAD
```

### Additional information

I have tried to add `net.tcp_mtu_probing=1` to `sysctl.conf` but file was reverted to the previous value after reboot.

agners, 2022-08-16:
This is unlikely related to MTU size of your provider. Ethernet commonly uses an MTU size of 1500, and your router will just split-up IP packets accordingly (fragments one packet into two smaller packets).

The error `Client.Timeout exceeded while awaiting headers` quite often appears when there are DNS issues. By default the DNS server provided by your DHCP server should be used. Can you try to set a static DNS server?

kuki13, 2022-08-24:
Sorry for the late response.

You were right. It was the DNS issue. Previously when I read about this issue a few people suggested changing DNS by `ha dns options --servers dns://8.8.8.8` option. I have tried that and this didn't help at all. Then I read about solving this issue by changing `MTU` options.

For others with a similar issue: configure a static DNS server by Home Assistant UI.

closed 2022-08-24 by kuki13