home-assistant/operating-system #2052: Unable to update HA Core and addons

Santanachia, 2022-08-10:
### Describe the issue you are experiencing

after upgrading to 8.4, the system began to run very slowly despite the fact that only 5% of the processor and 60% of the memory is used. On top of that I am not able to update either HA Core or addons (the installation process takes hours and nothing changes).
[image]

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

8.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. upgrade from previous version

### Anything in the Supervisor logs that might be useful for us?

```txt
ERROR (MainThread) [supervisor.utils] Can't execute run while a task is in progress
```

### Anything in the Host logs that might be useful for us?

```txt
nothing suspicious
```

### Additional information

I can't do a downgrade, the system runs so slow that ssh constantly disconnects.

Santanachia, 2022-08-10:
p.s.
When trying to install another addon:
```
Can't install homeassistant/aarch64-addon-ssh:9.6.0: 500 Server Error for http+docker://localhost/v1.41/auth: Internal Server Error ("Get "https://registry-1.docker.io/v2/": Get "https://auth.docker.io/token?account=vexator&client_id=docker&offline_token=true&service=registry.docker.io": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers) (Client.Timeout exceeded while awaiting headers)")
```

Santanachia, 2022-08-10:
update via ssh (port 22222) failed
```bash
Welcome to Home Assistant OS.

Use `ha` to access the Home Assistant CLI.
# ha os info
board: rpi3-64
boot: B
data_disk: .
update_available: false
version: "8.4"
version_latest: "8.4"
# ha os update --version=8.3
Error: Can't fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/8.3/haos_rpi3-64-8.3.raucb: Response payload is not completed
#
```

Santanachia, 2022-08-10:
is it possible to download the file manually and update from local file?

agners, 2022-08-10:
This sounds like internet connectivity issues. Can you ping github.com?

What is `ha network info` showing?

Santanachia, 2022-08-10:
```
# ping github.com
github.com is alive!

# ha network info
docker:
  address: 172.30.32.0/23
  dns: 172.30.32.3
  gateway: 172.30.32.1
  interface: hassio
host_internet: true
interfaces:
- connected: false
  enabled: false
  interface: eth0
  ipv4:
    address: []
    gateway: null
    method: disabled
    nameservers: []
  ipv6:
    address: []
    gateway: null
    method: disabled
    nameservers: []
  primary: false
  type: ethernet
  vlan: null
  wifi: null
- connected: true
  enabled: true
  interface: wlan0
  ipv4:
    address:
    - 192.168.8.100/24
    gateway: 192.168.8.1
    method: auto
    nameservers:
    - 192.168.8.1
  ipv6:
    address:
    - fd14:a32f:31a8:3f00:7976:7a2e:b232:ae34/64
    - fe80::d1a9:8e60:7e7d:64b5/64
    gateway: fe80::16a3:2fff:fe31:a83f
    method: auto
    nameservers:
    - fe80::16a3:2fff:fe31:a83f
  primary: true
  type: wireless
  vlan: null
  wifi:
    auth: wpa-psk
    mode: infrastructure
    signal: 54
    ssid: Vex 2.4
supervisor_internet: true
```

Santanachia, 2022-08-11:
@agners 
> This sounds like internet connectivity issues.

You are right. After connecting via LAN, it managed to update, as well as everything works faster. The unsolved problem remains - WiFi worked fine in 8.3, now it doesn't work properly.

agners, 2022-08-12:
There is only very few changes between 8.3 and 8.4, most of them are just bumps of stable versions of software.

https://github.com/home-assistant/operating-system/releases/tag/8.4

To me it seems unlikely a OS problem. Did maybe something in your environment change (new location with bad reception, new devices which interfere, etc?)

At the very least, I need host logs and detailed information of what Wi-Fi adapter you are using.

Also note that we generally recommend using Ethernet since it is much more reliable typically.

Santanachia, 2022-08-17:
The only thing that has changed is the OS. Immediately after the reboot, there were problems with integrations that required internet.

WiFi built into the RPi 3 B+

I know I should be using Ethernet, but I need a USB connection to the device where I don't have a cable.

This isn't the first time an OS update has messed up WiFi. Not so long ago: #1837

ArtBIT, 2022-08-24:
I am experiencing a similar issue. I cannot update Core nor any of the add-ons due to the fact that host does not have access to the internet, which it clearly has.
I tracked it down to the `http://version.home-assistant.io/online.txt` not being accessible, while `https://version.home-assistant.io/online.txt` is.
It seems that if you failed to update your supervisor while the http URL was still accessible, you are locked out of updating due to this change https://github.com/home-assistant/supervisor/commit/bcef34012d6c93b05b547281eba20c152e39e505#diff-b4926e9e651453994da0283d12755b49151fc9f1f71f9a1eb592f85fa60801eeR244 
Is there a way to trick the updater to skip the online check?

closed 2022-11-29 by github-actions[bot]