home-assistant/operating-system #2060: Recommended Bluetooth Dongle not working

thexperiments, 2022-08-13:
### Describe the issue you are experiencing

I bought one of the recommended Bluetooth dongle (ZEXMTE BT-505) to use it with my homeassistant setup.

Sadly it doesn't seem to work. The integration shows:
Retrying setup: Failed to start Bluetooth: adapter 'hci0' not found

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

8.4

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Plugin USB BT stick
2. Forward to VM
3. Observe Problem

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
On the console of hassos I see this information about Bluetooth:
# dmesg | grep -i bluetooth
[    0.374213] Bluetooth: Core ver 2.22
[    0.374622] NET: Registered PF_BLUETOOTH protocol family
[    0.375140] Bluetooth: HCI device and connection manager initialized
[    0.375841] Bluetooth: HCI socket layer initialized
[    0.376140] Bluetooth: L2CAP socket layer initialized
[    0.377144] Bluetooth: SCO socket layer initialized
[    2.270814] usb 2-3: Product: Bluetooth 5.1 Radio
[    4.502208] Bluetooth: hci0: command 0x1001 tx timeout
[   12.886226] Bluetooth: hci0: RTL: HCI_OP_READ_LOCAL_VERSION failed (-110)

# lsusb
Bus 002 Device 004: ID 0bda:a725
Bus 003 Device 001: ID 1d6b:0003
Bus 002 Device 005: ID 0658:0200
Bus 001 Device 001: ID 1d6b:0001
Bus 001 Device 002: ID 0627:0001
Bus 002 Device 003: ID 1cf1:0030
Bus 002 Device 001: ID 1d6b:0002

>> Bus 002 Device 004 should be the Bluetooth dongle
```

agners, 2022-08-13:
> I bought one of the recommended Bluetooth dongle (ZEXMTE BT-505) to use it with my homeassistant setup.

I don't have that particular dongle here, where is it being recommended?

agners, 2022-08-13:
FWIW, it seems the driver is present.

```
Bluetooth: hci0: RTL: HCI_OP_READ_LOCAL_VERSION failed (-110)
```

-110 is a time out. This could also be related to virtualization.

thexperiments, 2022-08-13:
It's recommended here: 
https://www.home-assistant.io/integrations/bluetooth/

Sure could be but the Conbee II and aeotect zwave dongle I have passed through in exactly the same way and these work just fine.

thexperiments, 2022-08-13:
I might try to pass it to an other VM there or test it on an other device. Might be possible that it is somehow not working in general, did not test it before.

rajbala, 2022-08-18:
I bought the other long-range bluetooth dongle (XDO BT802) in that same link that is noted as "known working" with HA. Like your adapter, this one also does not work.

adamfeldman, 2022-08-23:
This ZEXMTE long-range unit is working well for me. Passed-through to HA VM via KVM (Proxmox).

https://smile.amazon.com/dp/B09MZ8715D/

dumbo0001, 2022-08-23:
I had the same problem. It suddenly worked after retesting another day with some reboots.

rajbala, 2022-08-23:
> Is the bluetooth adapter visible to the host OS? What virtualization system?
> 
> This ZEXMTE long-range unit is working well for me. Passed-through to HA VM via KVM (Proxmox).
> 
> https://smile.amazon.com/dp/B09MZ8715D/

It is definitely visible to the OS.  I am not using virtualization.

The OS is Debian Bullseye on a Raspberry Pi 4.

rajbala, 2022-08-23:
> I had the same problem. It suddenly worked after retesting another day with some reboots.

Which bluetooth adapter are you using?  And which host hardware and OS, etc?

I have rebooted many times with no luck.

thexperiments, 2022-08-23:
Ok update, thanks for all the hints here and the general confirmation that it really should work.
I found the time to analyze a bit more and have the solution(for me).

The problem was that the ProxMox host machine had grabbed the bt device and tried to initialize it which interfered with forwarding it to the VM.

`
# unplug the bt stick
# unload the kernel modules on the proxmox host
modprobe -r btusb
modprobe -r btrtl

# blacklist bt usb also in the future
echo "blacklist btusb" > /etc/modprobe.d/bluetooth-blacklist.conf

#insert the bt stick again
`

worked for me so far, did not restart my proxmox yet and not planning to soon as other things are running on it.

thexperiments, 2022-08-23:
> I bought the other long-range bluetooth dongle (XDO BT802) in that same link that is noted as "known working" with HA. Like your adapter, this one also does not work.

there can be many reasons for this. As it is a different stick and a different environment and my problem is resolved in a way that does not fit for your scenario I would suggest you create a new separate issue.

closed 2022-08-25 by thexperiments

FlexxFR, 2022-11-15:
> Ok update, thanks for all the hints here and the general confirmation that it really should work. I found the time to analyze a bit more and have the solution(for me).
>
> The problem was that the [...]

Great solution, worked perfectly :)
Was searching for an answer, and this was it. Thnx!