home-assistant/operating-system #2063: Bluetotth Devices do always get disconnected on Pi4

garv3, 2022-08-15:
### Describe the issue you are experiencing

Whenever I try to control a BT device (here a SwitchBot Curtain), the device get's disconnected in the middle of transaction.
```
2022-08-15 01:24:05.079 DEBUG (MainThread) [bleak.backends.bluezdbus.manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_ED_FF_2A_66_XX_XX): ['org.bluez.Device1', {'ServicesResolved': <dbus_next.signature.Variant ('b', True)>}, []]
2022-08-15 01:24:05.082 DEBUG (MainThread) [bleak.backends.bluezdbus.client] Disconnecting (/org/bluez/hci0/dev_ED_FF_2A_66_XX_XX)
2022-08-15 01:24:07.221 DEBUG (MainThread) [bleak.backends.bluezdbus.manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_ED_FF_2A_66_XX_XX): ['org.bluez.Device1', {'ServicesResolved': <dbus_next.signature.Variant ('b', False)>}, []]
2022-08-15 01:24:07.224 DEBUG (MainThread) [bleak.backends.bluezdbus.manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_ED_FF_2A_66_XX_XX): ['org.bluez.Device1', {'Connected': <dbus_next.signature.Variant ('b', False)>}, []]
2022-08-15 01:24:07.224 DEBUG (MainThread) [bleak.backends.bluezdbus.client] Device disconnected (/org/bluez/hci0/dev_ED_FF_2A_66_XX_XX)
2022-08-15 01:24:07.224 DEBUG (MainThread) [bleak.backends.bluezdbus.client] _cleanup_all(/org/bluez/hci0/dev_ED_FF_2A_66_XX_XX)
2022-08-15 01:24:07.231 ERROR (MainThread) [switchbot.devices.device] Switchbot communication failed. Stopping trying
Traceback (most recent call last):
File "/usr/local/lib/python3.10/site-packages/switchbot/devices/device.py", line 81, in _sendcommand
return await self._send_command_locked(key, command)
File "/usr/local/lib/python3.10/site-packages/switchbot/devices/device.py", line 122, in _send_command_locked
await client.start_notify(read_char, _notification_handler)
File "/usr/local/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 809, in start_notify
raise BleakError(
bleak.exc.BleakError: Characteristic with UUID None could not be found!
```

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

8.4

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Try to control a BT device 

### Anything in the Supervisor logs that might be useful for us?

```txt
-
```

### Anything in the Host logs that might be useful for us?

```txt
-
```

agners, 2022-08-16:
Maybe this gets resolved by #2065 in case the SwitchBot Curtain is a HID or HID over GATT device.

If not, more logs from the BlueZ daemon would be interesting. You can gain access to the underlying OS using SSH on port 22222 (see [this how-to](https://developers.home-assistant.io/docs/operating-system/debugging#ssh-access-to-the-host)).

The following command should work then:
```
journalctl -u bluetooth.service
```

lymanbuttler, 2022-08-28:
This bug also affects the Pi 3. The Bluetooth module becomes unresponsive at the hardware or firmware level after some time, and can only be restored by rebooting the host system.

closed 2023-01-16 by github-actions[bot]