home-assistant/operating-system #2064: Cannot connect to Raspberry PI 4 over Bluetooth

gcampax, 2022-08-15:
### Describe the issue you are experiencing

I tried to set it up a RPi4 running the most recent Home Assistant as a bluetooth speaker.
With some fiddling, I got it to pair to my phone and laptop, but sound does not work.
The logs from the audio container show:
> E: [pulseaudio] bluez5-util.c: org.bluez.Media1.RegisterEndpoint() failed: org.freedesktop.DBus.Error.UnknownMethod: Method "RegisterEndpoint" with signature "oa{sv}" on interface "org.bluez.Media1" doesn't exist

This seems identical to https://github.com/Arkq/bluez-alsa/issues/542 which suggests it is a build configuration problem: bluez (bluetoothd) needs to be built with --enable-a2dp --enable-avrcp

### What operating system image do you use?

rpi4  (Raspberry Pi 4/400 32-bit OS)

### What version of Home Assistant Operating System is installed?

8.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Enable Bluetooth. This popped up in the UI and I confirmed.
2. Pair a phone or laptop with Bluetooth. I did it by hand by calling DBus methods using the web terminal (I wish there was a UI for it)
3. Connect to Home Assistant from the phone or laptop.

### Anything in the Supervisor logs that might be useful for us?

```txt
See above
```

### Anything in the Host logs that might be useful for us?

```txt
See above
```

agners, 2022-08-16:
Indeed, it seems that BlueZ is by default not compiled with this options in Buildroot. There is the Buildroot `BR2_PACKAGE_BLUEZ5_UTILS_PLUGINS_AUDIO` option, I'll enable it in OS 9.

agners, 2022-08-17:
Can you try the latest development build `9.0.dev20220816`? It should have the options configured correctly.

gcampax, 2022-08-17:
Happy to try it! How do I do that? I tried:
```
ha os update --version 9.0.dev20220816
```
and I got
> Error: Error raised from OTA Webserver: 404

gcampax, 2022-08-17:
Ok, testing this build I'm making progress, in that I'm able to connect from my phone in HSP/HFP mode, but actual audio playing fails.
I'm seeing these errors in the logs:
> E: [pulseaudio] backend-native.c: socket(SEQPACKET, SCO) Address family not supported by protocol
> E: [pulseaudio] backend-native.c: accept(): Not a socket

Probably some more options to enable in the kernel to have bluetooth audio support?

(A2DP audio support would be even better but that seems to fail for reasons I don't quite understand)

closed 2022-11-23 by github-actions[bot]