home-assistant/operating-system #2076: System Corrupt after updating from 8.4 to HAOS 8.5

bschatzow, 2022-08-23:
### Describe the issue you are experiencing

Lots of errors in the log, system unstable.

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

8.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Run 8.4
2. Update to 8.5
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
22-08-23 06:05:17 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_protocol.py", line 435, in _handle_request
    resp = await request_handler(request)
  File "/usr/local/lib/python3.9/site-packages/sentry_sdk/integrations/aiohttp.py", line 121, in sentry_app_handle
    reraise(*_capture_exception(hub))
  File "/usr/local/lib/python3.9/site-packages/sentry_sdk/_compat.py", line 54, in reraise
    raise value
  File "/usr/local/lib/python3.9/site-packages/sentry_sdk/integrations/aiohttp.py", line 111, in sentry_app_handle
    response = await old_handle(self, request)
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 138, in system_validation
    return await handler(request)
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 204, in token_validation
    return await handler(request)
  File "/usr/src/supervisor/supervisor/api/utils.py", line 60, in wrap_api
    answer = await method(api, *args, **kwargs)
  File "/usr/src/supervisor/supervisor/api/store.py", line 177, in store_info
    ATTR_ADDONS: [
  File "/usr/src/supervisor/supervisor/api/store.py", line 178, in <listcomp>
    self._generate_addon_information(self.sys_addons.store[addon])
  File "/usr/src/supervisor/supervisor/api/store.py", line 113, in _generate_addon_information
    ATTR_ADVANCED: addon.advanced,
  File "/usr/src/supervisor/supervisor/addons/model.py", line 222, in advanced
    return self.data[ATTR_ADVANCED]
  File "/usr/src/supervisor/supervisor/store/addon.py", line 19, in data
    return self.sys_store.data.addons[self.slug]
KeyError: 'df843657_spotify'
Lots of similar errors to:

2022-08-23 06:10:46.872 WARNING (MainThread) [homeassistant.components.binary_sensor] Setup of binary_sensor platform synology_dsm is taking over 10 seconds.

And this:
22-08-23 06:05:58 ERROR (MainThread) [supervisor.homeassistant.api] Error on call https://172.30.32.1:8123/api/config:
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

### System Health information

Can't get this

### Additional information

System is now unstable.  Can't get more info.

ronjtaylor, 2022-08-24:
I am running an Intel NUC i3 with 8GB RAM and a WD Green SSD m2.2280. All was running fine on 8.4 but when I updated to the 8.5 (from settings -> updates) it boots bot goes into a loop when it is trying to load docker.  I am looking at it on a console monitor connected to the HDMI port but the display is too fast and it reboots/crashes before I can see it.
How do I fix this?
Thank you

ocalvo, 2022-08-24:
> I am running an Intel NUC i3 with 8GB RAM and a WD Green SSD m2.2280. All was running fine on 8.4 but when I updated to the 8.5 (from settings -> updates) it boots but goes into a boot loop when it is trying to load docker. I am looking at it on a console monitor connected to the HDMI port but the display is too fast and it reboots/crashes before I can see it. How do I fix this? Thank you

If you have a backup, you could try to repave the boot drive (WD Green SSD m2.2280) with version 8.5 and then restore the backup.

ronjtaylor, 2022-08-24:
Yes I backed up HA ( full backup) at Midnight last night with an automation and luckily copied the backup to another device (a USB SSD).
Just to be clear I am assuming that by "repave" you mean download the image and rebuild then load backup?
Could you please clarify so I don't stuff it.
Thank you

ocalvo, 2022-08-24:
> Yes I backed up HA ( full backup) at Midnight last night with an automation and luckily copied the backup to another device (a USB SSD).
> Just to be clear I am assuming that by "repave" you mean download the image and rebuild then load backup?
> Could you please clarify so I don't stuff it.
> Thank you

Yes,  by repave I mean download the OS image and flash it to the disk.

ronjtaylor, 2022-08-24:
When I look to download the image I can only see 8.4 in the doco https://www.home-assistant.io/installation/generic-x86-64#write-the-image-to-your-boot-media
This is the URL for blancher
https://github.com/home-assistant/operating-system/releases/download/8.4/haos_generic-x86-64-8.4.img.xz

ronjtaylor, 2022-08-24:
I rolled back to 8.4

Mask3007, 2022-08-24:
Updated yesterday from 8.4 to 8.5 unfortunately having only a Backup 11 days old.

But since 8.5 having similar issues mainly on Python lib and connection components.

Therefore most integrations with connections (tuya, fritzbox, xiaomi, Alexa) and even a sensor with a curl command getting timeouts even in the internal network.

agners, 2022-08-24:
@ronjtaylor your problem is unlikely related to the original posters issue, as yours is a different platform. FWIW, on my end I successfully upgraded a Gen11 Intel NUC to 8.5. If you continue to have that problem on upgrade, please open a new issue.

The documentation is now referencing the 8.5 image.

ronjtaylor, 2022-08-24:
My NUC is a NUC8i3BEH which is an 8th generation.  The 8.4 OS works fine but the 8.5 did not.  That to me seems that there is something different about the 8.5.
One of the reasons I went with the HA OS was to try and avoid the more complex issues of Linux and the documentation stated that this version was the one recommended for those of us less skilled.
I decided on the NUC as a platform because it had M2.2280 SSD and it had better USB ports to run ZWave and Zigbee.
After that I am reliant on the expertise of the HA team.
The statistic that 16K people updated needs qualifying by platform success or failure?

niioannou, 2022-09-03:
I have installed version 8.4 on a Raspberry Pi 2 and everything works fine. The problem is that the upgrade to 8.5 keeps failing.
What am I suppose to do ?
Note that I do not like to reinstall the image from scratch.

Thank you

closed 2022-12-09 by github-actions[bot]