home-assistant/operating-system #2077: Home Assistant do not start after upgrade from 8.4 to 8.5

cesargtz1, 2022-08-24:
### Describe the issue you are experiencing

I have a Home Assitant running in a Virtual Machine ( VirtualBox ) which is running over Windows 11, I tried upgrade home assistant from 8.4 to 8.5 when I got notification from new release, however when system is booting to start with new release is stuck waiting for Docker application Container Engine.

`[  ***  ] A start job is running for Docker Application Container Engine ( Xmin Xs / no limit )`

[image]

[image]

I have wait until 10 hours and this process never ends

Initially I was suspecting an incorrect configuration or conflict with a bad syntax related to any file that I could modify in my already configured Home Assitant, but I tested by removing my existant VM and created a new VM then without touching any configuration or install any add on, just from fresh HA sitanllation I tried to upgrade directly and I got same result.

does any one is having similar Issues for HA running in virtual machines?

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

8.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Install Home Assistant Operating system 8.4 in a VM ( Virtualbox )
2. Finish the onboarding
3. Upgrade the Operating system from 8.4 to 8.5

### Anything in the Supervisor logs that might be useful for us?

```txt
No logs collected as OS is not starting
```

### Anything in the Host logs that might be useful for us?

```txt
screen captures only
```

agners, 2022-08-24:
This sounds somewhat similar to #2048 and #1705. What VirtualBox version are you using?

sygys, 2022-08-25:
Yet again an update that adds absolutely nothing and breaks everything. When are you guys planning to stop releasing meaningless updates that only seem to break shit.

agners, 2022-08-25:
@sygys why are you saying that? Did your update break too?

> and breaks everything

That seems wrong. According to https://analytics.home-assistant.io/ 21'278 installations are on 8.5. Yet we have just a couple of issues here.

cesargtz1, 2022-08-25:
I

> This sounds somewhat similar to #2048 and #1705. What VirtualBox version are you using?

based on the comments for  #1705 i remember that I increased the number of CPUs before the issue and upgrade, also when I created the new VM I selected 3 CPUs instead of 1 ( default ), I'll do some tests using single CPU

macflyhub, 2022-08-25:
I am on 8.5 and did a reboot and that was that. Stuck with broken HA. Waited ages. Tried 'banner', 'supervisor repair'. 'restart' etc.. nothing. Shutdown. Changed CPU cores from 2 to 3 and then things got worse. Switched back and then somehow it worked. Now my deCONZ is broken.

cosmoxeon, 2022-08-28:
Tried to update and it went into a perpetual restart loop. I am running HA on a NUC. Had to restore from backup

Tea-NT, 2022-10-12:
9.2 release a few days ago, use VM Work Station work fine. how about backup low version data and restore in last version? is that work?

marti6553, 2022-11-08:
I have the same problem on version 9.3

marti6553, 2022-11-08:
I had the same problem on version 9.3 in virtualbox I changed 4vCPU to 2vCPU and it booted up

cesargtz1, 2022-12-21:
> I had the same problem on version 9.3 in virtualbox I changed 4vCPU to 2vCPU and it booted up

correct the issue is due to I have increased the CPUs, I leave it in 2 CPUs only ( as default ) , the issue fixed. but I had to recreate again the VM in virtualbox.

closed 2022-12-21 by cesargtz1