home-assistant/operating-system #2081: ASIX AX88179 USB 3.0 Gigabit Ethernet Disconnects

jsiemonski, 2022-08-24:
### Describe the issue you are experiencing

I have an ASIX AX88179 USB 3.0 Gigabit Ethernet adapter on my generic x86-64 system. Since installing OS 8.5, it has been unstable. Disconnecting the USB side and reconnecting it brings HA back, but this is problematic. It appears to have been affected in the Linux kernel 5.15 update.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

8.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Just letting it run causes it. 

### Anything in the Supervisor logs that might be useful for us?

```txt
22-08-24 11:14:23 WARNING (MainThread) [supervisor.addons.options] Option 'RTSP_READTIMEOUT' does not exist in the schema for Docker Wyze Bridge (af71aba8_docker-wyze-bridge)
22-08-24 11:19:25 WARNING (MainThread) [supervisor.addons.options] Option 'RTSP_READTIMEOUT' does not exist in the schema for Docker Wyze Bridge (af71aba8_docker-wyze-bridge)
22-08-24 11:20:38 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/001/003 - None
22-08-24 11:20:42 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bus/usb/001/005 - None
22-08-24 11:24:27 WARNING (MainThread) [supervisor.addons.options] Option 'RTSP_READTIMEOUT' does not exist in the schema for Docker Wyze Bridge (af71aba8_docker-wyze-bridge)
22-08-24 11:29:29 WARNING (MainThread) [supervisor.addons.options] Option 'RTSP_READTIMEOUT' does not exist in the schema for Docker Wyze Bridge (af71aba8_docker-wyze-bridge)
22-08-24 11:30:11 WARNING (MainThread) [supervisor.addons.options] Option 'RTSP_READTIMEOUT' does not exist in the schema for Docker Wyze Bridge (af71aba8_docker-wyze-bridge)
22-08-24 11:30:13 INFO (SyncWorker_7) [supervisor.docker.interface] Stopping addon_af71aba8_docker-wyze-bridge application
22-08-24 11:30:15 INFO (SyncWorker_7) [supervisor.docker.interface] Cleaning addon_af71aba8_docker-wyze-bridge application
22-08-24 11:30:15 WARNING (MainThread) [supervisor.addons.options] Option 'RTSP_READTIMEOUT' does not exist in the schema for Docker Wyze Bridge (af71aba8_docker-wyze-bridge)
22-08-24 11:30:25 INFO (SyncWorker_0) [supervisor.docker.interface] Stopping addon_core_configurator application
22-08-24 11:30:29 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning addon_core_configurator application
22-08-24 11:30:38 INFO (SyncWorker_3) [supervisor.docker.interface] Stopping addon_a0d7b954_vscode application
22-08-24 11:30:42 INFO (SyncWorker_3) [supervisor.docker.interface] Cleaning addon_a0d7b954_vscode application
22-08-24 11:32:01 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup
22-08-24 11:32:01 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup
22-08-24 11:33:07 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
22-08-24 11:33:07 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_IPV6_ERROR/ContextType.DNS_SERVER
22-08-24 11:33:07 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_FAILED/ContextType.DNS_SERVER
22-08-24 11:33:07 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR
22-08-24 11:33:07 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
22-08-24 11:33:07 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
22-08-24 11:33:07 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
22-08-24 11:33:07 INFO (MainThread) [supervisor.resolution.check] System checks complete
22-08-24 11:33:07 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
22-08-24 11:33:07 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-08-24 11:33:07 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
22-08-24 11:33:07 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
22-08-24 11:33:22 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/mrlt8/docker-wyze-bridge/ repository
22-08-24 11:33:22 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository
22-08-24 11:33:22 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/home-assistant-addon repository
22-08-24 11:33:22 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
22-08-24 11:33:22 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/alexbelgium/hassio-addons repository
22-08-24 11:33:22 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/sabeechen/hassio-google-drive-backup repository
22-08-24 11:33:24 INFO (MainThread) [supervisor.store] Loading add-ons from store: 135 all - 0 new - 0 remove
22-08-24 11:33:24 INFO (MainThread) [supervisor.store] Loading add-ons from store: 135 all - 0 new - 0 remove
22-08-24 11:33:25 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
22-08-24 12:03:25 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
22-08-24 12:21:15 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/001/005 - None
22-08-24 12:21:22 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bus/usb/001/006 - None
22-08-24 12:32:02 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup
22-08-24 12:32:03 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup
22-08-24 12:33:07 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
22-08-24 12:33:07 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_IPV6_ERROR/ContextType.DNS_SERVER
22-08-24 12:33:07 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_FAILED/ContextType.DNS_SERVER
22-08-24 12:33:07 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR
22-08-24 12:33:07 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
[... 135 lines cut ...]
[13744.221210] ax88179_178a 1-5:1.0 enp0s21f0u5: unregister 'ax88179_178a' usb-0000:00:15.0-5, ASIX AX88179 USB 3.0 Gigabit Ethernet
[13744.221381] ax88179_178a 1-5:1.0 enp0s21f0u5: Failed to read reg index 0x0002: -19
[13744.221387] ax88179_178a 1-5:1.0 enp0s21f0u5: Failed to write reg index 0x0002: -19
[13744.228821] ax88179_178a 1-5:1.0 enp0s21f0u5 (unregistered): Failed to write reg index 0x0002: -19
[13744.228829] ax88179_178a 1-5:1.0 enp0s21f0u5 (unregistered): Failed to write reg index 0x0001: -19
[13744.228832] ax88179_178a 1-5:1.0 enp0s21f0u5 (unregistered): Failed to write reg index 0x0002: -19
[13748.812814] usb 1-5: new high-speed USB device number 6 using xhci_hcd
[13748.944674] usb 1-5: New USB device found, idVendor=0b95, idProduct=1790, bcdDevice= 1.00
[13748.944681] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[13748.944683] usb 1-5: Product: AX88179
[13748.944685] usb 1-5: Manufacturer: ASIX Elec. Corp.
[13748.944687] usb 1-5: SerialNumber: 000050B6F334E8
[13749.276517] ax88179_178a 1-5:1.0 eth0: register 'ax88179_178a' at usb-0000:00:15.0-5, ASIX AX88179 USB 3.0 Gigabit Ethernet, 00:50:b6:f3:34:e8
[13749.283820] ax88179_178a 1-5:1.0 enp0s21f0u5: renamed from eth0
[13752.150393] ax88179_178a 1-5:1.0 enp0s21f0u5: ax88179 - Link status is: 1
[13752.156744] IPv6: ADDRCONF(NETDEV_CHANGE): enp0s21f0u5: link becomes ready
[15319.605758] audit: type=1334 audit(1661359650.985:332): prog-id=19 op=LOAD
[15319.605867] audit: type=1334 audit(1661359650.985:333): prog-id=20 op=LOAD
[15319.857990] audit: type=1334 audit(1661359651.237:334): prog-id=21 op=LOAD
[15319.858036] audit: type=1334 audit(1661359651.237:335): prog-id=22 op=LOAD
[15349.922815] audit: type=1334 audit(1661359681.302:336): prog-id=0 op=UNLOAD
[15349.922823] audit: type=1334 audit(1661359681.302:337): prog-id=0 op=UNLOAD
[15350.099852] audit: type=1334 audit(1661359681.479:338): prog-id=0 op=UNLOAD
[15350.099860] audit: type=1334 audit(1661359681.479:339): prog-id=0 op=UNLOAD
```

agners, 2022-08-24:
There are two commits mentioning this Ethernet adapter in the 5.15.61 kernel change log: https://cdn.kernel.org/pub/linux/kernel/v5.x/ChangeLog-5.15.61.

HAOS 8.5 uses 5.15.60.

Best is you revert to 8.4 until 8.6 is out.

agners, 2022-09-13:
HAOS 9.0 will ship with 5.15.67.

closed 2022-09-13 by agners