home-assistant/operating-system #2085: NUC8i3BEH (8th Gen) boot loop after upgrading to 8.5

agners, 2022-08-25:
My NUC is a NUC8i3BEH which is an 8th generation.  The 8.4 OS works fine but the 8.5 did not.  That to me seems that there is something different about the 8.5.
One of the reasons I went with the HA OS was to try and avoid the more complex issues of Linux and the documentation stated that this version was the one recommended for those of us less skilled.
I decided on the NUC as a platform because it had M2.2280 SSD and it had better USB ports to run ZWave and Zigbee.
After that I am reliant on the expertise of the HA team.
The statistic that 16K people updated needs qualifying by platform success or failure?

_Originally posted by @ronjtaylor in https://github.com/home-assistant/operating-system/issues/2076#issuecomment-1226558689_

agners, 2022-08-25:
> My NUC is a NUC8i3BEH which is an 8th generation. The 8.4 OS works fine but the 8.5 did not. That to me seems that there is something different about the 8.5.

You can find the changelog here: https://github.com/home-assistant/operating-system/releases/tag/8.5

One thing which did change was the upstream stable kernel update to 5.15.60.

If Docker doesn't start correctly, the system should automatically revert back to the previous installation (the GRUB2 bootloader should choose the other "Slot" at startup). Did that not happen in your case?

You can also manually choose the other boot slot via up/down arrows when the GRUB2 bootloader appears.

If both boot slot don't boot, then this means that also OS 8.4 stopped working. Since OS 8.4 worked before, it would mean that something caused data corruption on the data partition.

ronjtaylor, 2022-08-25:
Thanks for your reply and input

As I said I selected this OS as it is supposed to be for less skilled people.  Reading the Change log for me is like reading another language.

No, the system did not revert back to the previous installation by itself.  I don't know what GRUB2 bootloader should choose or not choose or what I should manually choose as this knowledge needs to be up front and patent.  I did not know this and nothing was advised when I hit the install in settings.  Maybe that should be advised or maybe it is and I did not know here to look
Sorry but this is an issue I have with HA and the dissemination of knowledge.  I am 70 going on 71 so not as sharp as I used to be but I try.

agners, 2022-08-25:
I understand that this is a bit technical, normally this all should not be necessary, but since you encounter a problem, manual intervention is necessary.

[image]

At boot up you should see this screen, can you try both entries once?

ronjtaylor, 2022-08-25:
Thank you for the knowledge.

Well I can't do that at the moment because to get my house back on line I rebuilt my system by flashing the M2.2280 SSD with 8.4 and reloaded my HA backup from the night before (Midnight backup automation) so I guess that unless I run the update again the previous bootloader would not be there?
I have another personally issue as I have to fly my wife to Brisbane (we live in Cairns Australia) for surgery early in September so I have resolved to not do any updates until I get her home from the surgery as I want HA to be doing it's thing while we are away.  I hope you understand.

Time permitting, I am willing to try this when we get back but I will also be her carer then as well.

My NUC is powered by a UPS 1300W that also powers my Essential devices like Modem/Router, ether switch etc so power should be nice and clean I also run NUT integration that parents of my Synology NAS.

However I have learned bit about GRUB2 and it sounds like it may have solved my problem so long as there was no data corruption.

Thank you

ronjtaylor, 2022-08-25:
Upon reflection I recall that I do remember seeing this screen but it was really fast and I was unable to read what it said.
Is this normal?  Could it be delayed to allow time to read?

agners, 2022-08-26:
> Well I can't do that at the moment because to get my house back on line I rebuilt my system by flashing the M2.2280 SSD with 8.4 and reloaded my HA backup from the night before (Midnight backup automation) so I guess that unless I run the update again the previous bootloader would not be there?

Yeah in that case the old system is gone. Reflashing wipes the boot loader and old Operating System installations.

> [...] or surgery early in September so I have resolved to not do any updates until I get her home from the surgery as I want HA to be doing it's thing while we are away. I hope you understand.

Sure, good luck with your surgery.

> However I have learned bit about GRUB2 and it sounds like it may have solved my problem so long as there was no data corruption.

Maybe, yes. But it should actually automatically switch to the old version. Do you remember if you tried booting the system 3 times before re-flashing? The system only switches back to the old system after 3 retries...

ronjtaylor, 2022-08-26:
_Do you remember if you tried booting the system 3 times before re-flashing? The system only switches back to the old system after 3 retries..._

Yes I do and I can't be sure but I think is was actually more than 3 times that it got to the loading docker stage and then started to reboot.

Someone suggested I REPAVE with 8.5 but when I went to the Install doco, the URL still pointed to 8.4 so that was why I flashed with 8.4.  I now see that the doco refers to 8.5

My plan is that when we get back from my wife's surgery and the dust settles, I will try the updates again and see what happens. I can update this issue the if you wish?

One of the frustrating thing about using the Statistics side of HA is that when these things happen you ruin your stats.  Not earth shattering but an an old data annalist is bugs me.

Thanks for your help

HawkanForce, 2022-08-27:
I have similar problem with NUC 8i5BEK (16gb + 128gb m.2 ssd)
if there is a power outage the homeassistant gets stuck in a boot loop. I have only used HA since 3/2022 but it has always behaved this way. Today I tried to update to the new 8.5 version and it was again stuck in the boot loop. I recorded a video last time
when it boots it waits at this line first for a while
[image]
but then the rows go by so fast that I cannot see them
But seems that it says Failed to start docker application
[image]
video: https://www.youtube.com/watch?v=tSvnzXJW4Mc&t=1s

ronjtaylor, 2022-08-27:
The video shows exactly what my NUC did when I tried to update to 8.5

I have never experienced this issue with a power outage mainly because I have my NUC on a UPS feed, with about 30 minutes autonomy time.  I live in an area where cyclones cause power outages quite regularly during the summer cyclone season so the UPS is a vital part of my kit as is a 4.5KW Inverter Generator. I also run the NUT Integration which shuts down the NUC when the UPS signals it's running low on battery.

cosmoxeon, 2022-08-28:
Exactly the same issue happened with my install on a NUC. Gave up trying to decipher the logs as it rebooted just too fast. Restored a day old snapshot instead.

agners, 2022-08-30:
There is a way to access the logs from console by editing the boot command in GRUB2, I wrote a rough how-to here: https://github.com/home-assistant/operating-system/issues/2092#issuecomment-1231572414. 

@HawkanForce or @cosmoxeon, would be cool if you can give that a try on an affected system.

closed 2022-12-05 by github-actions[bot]