home-assistant/operating-system #2086: upgrade to 8.5 breaks usb-Texas_Instruments_TUSB3410 working

IGastel, 2022-08-25:
### Describe the issue you are experiencing

after upgrade from 8.4 to 8.5 my USB reviever (honeywell HGI80 which uses a TI USB3410 chip) is no longer showing in the hardware list. i have the following error in the log:
Failed to open /dev/serial/by-id/usb-Texas_Instruments_TUSB3410_Boot_Device_TUSB3410-if00-port0 (config: {'rtscts': False, 'baudrate': 115200, 'dsrdtr': False, 'xonxoff': True, 'timeout': 0}): [Errno 2] could not open port /dev/serial/by-id/usb-Texas_Instruments_TUSB3410_Boot_Device_TUSB3410-if00-port0: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-Texas_Instruments_TUSB3410_Boot_Device_TUSB3410-if00-port0'

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

8.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

upgraded via UI from 8.4 to 8.5

### Anything in the Supervisor logs that might be useful for us?

```txt
no errors
```

### Anything in the Host logs that might be useful for us?

```txt
Full logs
[    0.000000] Linux version 5.15.60 (builder@061dba7915c8) (x86_64-buildroot-linux-gnu-gcc.br_real (Buildroot -g04ac0c825) 9.4.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP Tue Aug 16 12:26:36 UTC 2022
[    0.000000] Command line: BOOT_IMAGE=(hd0,gpt2)/bzImage root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootwait zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor systemd.machine_id=9b9b238f0f424d6b8a87bb9d70100408 cgroup_enable=memory fsck.repair=yes console=ttyS0 console=tty1 rauc.slot=A
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] signal: max sigframe size: 1440
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000000805fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000806000-0x0000000000807fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000808000-0x000000000080ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000810000-0x00000000008fffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000900000-0x00000000beed7fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000beed8000-0x00000000befd9fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000befda000-0x00000000bf8edfff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bf8ee000-0x00000000bf9edfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bf9ee000-0x00000000bfaedfff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000bfaee000-0x00000000bfb6dfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bfb6e000-0x00000000bfb7dfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000bfb7e000-0x00000000bfbfdfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000bfbfe000-0x00000000bfeebfff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bfeec000-0x00000000bff6ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bff70000-0x00000000bfffffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000013fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.70 by EDK II
[    0.000000] efi: SMBIOS=0xbf922000 ACPI=0xbfb7d000 ACPI 2.0=0xbfb7d014 MEMATTR=0xbe351018 
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: QEMU Standard PC (i440FX + PIIX, 1996), BIOS 0.0.0 02/06/2015
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000000] kvm-clock: cpu 0, msr 129dbd001, primary cpu clock
[    0.000002] kvm-clock: using sched offset of 2375746258346213 cycles
[    0.000008] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000012] tsc: Detected 3292.372 MHz processor
[    0.000094] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000096] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000102] last_pfn = 0x140000 max_arch_pfn = 0x400000000
[    0.000136] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- UC  
[    0.000145] last_pfn = 0xbfeec max_arch_pfn = 0x400000000
[    0.013260] Secure boot could not be determined
[... 738 lines cut ...]
dashboards | 2
-- | --
resources | 3
views | 4
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | August 18, 2022 at 19:25
-- | --
current_recorder_run | August 24, 2022 at 22:07
estimated_db_size | 312.97 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

agners, 2022-08-25:
```
[    1.888321] usb 3-1: New USB device found, idVendor=10ac, idProduct=0102, bcdDevice= 1.01
[    1.889626] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.890734] usb 3-1: Product: TUSB3410 Boot Device
[    1.891643] usb 3-1: Manufacturer: Texas Instruments
[    1.892596] usb 3-1: SerialNumber: TUSB3410        
[    1.895562] usb 3-1: can't set config #1, error -32
```

Error -32 means "broken pipe", which often points to a USB hardware problem.

Can you try revert back to HAOS 8.4 to see if that fixes your poblem indeed? E.g. by selecting the other Boot slot at boot up in GRUB2, or by using:

```
ha os update --version 8.4
```

IGastel, 2022-08-25:
I downgraded, but my problem remained. strange coincidence that my USB device breaks at the moment i upgrade.

agners, 2022-08-25:
Thanks for confirming. Maybe a complete system reboot helps? I'd suggest to try different USB port/different cables as well.

lakofka, 2022-08-25:
@IGastel  When I upgraded to 8.4, Zigbee2MQTT wouldn't start because it couldn't find the coordinator USB stick, which wasn't identified by device ID. Changing it so it _was_ identified by device ID fixed it. I think this is perhaps the opposite of your problem but it suggests that something in the handling of USB devices could have changed. 
Don't know, I'm new here.

IGastel, 2022-08-25:
@lakofka i have 4 USB devices all identified by their device ID. i did this to prevent a reboot would giving my USB devices a new port number. only 1 device is not recognized anymore after reboot.

lakofka, 2022-08-25:
@IGastel sounds like your problem is different to what mine was, then. I wish I could be some help, good luck, I hope you find a solution.

IGastel, 2022-08-25:
After multiple restarts of the virtual machine, the USB device appeared again. appologies for waisting time.

closed 2022-08-25 by IGastel