home-assistant/operating-system #2087: HAOS won't boot without display connected

cvroque, 2022-08-26:
### Describe the issue you are experiencing

This one is a bit odd. I'm using the [MSI Cubi](https://www.msi.com/Desktop/Cubi/Specification) as my HA server. It used to run Windows for a few years, then I went with a supervised installation and a few weeks ago I migrated over to HAOS.

Now I'm having one issue where it won't recover from a power loss or SO update, when I connect a display to check and reboot it starts up normally. After some testing I noticed it can't boot HAOS without a display connected, I tried with my Debian Supervised installation and it manages to boot normally without any display connected.

I checked and made sure I'm on the latest BIOS and the problem persists. I'm aware that this might be related to my particular hardware since I found similar issues [¹](https://forum-en.msi.com/index.php?threads/headless-cubi.269556/) [²](https://www.reddit.com/r/MiniPCs/comments/paq9ky/do_you_have_an_msi_cubi_will_it_boot_headless/) online, but it works normally under Debian, Ubuntu and Windows.

I don't know much about Linux so I opened this issue in order to get help, it's important for my HA server to recover by itself after a power loss. Thanks in advance

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

8.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Disconnect any attached displays
2. Perform a Host Shutdown
3. Try to turn Hust on
4. Nothing happens
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Attached.
```

### Anything in the Host logs that might be useful for us?

```txt
Attached.
```

### System Health information

## System Information

version | core-2022.8.7
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.60
arch | x86_64
timezone | America/Sao_Paulo
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4782
Installed Version | 1.26.2
Stage | running
Available Repositories | 1101
Downloaded Repositories | 33

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 3 May 2023 at 21:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | us-east-1-1.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.5
-- | --
update_channel | stable
supervisor_version | supervisor-2022.08.3
agent_version | 1.2.1
docker_version | 20.10.14
disk_total | 116.7 GB
disk_used | 51.9 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | AdGuard Home (4.7.3), File editor (5.3.3), Frigate NVR (3.1), Home Assistant Google Drive Backup (0.108.4), Mosquitto broker (6.1.2), Samba share (10.0.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 15
views | 18
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 13 July 2022 at 21:23
-- | --
current_recorder_run | 26 August 2022 at 16:27
estimated_db_size | 1350.53 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

<details><summary>Sonoff</summary>

version | 3.1.0 (3871c0b)
-- | --
cloud_online | 2 / 2
local_online | 2 / 2

</details>

<details><summary>Xiaomi Gateway 3</summary>

version | 2.1.1 (b23a66a)
-- | --

</details>

### Additional information

[text_45.txt](https://github.com/home-assistant/operating-system/files/9436050/text_45.txt)
[text_46.txt](https://github.com/home-assistant/operating-system/files/9436051/text_46.txt)

agners, 2022-08-30:
Hm, this is kinda annoying to debug.

It would be interesting to know if the logs have any hint. Can you start without display, wait for a few minutes, turn off the system, then start with display to access the logs using:

```
journalctl -b -1
```

The `journalctl` command is only available on the local console or through host OS SSH access (see https://developers.home-assistant.io/docs/operating-system/debugging#ssh-access-to-the-host).

cvroque, 2022-08-30:
Yeah, this one might be annoying. 

I followed your instructions on local console but the arg -1 ended up being my last boot up from 4 days ago and 0 was the boot with display connected, it didn't even get to register the "boot" without display.

cvroque, 2022-09-05:
This weekend while I was in another city my house had a short power outage and my system went down and failed to recover.

After I came back I tried this:
`journalctl --list-boots`

But it only showed the boots from my previous test 5 days ago and the one tonight after I connected the display. After that I messed with pretty much every setting in the BIOS without any luck, so I decided to purchase an HDMI dummy.

Should we still try something else in order to fix this for other users or it's recommended to close the issue? Thanks again!

agners, 2022-09-05:
It seems that Mint Linux distribution had the same problem: https://www.reddit.com/r/MiniPCs/comments/paq9ky/do_you_have_an_msi_cubi_will_it_boot_headless/

Maybe it can be resolved with a particular kernel configuration, but its essentially impossible to track this down without having access to that device. If that work around works for you, I'd say we close it.

Btw, it might be that a future OS version (especially with a major kernel upgrade, scheduled with HAOS 10) fixes this. So maybe worth retrying then...

Martes-Wigglesworth, 2023-08-17:
Greetings everyone. 

I have just encountered this issue and want to push a PR for this issue which should resolve this with a grub default operand update.  

@agners Can this bug be re-openned so I may utilize this item number for my feature branch, or is there another similar issue which I may reference?

Martes-Wigglesworth, 2023-08-26:
Upon further investigation, it appears that my case may be caused by an older bios which seems to require a vga input to be physically attached to fully post. I will update my PR comments and see what @agners would like to do with my submitted PR.

closed 2023-12-03 by github-actions[bot]

JaCh0o, 2024-01-26:
Hello,
Got current version on my rpi5 and same issue, whenever display not connected I stucks without boot, when I connect HDMI monitor it boots properly. Any resolve?

jqfernandez, 2024-01-27:
> Hello, Got current version on my rpi5 and same issue, whenever display not connected I stucks without boot, when I connect HDMI monitor it boots properly. Any resolve?

For me, this problem happens when HAOS is only using `wlan0` wifi interface and no monitor. If I use `eth0` with an ethernet cable and no monitor it boots fine. This is not a solution but its a workaround I can live with.

JaCh0o, 2024-01-27:
To be honest that doesn't work in my case. Its always connected thru Ethernet and still it just don't boot if I won't connect monitor? Can anyone help with that bug?

seanthewebber, 2024-06-09:
Apologies for interjecting into a stale discussion, but I'd like to confirm that the issue isn't limited to the `wlan0` interface. My `Generic x86-64` system, running `core-2024.5.5`, is also affected. It has an Ethernet connection to `enp2s0`, and it fails to boot if no display is connected.

The root cause appears to be missing `GRUB_CMDLINE_LINUX` and `GRUB_CMDLINE_LINUX_DEFAULT` options in the `/etc/grub.d/40_custom` file of the Home Assistant Operating System (HAOS):

```
GRUB_CMDLINE_LINUX="nomodeset consoleblank=0"
GRUB_TIMEOUT=0
```

However, modifying `/etc/grub.d/40_custom` isn't straightforward because the file system in HAOS is read-only. Does anyone know how to make the file system writable, or should I submit a pull request to the main repository?

Martes-Wigglesworth, 2024-06-10:
anyone know how to make the file system writable, or should I submit a pull request to the main repository? 

Greetings @seanthewebber I too went down the same route of investigation that you have just experienced, and it appears that you need to augment the core source configuration for grub configuration, then build it into the subsequently installed version of HaOS. 
(This is why I submitted the PR and briefly discussed it with one of the maintainers.) 

It was later determined that my issue would not be resolved when utilizing the HP thin client version I have, because it seemed to be baked into the bios that if there is no vga termination, then there is a problem and the system should not post at all. 

However, you can check out my PR, and maybe resubmit or just re-open it for discussion. 

I believe that the original discussion also went down the lines of "should we inject this change for older equipment, since it is an edge case anyhow?" 

I don't believe this update will break anything for current devices, however, I am not an HaOS expert either. 

Respectfully, 

Martes Wigglesworth 
***@***.***

cino893, 2025-03-18:
Hi @Martes-Wigglesworth  could you share script to inject this change? I have this issue for the first time. Ubuntu booted without screen with no issues.

Martes-Wigglesworth, 2025-03-18:
> Hi [@Martes-Wigglesworth](https://github.com/Martes-Wigglesworth) could you share script to inject this change? I have this issue for the first time. Ubuntu booted without screen with no issues(so my guesses are that they are injecting this override for lack of monitor).

Greetings @cino893 I thought that I linked the old PR in these comments, however, I did not. 

Hence I will need to look it up and update my comment(s) with said link to my old PR which was closed.

Hopefully that should provide some insight as to what I was referencing during these discussions.

cino893, 2025-03-19:
I guess it's here
https://github.com/Martes-Wigglesworth/operating-system/commit/e8da915a465b968426f0b17203509c6c807d8a03
But this commit is pretty big
is there option to add text in some config file or run command to deal with this issue?

Martes-Wigglesworth, 2025-03-19:
> I guess it's here [Martes-Wigglesworth@e8da915](https://github.com/Martes-Wigglesworth/operating-system/commit/e8da915a465b968426f0b17203509c6c807d8a03) But this commit is pretty big is there option to add text in some config file or run command to deal with boot without display issue?

Greetings again @cino893.

I have updated my previous comment to the specific PR which was closed.

I only updated a single linked line in grub.cfg which is referenced within the commit for [(file(s) updated as per PR-2691|](https://github.com/home-assistant/operating-system/pull/2691/files).

Utilize the PR hyperlink, listed above, to view the file and its path.

Martes-Wigglesworth, 2025-03-23:
> Thank you 👍🏻
> 
> Unfortunately adding it to /mint/boot/EFI/grub.cfg didn't worked on MSI Cubi
> 
> I see that bios isn't even loading GRUB
> 
> I don't know how they solve it on Ubuntu for example.
> 
> I think I will buy dummy HDMI adapter

I believe that may be the solution for now, as that seemed to be the only thing that worked for me with a VGA version of your experience, on my end.  

Let me know how it turns out.  

I ended up forgetting to re-insert the VGA dongle, during my recent move, and have since lost it, and need to repurchase one for this little box. 

(I put everything on a ups so it never needed to reboot again, and then I broke that stuff down and migrated to a container and a new Kubernetes cluster for what I was doing.  However, I found use for this little box again, and ran into this issue around the time you started to ping this thread, so it brought back what I did previously.  LOL!!!)