home-assistant/operating-system #2088: Upgrade from 8.4 to 8.5 breaks IPv4 network connectivity

meyergru, 2022-08-27:
### Describe the issue you are experiencing

Upgrade went O.K., until I rebooted my virtual machine, only to find I could no longer access my instance.

Using a virtual console, I found that IPv6 connectivity was still there, whereas IPv4 was not assigned.

I was able to  set a static network configuration, reconfigure the network via GUI to DHCP and start over.
 

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

8,5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgrade from 8.4 to 8.5
2. Reboot
3. Look at "net info", which is empty for IPv4.

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing useful, since the reboot occured 3 days after the upgrade.
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing useful, since the reboot occured 3 days after the upgrade.
```

herrfeuer, 2022-08-27:
In my Setup it changed IP of deconz Container. Now deconz Integration broke and Zigbee isnt working anymore.

Hexa07, 2022-08-28:
Hi, I got a similar problem. My home assistant's IP address was 192.168.0.x, but after upgrading to 8.5 and an additional reboot, it changed to 172.23.245.x. Home Assistant cannot obtain an IP address from the DHCP anymore, and assigning manually an IP does not work either (the Web UI cannot be reached using this IP). 
The only way to access the Web UI now is to access it via the new IP 172.23.245.65, but it only works from the virtual machine host itself. 

I upgraded from 8.4 to 8.5 earlier today, and it worked fine for a few hours, until the next reboot. 

I am hosting this VM using VMware Workstation Pro 16, on a Windows 10 Pro computer.
After reading this GitHub issue, I made a fresh install using version 8.5 on this host, same result; the machine never got an IP 192.168.0.x from the DHCP. 
I didn't have time to perform a fresh install using 8.4, but I assume this should work. I will let you know the result when I get a chance to try. 

Let me know if you need additional information to reproduce or troubleshoot this problem. Thanks.

meyergru, 2022-08-28:
If you can access SSH or console, you can set the IP manually once and then access the web UI again in order to re-configure DHCP:

`ha network update enp0s18 --ipv4-method static --ipv4-address 192.168.0.x --ipv4-gateway 192.168.0.1 --ipv4-nameserver 192.168.0.1
`

Hexa07, 2022-08-28:
Thanks a lot, I will give it a try!

ronaldvdmeer, 2022-08-30:
I'm wondering if I might have the same. I have multiple interfaces connected to my HA setup and suddenly HA is not longer reachable on one of the interfaces. Routing problem?

agners, 2022-08-30:
@herrfeuer IP addresses of the add-ons are not stable. Essentially, they are assigned by Docker in order the add-ons start. A reboot can change the orders the add-ons getting started, so they can have any IP address.

The Supervisor makes sure the add-ons are properly named, and referencing them by the add-on hostname as shown in the add-on dialog (e.g. `core-deconz` in this case).

agners, 2022-08-30:
`172.23.245.x` is not a range 

> My home assistant's IP address was 192.168.0.x, but after upgrading to 8.5 and an additional reboot, it changed to 172.23.245.x.

@Hexa07 `172.23.245.x` is not a range used by HAOS or the Supervisor. It seems that this is an IP assigned by VMware. Please review your VM configuration, especially the assigned network cards. If you still having issues, its probably better to post in the community forums at https://community.home-assistant.io/ to find what causes your VM to get a wrong IP address.

meyergru, 2022-08-30:
I saw 172.xxxxx as well when my DHCP assignment was gone. That seems to be a docker bridge adress, not something imposed by the VM host/network.

agners, 2022-08-30:
@meyergru the IP address configuration is applied by the Supervisor. This sounds very much like a bug we have been seen in Supervisor (see https://github.com/home-assistant/supervisor/pull/3676), however, this has been fixed a while ago.

What recently has been fixed is a problem where IPv4 address would get disabled if the DHCP server was down right at the moment of boot (https://github.com/home-assistant/supervisor/pull/3806). Could it be that your DHCP was down at bootup (or Ethernet plugged out?)

In any case, this is very likely a Supervisor problem, not a OS problem.

agners, 2022-08-30:
> I saw 172.xxxxx as well when my DHCP assignment was gone. That seems to be a docker bridge adress, not something imposed by the VM host/network.

The add-ons Docker bridge uses `172.30.32.x/23`, and the default Docker is `172.17.0.1/16`. However, both do not match the mentioned `172.23.245.x`... So very likely something coming from outside of HAOS via DHCP (my guess is a network interface configured for NAT or something).

agners, 2022-08-30:
@meyergru do you have host level SSH access enabled (on port 2222)?

If so, the Supervisor logs right after startup of HAOS 8.5 would be interesting. Something along these lines should get them (if not reboot has taken place since then):

```
journalctl -b 0 -o cat CONTAINER_NAME=hassio_supervisor | head -n 150
```

agners, 2022-08-30:
> Actually the bootup was a Proxmox start with all machines which may delay the DHCP response (although the DHCP server is separate from the other VMs). So it depends on the definition of what is considered "down".

I'd say that should not count as down.

> journalctl is not installed...

You have to access the host OS SSH server, see https://developers.home-assistant.io/docs/operating-system/debugging#ssh-access-to-the-host.

meyergru, 2022-08-30:
The procedure to enable "real" root access somehow contradicts the "no in-between boot" requirement: when I attach a USB drive, I have to reboot the VM in order to make it visible. Is the another option to add the authorized key file and activate sshd on port 22222?

ronaldvdmeer, 2022-08-30:
I've noticed my problem goes away when I setup a static IP address. Same IP, subnet, gateway and DNS settings.
My problem was with 2 interfaces connected at the same time. One in the 192.168.1.x range and another in the 192.168.20.x range. I could temporarily solve it by disabling the 192.168.20.x interface in the hypervisor, boot the system and enable the interface after worth. But when I boot with both interfaces connected only the 192.168.20.x interface would work.

Now with the static settings it seems to work on boot as well. Something perhaps with routing tables and related to this problem or something completely different?

agners, 2022-08-30:
> The procedure to enable "real" root access somehow defeats the "no in-between boot" requirement: when I attach a USB drive, I have to reboot the VM in order to make it visible. Is there another option to add the authorized_keys file and activate sshd on port 22222?

Other then typing in shell, not really :cold_sweat: We intend to add it to the frontend at one point, but haven't happened so far.

Can you not hot-add USB devices? In Virtual Manager/KVM that works for me usually.

You should be able to access the logs of the previous boot using `-b -1` as well:

```
journalctl -b -1 -o cat CONTAINER_NAME=hassio_supervisor | head -n 150
```

agners, 2022-08-30:
> Now with the static settings it seems to work on boot as well. Something perhaps with routing tables and related to this problem or something completely different?

From what I understand NetworkManager declares one of the devices as primary (also visible in `ha network info`). How that is exactly determined I haven't looked into. It seems to pick the Ethernet interface typically, so it works for most cases.

Do both of your network have Internet connectivity? Next time it happens, can you check if the primary changes?

In any case, your case seems different to the issue this thread is about. It feels like this might lead to a feature request (make primary interface configurable). Can you open a separate issue?

ronaldvdmeer, 2022-08-30:
> In any case, your case seems different to the issue this thread is about. It feels like this might lead to a feature request (make primary interface configurable). Can you open a separate issue?

Thank you. Creating a new issue. https://github.com/home-assistant/operating-system/issues/2095

closed 2022-08-30 by agners

meyergru, 2022-08-30:
Yup, that most likely was my issue. Matter-of-fact, I use DHCPv4 and SLAAC but no DHCPv6, so that failed and caused the problem. Thanks!

Hexa07, 2022-09-03:
> @Hexa07 `172.23.245.x` is not a range used by HAOS or the Supervisor. It seems that this is an IP assigned by VMware. Please review your VM configuration, especially the assigned network cards. If [...]

Hi @agners , you were absolutely right, for some reason, the VMware network configuration changed, and the bridge network adapter could not reach the DHCP. I think it could have been caused because the bridge adapter selection in VMware was on "automatic". I assigned the bridged adapter "manually," and after rebooting HA it got an IP from the DHCP.

Thank you for your help!

agners, 2022-09-03:
@Hexa07 glad you found the issue, thanks for the update!