home-assistant/operating-system #2095: HA not accessible on one NIC when connected to multiple

ronaldvdmeer, 2022-08-30:
### Describe the issue you are experiencing

Today my HA setup was unavailable when I woke up and when I did some searching I saw it was caused by my setup having two network interfaces. I have an interface for my main network 192.168.1.1/24 and another interface for my IoT devices. Both interfaces connected to the same HA. Both also have internet access. When I disconnected the IoT interface (192.168.20.1/24) I was able again to contact the HA setup on my main network. 

As part of the debug proces i've changed from DHCP to Static IP on both interfaces and this seems to solve the problem. Now HA is accessable via both interfaces once again.

[image]

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

8.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Connect both network interfaces. 
2. Configure interface in HA as DHCP
3. Reboot system

```HA is only accessible on the IoT interface```

5. Disconnect the Iot Interface
 
```HA is only accessible on the Main interface```

### Anything in the Supervisor logs that might be useful for us?

```txt
Not seeing anything
```

### Anything in the Host logs that might be useful for us?

```txt
Not seeing anything
```

### System Health information

Home Assistant 2022.8.7
Supervisor 2022.08.5
Operating System 8.5
Frontend-versie: 20220802.0 - latest

### Additional information

Only today it stopped working. In the previous months I rebooted several times without any problem.

ronaldvdmeer, 2022-08-30:
While not being able to connect to HA via the main network.

```ha network info``` 
[image]

While able to connect to HA via the main network
[image]

Primary switched?

agners, 2022-08-30:
I am not 100% sure, but it seems to me that primary is whatever the default route points to (at least the docs suggest so https://people.freedesktop.org/~lkundrak/nm-docs/gdbus-org.freedesktop.NetworkManager.html#gdbus-property-org-freedesktop-NetworkManager.PrimaryConnection). Now the route is probably installed when the DHCP packet is received, and the DHCP server announces a gateway. Which one ends up to be the primary is probably random (maybe the last packet received).

Since you are dual-homed, and both offer a connection to the Internet, there is no good way for the system to autodetect.

However, from what I understand your main complaint is that you consider `enp0s18` as you main network interface, HA Core should be reachable on it by default. The Core has network configuration under [Settings >> System >> Network](https://my.home-assistant.io/redirect/network). The auto-detection is based on routing (which explains the behavior), but you have the option to select a specific network interface.

agners, 2022-08-30:
Actually, that setting in Home Assistant Core only affects multicasts, not accessibility (in a quick test HA Core listens on all available interfaces). Are you using the mDNS hostname to access the instance? If so, that would explain it as the hostname is only announced on one network interface.

ronaldvdmeer, 2022-08-30:
> Actually, that setting in Home Assistant Core only affects multicasts, not accessibility (in a quick test HA Core listens on all available interfaces). Are you using the mDNS hostname to access the instance? If so, that would explain it as the hostname is only announced on one network interface.

No I’m using the IP address and my phone has the nabucasa link configured. Both won’t work.

ronaldvdmeer, 2022-08-30:
> However, from what I understand your main complaint is that you consider `enp0s18` as you main network interface, HA Core should be reachable on it by default. Home Assistant Core has network configuration under [Settings >> System >> Network](https://my.home-assistant.io/redirect/network). The auto-detection is based on routing (which explains the behavior), but you have the option to select a specific network interface.

My complaint is that without any noticeable change HA is no longer reachable on the main interface preventing me and others in my household from accessing it. The IoT network is restricted and can only be accessed when I connect to the wifi access point associated with it. 

It sounds possible that a dhcp lease renewing was the reason.

agners, 2022-08-30:
> No I’m using the IP address and my phone has the nabucasa link configured. Both won’t work.

With IP address you mean `192.168.1.30` then? In my test with two interfaces, Home Assistant Core was accessible through both IPs, primary or not. So even if the IoT network is considered primary, HA should listen on `192.168.1.30`, and send answers through the appropriate link (since the Linux kernel will pick the route with the best match from the routing table).

Primary in the end should only be relevant through which way HAOS will connect to the internet... Which shouldn't matter, if both networks have internet connectivity.

In the situation where the IoT network is considered primary, and you are not able to access Home Assistant, can you ping `192.168.1.30`?

ronaldvdmeer, 2022-08-30:
> In the situation where the IoT network is considered primary, and you are not able to access Home Assistant, can you ping `192.168.1.30`?

I cannot ping it. However I just tested something else. In the screenshots i've posted earlier I saw I had ipv6 enabled on the IoT interface. So I re-enabled it again and once again I was only able to access the 192.168.20.29 address. When I disabled it I'm only able to access HA 192.168.1.30 and 192.168.20.29 becomes unavailable.

agners, 2022-08-30:
Hm interesting is that the primary interface seems to switch at boot (at 1:50 in the video). This is probably caused by our automatic Supervisor save.

But anyhow, it should :tm: all not matter! Both IP's should remain accessible. Both network should have proper routing entries, and the kernel should route responses accordingly. Can you check routing tables?

```
ha> login
docker exec homeassistant ip route
```

agners, 2022-08-30:
Do you bridge both network interfaces to the same L2 network (collision domain)? It could be spanning tree protocol at play.

ronaldvdmeer, 2022-08-30:
Hmmm HA runs on a hypervisor with a bond linking 2 physical NICs together on one switch configured with port aggregation. Spanning tree on the switch is configured as RSTP.

agners, 2022-08-31:
What happens if you disable it? As long as you have no loops in the network it should be fine to do so.

agners, 2022-08-31:
Just as a side note: IMHO, having two independent logical network on the same L2 network defeats the purpose to some degree: An attacker gaining access to a device on the IoT network might be able to catch packets on the network interface/change the IP to be part of the other network.

ronaldvdmeer, 2022-08-31:
> Can you check routing tables?
> `ha> login`
> `docker exec homeassistant ip route`
 
[image]

> What happens if you disable it? As long as you have no loops in the network it should be fine to do so.
Nothing changes it seems. 

> Just as a side note: IMHO, having two independent logical network on the same L2 network defeats the purpose to some degree: An attacker gaining access to a device on the IoT network might be able to catch packets on the network interface/change the IP to be part of the other network.

I'm aware of the security concerns and although I would like another solution some IoT devices force me to be within the same network/subnet with my HA setup. The fact that both NICS are connected to the same L2 network is mostly for the link aggregation and bandwith advantages.

agners, 2022-08-31:
The routes look good, clearly the kernel will use the respective network interfaces to send back responses for the two networks. Since ping also does not work, it can't be an issue of HA Core not listening.

IMHO, this is clearly a L2 issue: One of your Ethernet links is not connected sometimes.

> I'm aware of the security concerns and although I would like another solution some IoT devices force me to be within the same network/subnet with my HA setup.

I understand, but that does not mean you need to run the two network on the same L2 network. You can separate them physically still, or you can use VLANs.

> The fact that both NICS are connected to the same L2 network is mostly for the link aggregation and bandwidth advantages.

That doesn't really make sense to me: Link aggregation requires to setup bonding in Linux (see https://wiki.linuxfoundation.org/networking/bonding). Also, you would then end up with a single interface. If you enable link aggregation only on one side, that hardly can function properly. 

Furthermore, this is a virtual machine isn't it? Virtual interface do not have limitations in speed, so you can safely just use a single one, and assign that one interface both IP addresses. If your physical machine has two Ethernet ports, you would do the link aggregation between Switch and your virtual machine host. If you want to use bridging, you would then bridge that aggregated link interface.

ronaldvdmeer, 2022-08-31:
> > I'm aware of the security concerns and although I would like another solution some IoT devices force me to be within the same network/subnet with my HA setup.
> 
> I understand, but that does not mean you need to run the two network on the same L2 network. You can separate them physically still, or you can use VLANs.
> 
> > The fact that both NICS are connected to the same L2 network is mostly for the link aggregation and bandwidth advantages.
> 
> That doesn't really make sense to me: Link aggregation requires to setup bonding in Linux (see https://wiki.linuxfoundation.org/networking/bonding). Also, you would then end up with a single interface. If you enable link aggregation only on one side (e.g switch, but don't configure it on the connected host), that hardly can function properly.
> 
> Furthermore, this is a virtual machine isn't it? Virtual interface do not have limitations in speed, so you can safely just use a single one, and assign that one interface both IP addresses. If your physical machine has two Ethernet ports, you would do the link aggregation between Switch and your virtual machine host. If you want to use bridging, you would then bridge that aggregated link interface.

We are going completely offtopic but on the same L2 switch is a NAS that is also connected with link aggregation. This way multiple connections with 1gbit speed are achievable between the NAS and the Hypervisor. 

> The routes look good, clearly the kernel will use the respective network interfaces to send back responses for the two networks. Since ping also does not work, it can't be an issue of HA Core not listening.
> 
> IMHO, this is clearly a L2 issue: One of your Ethernet links is not connected sometimes.

Hmm interesting but how does this explain that when enabling ipv6 on one of the interfaces suddenly the other network gets available. Doesnt that sound more like something related to the operating system instead of the L2 switch?

agners, 2022-08-31:
> We are going completely offtopic but on the same L2 switch is a NAS that is also connected with link aggregation. This way multiple connections with 1gbit speed are achievable between the NAS and the Hypervisor.

Yes, but why does the VM has multiple network interfaces?

> Also doing a ping from HA to a IoT device in the 192.168.20.X network works.

That is news. Now it definitely sounds like a routing issue. But not a HAOS routing issue, as its routing table looks fine :)

Does your Windows machine you are pinging from is in both network too (on IP level?)?

You can also `tcpdump` the traffic from within the Core container:

```
# docker exec -it homeassistant /bin/bash
bash-5.1# apk add tcpdump
bash-5.1# tcpdump -f icmp
```

agners, 2022-08-31:
Btw, maybe its some smarts from your router? I remember playing with a Zyxel firewall at one point, and it had some auto detection when two networks were connected to the same switch (as it kinda moots the firewalls point). It took me a while to figure out why things did not work correctly.

agners, 2022-08-31:
[image]

It is weird that the time here is quite a bit higher. Also different TTL. Are those packets maybe sent to the GW and routed through it? That would then point to L2 issue again.

ronaldvdmeer, 2022-08-31:
> Yes, but why does the VM has multiple network interfaces?

Basically it because this integration issue: https://github.com/rytilahti/python-miio/issues/422 and https://python-miio.readthedocs.io/en/latest/troubleshooting.html#discover-devices-across-subnets

[image]

I did some testing outside home-assistant with a LXC container running debian. It seems like the problem is reproducable on that container as well however when I configure both interfaces to have a vlan they are both reachable. vNIC1 is configured with vlan 10 and vNIC2 is configured with vlan 20. Now that works. But as soon as I change vNIC1 to no-vlan only one interface is reachable once again. 

[image]

Not sure if that causes a problem but on HA OS there are 2 default routes instead of one on my test machine.

agners, 2022-08-31:
> Not sure if that causes a problem but on HA OS there are 2 default routes instead of one on my test machine.

It should not matter, as to determine the interface to send back your ping package should use the more specific routes.

> On topic: I did some testing outside home-assistant with a LXC container running debian. It seems like the problem is reproducable on that container as well however when I configure both interfaces to have a vlan they are both reachable. vNIC1 is configured with vlan 10 and vNIC2 is configured with vlan 20. Now that works. But as soon as I change vNIC1 to no-vlan only one interface is reachable once again.

Maybe a full VM would be better test case, as containers use the host network stack essentially (with networking namespace of course, but still).

In any case, to me it still seems like something n L2 is not proper. `tcpdump` should really help to see which packets make it where. You can also filter by network interface using e.g. `tcpdump -i enp0s18`.

ronaldvdmeer, 2022-08-31:
Hmmm. I do see the icmp traffic with tcpdump on the interface.

[image]

ronaldvdmeer, 2022-09-03:
Any idea @agners ?

agners, 2022-09-05:
@ronaldvdmeer this is not a HAOS issue, and I am a bit hesitant to fix other people's network issues :sweat_smile: 

It is also hard when I don't have the full picture of your network, and questions don't get answered, like this one:

> Does your Windows machine you are pinging from is in both network too (on IP level?)?

Why is the ICMP request for `192.168.20.29` sent from a `192.168.10.100` IP? What I expect Linux to do is send the answer out on the `enp0s18` interface, as that is the interface which is _directly_ connected to the `192.168.10.x` network.

I recommend to post this issue at https://community.home-assistant.io/, along with the full configuration information of your network, including your Windows host you are pinging from, router and their configuration, your hypervisor network configuration, the virtual machine's virtual network interface configuration and HAOS configuration.

ronaldvdmeer, 2022-09-05:
I understand completely and appreciate the time you already have taken to answer my questions. My concern was that it has something to do with HA because nothing changed in the network but HA was upgraded. 

I’ll be closing this now because it seems to be related to something else.

closed 2022-09-05 by ronaldvdmeer