home-assistant/operating-system #2098: var / log getting to huge, can't reduce it.

malosaa, 2022-08-31:
### Describe the issue you are experiencing

I did check the forums and github, but this issue is still there.

the problem i have is that in HASSOS the var/log is getting to huge, and i can't reduce it.
my hass system is almost out of space.

i did ssh with putty and try to run command (journalctl --vacuum-size=100m)
but journalctl does not work, it says command not found.

the logs are located in var/logs/

this is a fraction of the folders
```
321.6M  ./journal/d7a06afcf9f749a6822aecdbc8e7347e
3.0G    ./journal/5703c6ebabe24d8a85d1e7c12ff1441f
8.8G    ./journal/46db4e65632c431489ba1567ed237247
12.1G    ./journal
```

any idea how to do it ?

regards 

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

2022.8.7

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

nope

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

agners, 2022-09-01:
Hm, it seems that there are logs from multiple machine IDs. What was the original version of your installation?

You can print the currently active machine ID with
```
systemd-machine-id-setup --print
```

It should be save to just delete all the others.

I assume `d7a06afcf9f749a6822aecdbc8e7347e` is your active machine ID. The vacuuming process only applies to archived logs, so it seems you have 221.6MB active logs.

malosaa, 2022-09-01:
@agners  

many thanks for the reply,

its this one 
`5703c6ebabe24d8a85d1e7c12ff1441f`

agners, 2022-09-02:
> The question is how can i have multiple device id's ? thats weird.

The system probably lost it at one point, hence my question: What was the original version of your installation?

> and how can i delete the other ones

```
rm -r ./journal/46db4e65632c431489ba1567ed237247
```

>  and reduce this active one ?

The vaccuum command should do it, but its a one time thing. In the end, it should not grow beyond 4GB with the recent HAOS releases, so I'd recommend just leave the active one as is.

malosaa, 2022-09-20:
> The system probably lost it at one point, hence my question: What was the original version of your installation?
>
>
> ```
> rm -r ./journal/46db4e65632c431489ba1567ed237247
> ```
>
>
> The vaccuum command [...]

Sorry for the late reply, i did totally forget to reply.

It did work great and got rid of the old logs.
I don't know why it happened but its fixed.

many thanks again agners!

closed 2022-09-20 by malosaa

mikegodin23, 2023-05-14:
Hi.
I have a similar issue, but with another folder:
/var/log.hdd/
exactly a subfolder in
/var/log.hdd/journal
It creates more than 10Mb of files per hour.
Strangely the name of the folder
4d4f6d7f9df948e6afd380c743b4ba53
is the same as the one that is created in the
/var/log/journal 
Is it safe to remove it?
If so, is there a way to delete it permanently?
Thanks.
Miguel