home-assistant/operating-system #2105: Netgear NIGHTHAWK routers cannot install HAOS

GorgonDeDragon, 2022-09-02:
### Describe the issue you are experiencing

Using a NETGEAR Nighthawk router seems to cause HAOS install to get stuck in a loop and is unable to finish installing before getting stuck in this loop

`22-09-02 11:16:16 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
22-09-02 11:16:46 INFO (SyncWorker_2) [supervisor.docker.interface] Updating image ghcr.io/home-assistant/raspberrypi4-64-homeassistant:landingpage to ghcr.io/home-assistant/raspberrypi4-64-homeassistant:2022.8.7
22-09-02 11:16:46 INFO (SyncWorker_2) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/raspberrypi4-64-homeassistant with tag 2022.8.7.
22-09-02 11:22:21 ERROR (SyncWorker_2) [supervisor.docker.interface] Can't install ghcr.io/home-assistant/raspberrypi4-64-homeassistant:2022.8.7: 500 Server Error for http+docker://localhost/v1.41/images/create?tag=2022.8.7&fromImage=ghcr.io%2Fhome-assistant%2Fraspberrypi4-64-homeassistant&platform=linux%2Farm64: Internal Server Error ("Get "https://ghcr.io/v2/": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)")
22-09-02 11:22:21 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec`

i have tested this on 3 other 8.0 installs and nothing has worked. the CLI does not install so i cannot access the startup screen,

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

8.5 - 8.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Own Nighthawk router and plug Raspberry Pi into nighthawk router

attempt to install HAOS (through the raspberry pi or through flashing as told in the guide)

open "preparing Home Assistant" in browser

observe

### Anything in the Supervisor logs that might be useful for us?

```txt
22-09-02 11:16:16 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
22-09-02 11:16:46 INFO (SyncWorker_2) [supervisor.docker.interface] Updating image ghcr.io/home-assistant/raspberrypi4-64-homeassistant:landingpage to ghcr.io/home-assistant/raspberrypi4-64-homeassistant:2022.8.7
22-09-02 11:16:46 INFO (SyncWorker_2) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/raspberrypi4-64-homeassistant with tag 2022.8.7.
22-09-02 11:22:21 ERROR (SyncWorker_2) [supervisor.docker.interface] Can't install ghcr.io/home-assistant/raspberrypi4-64-homeassistant:2022.8.7: 500 Server Error for http+docker://localhost/v1.41/images/create?tag=2022.8.7&fromImage=ghcr.io%2Fhome-assistant%2Fraspberrypi4-64-homeassistant&platform=linux%2Farm64: Internal Server Error ("Get "https://ghcr.io/v2/": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)")
22-09-02 11:22:21 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
```

### Anything in the Host logs that might be useful for us?

```txt
22-09-02 11:16:16 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
22-09-02 11:16:46 INFO (SyncWorker_2) [supervisor.docker.interface] Updating image ghcr.io/home-assistant/raspberrypi4-64-homeassistant:landingpage to ghcr.io/home-assistant/raspberrypi4-64-homeassistant:2022.8.7
22-09-02 11:16:46 INFO (SyncWorker_2) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/raspberrypi4-64-homeassistant with tag 2022.8.7.
22-09-02 11:22:21 ERROR (SyncWorker_2) [supervisor.docker.interface] Can't install ghcr.io/home-assistant/raspberrypi4-64-homeassistant:2022.8.7: 500 Server Error for http+docker://localhost/v1.41/images/create?tag=2022.8.7&fromImage=ghcr.io%2Fhome-assistant%2Fraspberrypi4-64-homeassistant&platform=linux%2Farm64: Internal Server Error ("Get "https://ghcr.io/v2/": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)")
22-09-02 11:22:21 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
```

### System Health information

cant acess

agners, 2022-09-03:
This typically indicates DNS issues. Did you plug-in Ethernet at startup? Maybe another reboot helps?

You can configure a DNS server manually by connecting a keyboard and display and type the following into the Home Assistant OS CLI:

```
ha > network update eth0 --ipv4-nameserver 8.8.8.8
```

GorgonDeDragon, 2022-09-03:
as of 7.0 (and onwards) none have reached the CLI it seems, creating a blank screen.

i ran it back to 6.7 and thats the only one that seems to fully boot right, up untill it needs to download files where it gets stuck on setup (i have let it sit for 2 hours)

i will try out the net update and report back on that

GorgonDeDragon, 2022-09-04:
so far ive gotten core to update to OS 8.5 and open to CLI the after putting in your command as that was the only one that could update.

i did your command again and 2 things im still getting is an unhealthy system and 500 server errors. i have tried updating Docker, Supervisor and DNS with no luck.

agners, 2022-09-05:
> so far ive gotten core to update to OS 8.5 and open to CLI the after putting in your command as that was the only one that could update.

Hm, so OS update worked then?

What is `ha os network` and `ha supervisor info` showing?

GorgonDeDragon, 2022-09-05:
So Super Info gets me this
`addons: []
addons_repositories:
- https://githubg.com/hassio-addons/repository
arch: aarch64
channel: stable
debug: false
debug_block: false
ip_adress: 172.30.32.2
logging: info
supportted: true
timezone America/New_York
Update_available: true
version: 2021.10.6
version_latest: 2022.08.5
wait_boot: 5`

ha> os network only brings up the OS command help screen

agners, 2022-09-06:
Oh wow, your Supervisor is very old: `2021.10.6`. Is there something in place to actively block Supervisor updates?

Which version of HAOS is that?

The second command should read `ha network`, sorry about that. Ideally run both commands with a HAOS 8.5 installed, it is easier to debug with the latest HAOS version.

GorgonDeDragon, 2022-09-06:
> Oh wow, your Supervisor is very old: `2021.10.6`. Is there something in place to actively block Supervisor updates?

unless a fresh install automatically blocks it. I'm afraid not as I am just getting 500 errors. any repair command seems to be swatted down by an "unhealthy system" (my pi and the sd cards are no older then a month) 

With HAOS 8.5 i only seem to find more issues with the HDMI failing halfway through startup to the CLI and still having it unable to download any more updates. i can only observe it on the myip:8123 onboarding screen.

agners, 2022-09-08:
> With HAOS 8.5 i only seem to find more issues with the HDMI failing halfway through startup to the CLI ...

What do you mean by that?

Is the above output of `ha supervisor info` from a new HAOS 8.5 install?

closed 2022-12-14 by github-actions[bot]

kasekesy, 2023-02-10:
getting same type of errors netgear router

23-02-10 00:41:23 INFO (SyncWorker_0) [supervisor.docker.interface] Updating image [http://ghcr.io/.../raspberrypi3-64-homeassistant:landingpage](http://ghcr.io/home-assistant/raspberrypi3-64-homeassistant:landingpage?fbclid=IwAR22lo99sY9BycD-N28FmHT8fjIltQmvit0ntzfV-u9WZVCk3PfJXbzNlZA) to [http://ghcr.io/hom.../raspberrypi3-64-homeassistant:2023.2.3](http://ghcr.io/home-assistant/raspberrypi3-64-homeassistant:2023.2.3?fbclid=IwAR2TFWYoXBt_r-OcWnUBJU0W69Y6RIoYDrsadPslXoOMgB-Hn8oXpDmKl00)
23-02-10 00:41:23 INFO (SyncWorker_0) [supervisor.docker.interface] Downloading docker image [ghcr.io/home-assistant/raspberrypi3-64-homeassistant](http://ghcr.io/home-assistant/raspberrypi3-64-homeassistant?fbclid=IwAR0MA9vFPvKcSRELfgVxME3m7Yal5RoVHdPiiwtWEKF7kW5MnqTRG4gyUPo) with tag 2023.2.3.
23-02-10 00:47:03 ERROR (SyncWorker_0) [supervisor.docker.interface] Can't install [http://ghcr.io/hom.../raspberrypi3-64-homeassistant:2023.2.3](http://ghcr.io/home-assistant/raspberrypi3-64-homeassistant:2023.2.3?fbclid=IwAR11dBpyH9w1JkqpIkS18xsRL_UaazNjgfLVeYBIyNaPv0TUcYrRRD8IYlE): 500 Server Error for http+docker://localhost/v1.41/images/create?tag=2023.2.3&fromImage=ghcr.io%2Fhome-assistant%2Fraspberrypi3-64-homeassistant&platform=linux%2Farm64: Internal Server Error ("Get "[https://ghcr.io/v2/](https://l.facebook.com/l.php?u=https%3A%2F%2Fghcr.io%2Fv2%2F%3Ffbclid%3DIwAR0b8-s_B216NDbzU0xElRm12tbVUwlb0EPU1k8pReEfDFIBnr-gq0d1SrU&h=AT1KM1VwYooFlqQ_FEkEAf1QJbEWBmKSE4ck4t7wbDWjVAPebCmlM1n7c3JiLte0utgPWuVTyzhPSqn0REmIe7zgRc5jsAvqwcwhqhzbeh-0QepHJfll_TRvfFkq800afA&__tn__=-UK-R&c[0]=AT36mSHwlcHnEAQtc0X9jA81qVs6bNHwPlIGbdfh4DRwIfj2R4K6Oyi5P02JGF7xXlQmuBi-l_6acGpzDx5LG6GmjDLytcLwoJh_CYD0SqIt9QvgfTKxR5bpoLTm6nSNV0frfuHBmhmqL-Wl57Co8QtA7_vdyD72)": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)")
23-02-10 00:47:03 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec

agners, 2023-02-14:
There is a newer issue with more details, see #2182. Ideally you post your case there as well, along with information what Netgear router and which firmware version you are using.