home-assistant/operating-system #2110: Bluetooth not available after upgrade to 8.5

maretodoric, 2022-09-04:
### Describe the issue you are experiencing

After upgrade from 8.4 to 8.5 my bluetooth controller seems to be missing from bluetoothctl, hcitool, hciconfig, rfkill list. But dmesg shows it's loaded, kernel modules are loaded as well.

### What operating system image do you use?

rpi3 (Raspberry Pi 3 32-bit OS)

### What version of Home Assistant Operating System is installed?

8.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. upgrade ha os from 8.4 to 8.5

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing
```

### Additional information

I've tried downgrading back to 8.4 but didn't resolve the issue amd I've tried full power cycle without luck.

agners, 2022-09-05:
Bluetooth on Raspberry Pi 3 32-bit with HAOS 8.5 seems to work on my end:

```
# bluetoothctl 
Agent registered
[CHG] Controller B8:27:EB:63:F4:E1 Pairable: yes
[bluetooth]# quit
# ha os info
board: rpi3
boot: B
data_disk: /dev/mmcblk0p8
update_available: true
version: "8.5"
version_latest: 9.0.rc1
```

Did you do configuration changes in `config.txt` maybe? If you have access to the HAOS console, can you check these two services?

```
systemctl status hciuart.service
systemctl status bthelper@hci0.service
```

maretodoric, 2022-09-05:
I haven't done any changes in a whilee.
I checked, and it seems they're not running and hciuart is timing out while trying to start

```
# systemctl status hciuart
× hciuart.service - Configure Bluetooth Modems connected by UART
     Loaded: loaded (/usr/lib/systemd/system/hciuart.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sun 2022-09-04 21:53:26 UTC; 13h ago
    Process: 358 ExecStart=/usr/bin/btuart (code=exited, status=1/FAILURE)

Sep 04 21:52:55 homeassistant systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Sep 04 21:53:26 homeassistant btuart[388]: Initialization timed out.
Sep 04 21:53:26 homeassistant btuart[388]: bcm43xx_init
Sep 04 21:53:26 homeassistant systemd[1]: hciuart.service: Control process exited, code=exited, status=1/FAILURE
Sep 04 21:53:26 homeassistant systemd[1]: hciuart.service: Failed with result 'exit-code'.
Sep 04 21:53:26 homeassistant systemd[1]: Failed to start Configure Bluetooth Modems connected by UART.
# systemctl status bthelper@hci0.service
○ bthelper@hci0.service - Raspberry Pi bluetooth helper
     Loaded: loaded (/usr/lib/systemd/system/bthelper@.service; static)
     Active: inactive (dead)
# systemctl start hciuart
Job for hciuart.service failed because the control process exited with error code.
See "systemctl status hciuart.service" and "journalctl -xeu hciuart.service" for details.
# systemctl status hciuart
× hciuart.service - Configure Bluetooth Modems connected by UART
     Loaded: loaded (/usr/lib/systemd/system/hciuart.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Mon 2022-09-05 11:22:17 UTC; 33s ago
    Process: 28524 ExecStart=/usr/bin/btuart (code=exited, status=1/FAILURE)

Sep 05 11:22:07 homeassistant systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Sep 05 11:22:17 homeassistant btuart[28548]: Initialization timed out.
Sep 05 11:22:17 homeassistant btuart[28548]: bcm43xx_init
Sep 05 11:22:17 homeassistant systemd[1]: hciuart.service: Control process exited, code=exited, status=1/FAILURE
Sep 05 11:22:17 homeassistant systemd[1]: hciuart.service: Failed with result 'exit-code'.
Sep 05 11:22:17 homeassistant systemd[1]: Failed to start Configure Bluetooth Modems connected by UART.
# systemctl status bluetooth
● bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2022-09-04 21:55:57 UTC; 13h ago
       Docs: man:bluetoothd(8)
   Main PID: 5033 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 2055)
     Memory: 652.0K
     CGroup: /system.slice/bluetooth.service
             └─5033 /usr/libexec/bluetooth/bluetoothd

Sep 04 21:55:56 homeassistant systemd[1]: Starting Bluetooth service...
Sep 04 21:55:57 homeassistant bluetoothd[5033]: Bluetooth daemon 5.63
Sep 04 21:55:57 homeassistant systemd[1]: Started Bluetooth service.
Sep 04 21:55:57 homeassistant bluetoothd[5033]: Starting SDP server
Sep 04 21:55:57 homeassistant bluetoothd[5033]: Bluetooth management interface 1.21 initialized
```

In home assistant container if i run rfkill list i get no bluetooth device available (weird) but it appears in dmesg during boot
```
# dmesg |grep -Ei 'hci|blue'
[    2.581962] sdhci: Secure Digital Host Controller Interface driver
[    2.584937] sdhci: Copyright(c) Pierre Ossman
[    2.587646] sdhci-pltfm: SDHCI platform and OF driver helper
[  109.873658] Bluetooth: Core ver 2.22
[  109.873862] NET: Registered PF_BLUETOOTH protocol family
[  109.873874] Bluetooth: HCI device and connection manager initialized
[  109.873905] Bluetooth: HCI socket layer initialized
[  109.873919] Bluetooth: L2CAP socket layer initialized
[  109.873948] Bluetooth: SCO socket layer initialized
[  173.770245] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[  173.770272] Bluetooth: BNEP filters: protocol multicast
[  173.770293] Bluetooth: BNEP socket layer initialized
```

lsmod also shows modules loaded
```
# lsmod | grep -Ei 'blue|hci|bcm43'
hci_uart               40960  0
btbcm                  16384  1 hci_uart
bluetooth             425984  9 hci_uart,bnep,btbcm
ecdh_generic           16384  1 bluetooth
rfkill                 32768  4 bluetooth,cfg80211
```

agners, 2022-09-05:
`hciuart.service` looks different here:

```
Sep 05 08:04:46 homeassistant systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Sep 05 08:04:47 homeassistant btuart[374]: bcm43xx_init
Sep 05 08:04:47 homeassistant btuart[374]: Set Controller UART speed to 460800 bit/s
Sep 05 08:04:47 homeassistant btuart[374]: Flash firmware /lib/firmware/brcm/BCM43430A1.hcd
Sep 05 08:04:47 homeassistant btuart[374]: Set BDADDR UART: b8:27:eb:63:f4:e1
Sep 05 08:04:47 homeassistant btuart[374]: Set Controller UART speed to 460800 bit/s
Sep 05 08:04:47 homeassistant btuart[374]: Device setup complete
Sep 05 08:04:47 homeassistant systemd[1]: Started Configure Bluetooth Modems connected by UART.
```

What device tree overlays are loaded? `cat /mnt/boot/config.txt | grep dtoverlay`

Maybe a complete poweroff with a minute or so disconnected from power helps?

maretodoric, 2022-09-05:
None of the overlays appear to be loaded, (they're commented), how does it look like on your end?

```
# cat /mnt/boot/config.txt | grep dtoverlay
#dtoverlay=lirc-rpi
#dtoverlay=vc4-fkms-v3d
```

Tried to power off multiple times , none worked unfortunately. Had problems earlier when restoring from backup failed  to start bluetooth and full power cycle fixed the issue so i tried it again here, but didn't seem to work.

Ever since i've restored previous backup, downgraded OS version, updated again with full power cycle each time between and nothing seem to have worked.

agners, 2022-09-05:
```
# cat /mnt/boot/config.txt | grep dtoverlay
#dtoverlay=lirc-rpi
#dtoverlay=miniuart-bt
#dtoverlay=rpi-rf-mod
dtoverlay=vc4-fkms-v3d
#dtoverlay=vc4-fkms-v3d
```

The `config.txt` doesn't get upgrades, so mine is probably a newer installation. I don't think that the vc4-fkms-v3d one matters, as its on the `[pi4]` section.

> Tried to power off multiple times , none worked unfortunately. Had problems earlier when restoring from backup failed to start bluetooth and full power cycle fixed the issue so i tried it again [...]

That is weird, that almost sounds like hardware broke? :confused: 

Maybe something mis-configured and accidentally accessing the same serial port (ZHA/deCONZ...)?

maretodoric, 2022-09-11:
I have just migrated to ho elitedesk i had near and installed raspberry pi os on pi3 that had "Bluetooth failure".
Raspberry pi os can see Bluetooth adapter just fine. It's visible in rfkill list, adapters shows up in bluetoothctl list. Still have no clue what happened after upgrade last time

agners, 2022-09-12:
Ok, that is weird. The most likely explanation seems to me like some add-on or integration messed with the UART. Glad Bluetooth works again with a new installation.

closed 2022-09-12 by agners