home-assistant/operating-system #2115: Firefox Functionality

Hamish1066, 2022-09-06:
### Describe the issue you are experiencing

When editing an entity item in the dashboard, I can't use the clickable option to move an entity up or down on the list. It works fine on the same computer with Chrome and Safari and on the MacOS app.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

8.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Open to edit an entities card in dashboard (accessed through Firefox)
2. Try to move an entity using the button
3. It doesn't work
...

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
This is on a Macbook Pro using MacOS 12.5.1. It also doesn't work on the iOS app, but I can do a separate bug report for that if necessary.
```

agners, 2022-09-08:
@Hamish1066 this repository is about Home Assistant Operating System issues. Your problem sounds more a frontend issue. Please check if you can reproduce the issue with the latest Home Assistant Core/Frontend version, and report an issue at https://github.com/home-assistant/frontend/ if the problem still persists.

Hamish1066, 2022-09-08:
Thanks. Yes, it's working fine after I upgraded to 9.0.