home-assistant/operating-system #2119: USBIP driver and kernel module missing from hassio

irakhlin, 2022-09-08:
### Describe the issue you are experiencing

The most recent release of home assistant made a few references to using USBIP modules to utilize as an option for remote bluetooth controllers, which I was able to get working in a core install of home assistant. Unfortunately the usbip kernel module and utility is missing from the build. Taking a look I was able to enable it in a few places and build the OS successfully https://github.com/ilya-ws/operating-system/tree/usbip

I can verify the modules build but I am still yet to still it all out. Is this something that is unintentionally missing?

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

8.5

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

USBIP modules not compiled by the build and is unavailable in the os.

### Anything in the Supervisor logs that might be useful for us?

```txt
n/a
```

### Anything in the Host logs that might be useful for us?

```txt
n/a
```

irakhlin, 2022-09-08:
... just came across this https://github.com/home-assistant/operating-system/pull/2047/commits/fa777e0336ae419e6569a6d83f71a4b2ed833bd1 looks like you guys got it.

irakhlin, 2022-09-09:
Was able to get usbip working on my build by simply modprobe loading the vhci driver. On the most recently released 9.0rc2 I am still not able to find the usbip binaries.

irakhlin, 2022-09-10:
Update: After some additional testing of my build which did contain usbip, it would seem you may also want to build package hwdata at least for USB. USBIP will warn that it cannot find hwdata which is fine but the result is that it will not be able to identify the remote usb device. This could become an issue for auto discovery.

BR2_PACKAGE_HWDATA=y
BR2_PACKAGE_HWDATA_IAB_OUI_TXT=n
BR2_PACKAGE_HWDATA_PCI_IDS=n
BR2_PACKAGE_HWDATA_PNP_IDS=n
BR2_PACKAGE_HWDATA_USB_IDS=y

https://fossies.org/linux/buildroot/package/hwdata/Config.in and  https://github.com/vcrhonek/hwdata for reference. It seems there is also a python binding available, not sure if this is needed as I am not sure what the plans are for further integration

agners, 2022-09-12:
Not shipping USB IP binaries with HAOS has been a conscious decision. The idea is to keep HAOS as lean as possible.

On HAOS configuring USB/IP should be done by an add-on. With the `kernel_modules` configuration, an add-on can load kernel modules. See also: https://developers.home-assistant.io/docs/add-ons/configuration

irakhlin, 2022-09-12:
> Not shipping USB IP binaries with HAOS has been a conscious decision. The idea is to keep HAOS as lean as possible.
>
> On HAOS configuring USB/IP should be done by an add-on. With the `kernel_modules` [...]

@agners I understand now the idea behind including just the kernel modules however I am still hitting an issue when I attempt to do this in a addon. 

I created a debugging addon with ssh enabled on a custom port and given it full_access, kernel_modules and devicetree options. Inside the addon I have the usbip binaries. First issue I ran into was the vhci-hcd module not being loaded, I was not able to figure out how to load this module in the context of an addon as modprobe is not available. I was able to ssh into my system on port 22222 and modprobe vhci-hcd runs fine at that point. The issue I could not get passed is the following (running inside my addon container):

```
root@7cbb8c6e-usbipclient:/sys/devices/platform/vhci_hcd.0# usbip list -r 10.0.0.77
Exportable USB devices
======================
 - 10.0.0.77
      1-1.4: Realtek Semiconductor Corp. : unknown product (0bda:a725)
           : /sys/devices/platform/soc/20980000.usb/usb1/1-1/1-1.4
           : Wireless / Radio Frequency / Bluetooth (e0/01/01)
           :  0 - Wireless / Radio Frequency / Bluetooth (e0/01/01)
           :  1 - Wireless / Radio Frequency / Bluetooth (e0/01/01)
```

```
root@7cbb8c6e-usbipclient:/sys/devices/platform/vhci_hcd.0# usbip --debug attach -r 10.0.0.77 -b 1-1.4
usbip: debug: usbip.c:129:[run_command] running command: `attach'
libusbip: debug: vhci_driver.c:269:[usbip_vhci_driver_open] available ports: 16
libusbip: debug: vhci_driver.c:281:[usbip_vhci_driver_open] available controllers: 1
libusbip: debug: vhci_driver.c:128:[refresh_imported_device_list] controller 0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 0 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 1 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 2 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 3 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 4 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 5 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 6 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub hs port 7 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 8 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 9 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 10 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 11 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 12 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 13 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 14 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:67:[parse_status] hub ss port 15 status 4 speed 0 devid 0
libusbip: debug: vhci_driver.c:69:[parse_status] sockfd 0 lbusid 0-0
libusbip: debug: vhci_driver.c:104:[parse_status] exit
usbip: debug: usbip_attach.c:100:[import_device] got free port 0
libusbip: debug: vhci_driver.c:367:[usbip_vhci_attach_device2] writing: 0 3 65541 2
libusbip: debug: vhci_driver.c:372:[usbip_vhci_attach_device2] attach attribute path: /sys/devices/platform/vhci_hcd.0/attach
usbip: debug: sysfs_utils.c:18:[write_sysfs_attribute] error opening attribute /sys/devices/platform/vhci_hcd.0/attach
libusbip: debug: vhci_driver.c:376:[usbip_vhci_attach_device2] write_sysfs_attribute failed
usbip: error: import device
```

While I do see /sys/devices/platform/vhci_hcd.0/attach mounted in the container:
```
root@7cbb8c6e-usbipclient:/sys/devices/platform/vhci_hcd.0# ls -l
total 0
--w------- 1 root root 4096 Sep 12 16:00 attach
--w------- 1 root root 4096 Sep 12 16:06 detach
lrwxrwxrwx 1 root root    0 Sep 12 16:06 driver -> ../../../bus/platform/drivers
-rw-r--r-- 1 root root 4096 Sep 12 16:06 driver_override
-r--r--r-- 1 root root 4096 Sep 12 16:06 modalias
-r--r--r-- 1 root root 4096 Sep 12 16:00 nports
drwxr-xr-x 2 root root    0 Sep 12 16:06 power
-r--r--r-- 1 root root 4096 Sep 12 16:00 status
[... cut ...]
-rw-r--r-- 1 root root 4096 Sep 12 16:00 uevent
drwxr-xr-x 5 root root    0 Sep 12 16:06 usb3
drwxr-xr-x 5 root root    0 Sep 12 16:06 usb4
-rw-r--r-- 1 root root 4096 Sep 12 16:06 usbip_debug
```

My understanding is that /sys is mounted as read only: 
```
root@7cbb8c6e-usbipclient:/sys/devices/platform/vhci_hcd.0# cat attach
cat: attach: Permission denied
root@7cbb8c6e-usbipclient:/sys/devices/platform/vhci_hcd.0# echo 'test' > attach
-bash: attach: Read-only file system
```

Please let me know what I am missing

irakhlin, 2022-09-12:
@agners It would seem that this may need to be added as functionality to supervisor? Am I correct in this assumption? Something similar to https://github.com/home-assistant/supervisor/blob/4f272ad4fdda3289e1461f913cd8153237a6e81d/supervisor/docker/addon.py#L383  where 
/sys/devices/platform/vhci_hcd.[0...x] are mounted with rw instead of ro? Potentially hidden behind a config switch to enable usbip?

agners, 2022-09-12:
Hm, I see, yeah ideally I guess that would be a new Supervisor option. For now, maybe `full_access` does the job? Or mount privileges and mount sysfs rw yourself? :imp:

irakhlin, 2022-09-12:
@agners Unfortunately full_access does not seem to be enough, I was not sure how to manually add additional mount directives as the hassio addon template does not allow you to manually mount /sys so I modified the container using an updated version of portainer to mount the following directories with RW permission:
/sys/devices/platform/vhci_hcd.0:/sys/devices/platform/vhci_hcd.0 rw 
/sys/bus/platform/drivers/vhci_hcd:/sys/bus/platform/drivers/vhci_hcd rw

This seems to work and the addon container can now correctly binds to a remote usb device. So I guess to say you were initially correct this is not a bug in home assistant os but did you want me to open a bug/feature request in supervisor ? It would seem maybe a single option of enabling "usbip" or something more generic may make most sense but it is not my project !

agners, 2022-09-12:
> It would seem maybe a single option of enabling "usbip" or something more generic may make most sense but it is not my project !

Yeah something more generic would be nice indeed. Best is to open an issue in the Supervisor issue tracker, so this can be discussed at the right place.

closed 2022-12-18 by github-actions[bot]