home-assistant/operating-system #2122: Missing bluetooth driver

b8vrede, 2022-09-11:
### Describe the issue you are experiencing

When attempting to use the on-board bluetooth of my Dell Optiplex 3060, the bluetooth adaptor is not loaded correctly due to missing firmware.

```
[core-ssh dbus]$ dmesg | grep -i blue
[    0.207441] Bluetooth: Core ver 2.22
[    0.207450] NET: Registered PF_BLUETOOTH protocol family
[    0.207453] Bluetooth: HCI device and connection manager initialized
[    0.207457] Bluetooth: HCI socket layer initialized
[    0.207461] Bluetooth: L2CAP socket layer initialized
[    0.207466] Bluetooth: SCO socket layer initialized
[    6.786309] bluetooth hci0: Direct firmware load for qca/rampatch_usb_00000302.bin failed with error -2
[    6.786313] Bluetooth: hci0: failed to request rampatch file: qca/rampatch_usb_00000302.bin (-2)
```

I already tried to add the firmware manually using `apk add linux-firmware-qca` but later found out that it isn't possible to do such operation in Home Assistant OS.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

8.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.
2.
3.
...
-

### Anything in the Supervisor logs that might be useful for us?

```txt
[core-ssh dbus]$ dmesg | grep -i blue
[    0.207441] Bluetooth: Core ver 2.22
[    0.207450] NET: Registered PF_BLUETOOTH protocol family
[    0.207453] Bluetooth: HCI device and connection manager initialized
[    0.207457] Bluetooth: HCI socket layer initialized
[    0.207461] Bluetooth: L2CAP socket layer initialized
[    0.207466] Bluetooth: SCO socket layer initialized
[    6.786309] bluetooth hci0: Direct firmware load for qca/rampatch_usb_00000302.bin failed with error -2
[    6.786313] Bluetooth: hci0: failed to request rampatch file: qca/rampatch_usb_00000302.bin (-2)
```

### Anything in the Host logs that might be useful for us?

```txt
-
```

agners, 2022-09-12:
Will be part of 9.0 release.

closed 2022-09-12 by agners

b8vrede, 2022-09-15:
Just tested the 9.0 release, works perfectly, thanks @agners!