home-assistant/operating-system #2129: HassOS documentation is vague on USB stick config updates via UI

maxromanovsky, 2022-09-13:
### Describe the issue you are experiencing

HassOS documentation is vague on USB stick config updates via UI.
It's not clear how to do it from UI/API, and it's not possible to do it via CLI if you're just enabling SSH.
I remember before configuration UI redesign there was a dropdown to reload config, but I can't find it anymore, even through the new search bar.

Documentation: https://github.com/home-assistant/operating-system/blob/dev/Documentation/configuration.md#automatic

At first I thought to publish it under https://github.com/home-assistant/home-assistant.io, but this documentation is not there AFAIK

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

8.5

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

n/a

### Anything in the Supervisor logs that might be useful for us?

```txt
n/a
```

### Anything in the Host logs that might be useful for us?

```txt
n/a
```

agners, 2022-09-14:
> It's not clear how to do it from UI/API, and it's not possible to do it via CLI if you're just enabling SSH.

Since the UI redesign it is no longer possible. But it is possible via CLI or SSH with the command `ha os import`. I'll adjust the documentation.

maxromanovsky, 2022-09-14:
Thanks! `ha os import` worked like a charm

closed 2022-09-14 by agners