home-assistant/operating-system #2130: xu4 failed upgrading from 9.0.rc2 to 9.0

TonyApuzzo, 2022-09-14:
### Describe the issue you are experiencing

when upgrading from HAOS 9.0.rc2 to 9.0 on my ODroid XU-4 (actually HC1), the system failed to reboot.  I performed multiple power-cycles and it would not boot.

### What operating system image do you use?

odroid-xu4 (Hardkernel ODROID-XU4)

### What version of Home Assistant Operating System is installed?

9.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. be running OS 9.0.rc2 on HAOS on ODroid XU4 with external SSD data partition
2. attempt to upgrade from the notifications in HASS
3. notice that system did not boot
4. login to serial console and power-cycle the system
5. get non-boot log that I posted below
6. tried multiple times with same result
...

### Anything in the Supervisor logs that might be useful for us?

```txt
n/a
```

### Anything in the Host logs that might be useful for us?

```txt
Boot log from the serial console:

root@OpenWrt-Main:~# screen /dev/ttyUSB0 115200,cs8,ixon
/dev/null/utmp: Not a directory
[    5.210885] hub 5-0:1.0: 1 port detected
[    5.212392] mmc0: error -110 reading general info of SD ext reg
[    5.216271] usb usb6: We don't know the algorithms for LPM for this host, disabling LPM.
[    5.221355] mmc0: error -110 whilst initialising SD card
[    5.229245] usb usb6: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 5.15
[    5.242607] usb usb6: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    5.249451] usb usb6: Product: xHCI Host Controller
[    5.254212] usb usb6: Manufacturer: Linux 5.15.67 xhci-hcd
[    5.259733] usb usb6: SerialNumber: xhci-hcd.9.auto
[    5.266711] hub 6-0:1.0: USB hub found
[    5.269195] hub 6-0:1.0: 1 port detected
[    5.276261] mmc_host mmc0: Bus speed (slot 0) = 50000000Hz (slot req 100000Hz, actual 100000HZ div = 250)
[    5.288200] s3c-rtc 101e0000.rtc: rtc disabled, re-enabling
[    5.296003] s3c-rtc 101e0000.rtc: registered as rtc1
[    5.331729] exynos-bus: new bus device registered: soc:bus-wcore ( 88700 KHz ~ 532000 KHz)
[    5.343316] exynos-bus: new bus device registered: soc:bus-noc ( 66600 KHz ~ 111000 KHz)
[    5.354515] exynos-bus: new bus device registered: soc:bus-fsys-apb (111000 KHz ~ 222000 KHz)
[    5.365868] exynos-bus: new bus device registered: soc:bus-fsys2 ( 75000 KHz ~ 200000 KHz)
[    5.378052] exynos-bus: new bus device registered: soc:bus-mfc ( 83250 KHz ~ 333000 KHz)
[    5.391570] exynos-bus: new bus device registered: soc:bus-gen ( 88700 KHz ~ 266000 KHz)
[    5.401349] exynos-bus: new bus device registered: soc:bus-peri ( 66600 KHz ~  66600 KHz)
[    5.412238] exynos-bus: new bus device registered: soc:bus-g2d ( 83250 KHz ~ 333000 KHz)
[    5.422237] exynos-bus: new bus device registered: soc:bus-g2d-acp ( 66500 KHz ~ 266000 KHz)
[    5.429001] mmc_host mmc0: Bus speed (slot 0) = 200000000Hz (slot req 200000000Hz, actual 200000000HZ div = 0)
[    5.432666] exynos-bus: new bus device registered: soc:bus-jpeg ( 75000 KHz ~ 300000 KHz)
[    5.450744] exynos-bus: new bus device registered: soc:bus-jpeg-apb ( 83250 KHz ~ 166500 KHz)
[    5.460406] exynos-bus: new bus device registered: soc:bus-disp1-fimd (120000 KHz ~ 200000 KHz)
[    5.470759] exynos-bus: new bus device registered: soc:bus-disp1 (120000 KHz ~ 300000 KHz)
[    5.480540] exynos-bus: new bus device registered: soc:bus-gscl-scaler (150000 KHz ~ 300000 KHz)
[    5.491735] exynos-bus: new bus device registered: soc:bus-mscl ( 84000 KHz ~ 666000 KHz)
[    5.502264] samsung-i2s 3830000.i2s-sec: DMA channels sourced from device 3830000.i2s
[    5.550411] printk: console [netcon0] enabled
[    5.553360] netconsole: network logging started
[    5.560499] input: gpio-keys as /devices/platform/gpio-keys/input/input0
[    5.579130] ALSA device list:
[    5.580695]   #0: Odroid-XU4
[    5.587820] Waiting for root device PARTUUID=48617373-08...
[    5.635394] mmc0: error -110 reading general info of SD ext reg
[    5.640730] mmc0: error -110 whilst initialising SD card
[    5.748596] usb 5-1: new high-speed USB device number 2 using xhci-hcd
[    6.464379] usb 5-1: New USB device found, idVendor=0bda, idProduct=8153, bcdDevice=30.00
[    6.472101] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=6
[    6.478952] usb 5-1: Product: USB 10/100/1000 LAN
[    6.483271] usb 5-1: Manufacturer: Realtek
[    6.487303] usb 5-1: SerialNumber: 000001000000
[    6.598919] usb usb3-port1: Cannot enable. Maybe the USB cable is bad?
[    7.638589] usb 3-1: new high-speed USB device number 3 using xhci-hcd
[    7.832622] usb 3-1: New USB device found, idVendor=152d, idProduct=0578, bcdDevice=31.01
[    7.839877] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    7.846749] usb 3-1: Product: USB to SATA bridge
[    7.851442] usb 3-1: Manufacturer: JMicron
[    7.855304] usb 3-1: SerialNumber: DB00000000013B
[    7.897544] scsi host0: uas
[    7.909241] scsi 0:0:0:0: Direct-Access     JMicron                   3101 PQ: 0 ANSI: 6
[    7.929557] sd 0:0:0:0: Attached scsi generic sg0 type 0
[    7.946239] sd 0:0:0:0: [sda] 468862128 512-byte logical blocks: (240 GB/224 GiB)
[    7.952706] sd 0:0:0:0: [sda] 4096-byte physical blocks
[    7.960493] sd 0:0:0:0: [sda] Write Protect is off
[    7.968900] sd 0:0:0:0: [sda] Disabling FUA
[    7.971690] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    7.985020] sd 0:0:0:0: [sda] Optimal transfer size 33553920 bytes not a multiple of physical block size (4096 bytes)
[    8.071185]  sda: sda1 sda2
[    8.107755] sd 0:0:0:0: [sda] Attached SCSI disk
[   10.149176] usb usb6-port1: Cannot enable. Maybe the USB cable is bad?
[   14.469038] usb usb6-port1: Cannot enable. Maybe the USB cable is bad?
[   35.683384] vdd_ldo12: disabling
[   35.695218] vdd_ldo16: disabling
[   35.720329] vdd_ldo24: disabling
[   94.234761] random: crng init done
```
```

### System Health information

n/a

### Additional information

Finally I removed the SD card and inserted a newly burned SD card with 9.0 and then it booted, but HASS would not come up on the internal data partition.

NOTE that I have an external SSD /dev/sda1 for data and I had to re-label the SD card data partition in order to get HAOS to start, but once I did that, everything came back to life from my original install.

I suppose this is a low-priority issues since I can't reproduce it anymore, but it is odd that I was able to seamlessly upgrade from HAOS 8.x to the 9.0 rc1 and rc2 without trouble but then failed when trying to do 9.0 release.

agners, 2022-09-14:
Mine seems to have booted fine after 9.0 upgrade, however I do use the internal eMMC.

This errors suggest that a problem with the SD card existed:

```
[    5.635394] mmc0: error -110 reading general info of SD ext reg
[    5.640730] mmc0: error -110 whilst initialising SD card
```

> a newly burned SD card

Was that a different SD card?

> I suppose this is a low-priority issues since I can't reproduce it anymore, but it is odd that I was able to seamlessly upgrade from HAOS 8.x to the 9.0 rc1 and rc2 without trouble but then failed when trying to do 9.0 release.

Indeed, that is quite odd. But other than a new stable kernel (which of course can introduce bugs), there is really not much changed between 9.0.rc2 and 9.0. Did you try to booting from the same SD card again?

closed 2022-09-14 by TonyApuzzo

agners, 2022-09-14:
Home Assistant can be quite demanding on SD cards, especially since we write data to databases and logs continuously. If you have logging enabled, try to decrease severity (e.g. only warning, switch to info/debug only when necessary). Also increasing `commit_interval` on the [recorder](https://www.home-assistant.io/integrations/recorder) can help. OS 9.0 should decrease writes a bit more too as it uses higher commit interval on ext4.

MihaT2021, 2022-09-23:
I downgrade to o.s. 8.5  and ssd is visible. on o.s 9.0 it is not.

https://github.com/home-assistant/operating-system/releases/
Hardkernel ODROID

    ODROID-XU4: Reset Ethernet properly on reboot (https://github.com/home-assistant/operating-system/issues/1488) by @agners in https://github.com/home-assistant/operating-system/pull/2108
    Drop unused cgroup_enable=memory kernel parameter by @agners in https://github.com/home-assistant/operating-system/pull/2124
    Linux: Update kernel 5.15.67 by @agners in https://github.com/home-assistant/operating-system/pull/2126
    Backport patch to fix ODROID-N2 HDMI by @agners in https://github.com/home-assistant/operating-system/pull/2128