home-assistant/operating-system #2134: Update from 8.4 to 8.5 or 9.0 not working

C0D3-IT, 2022-09-15:
### Describe the issue you are experiencing

Hey guys,
when I try to update from 8.4 to 8.5 or 9.0 via gui or ssh it seems to work. There is no error message.

After the device is restarted, the OS version is still 8.4 and the update message appears again.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

8.4

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Initiate update via gui or ssh
2. wait for the system to be restarted
3. the update was not done
...

### Anything in the Supervisor logs that might be useful for us?

```txt
found nothing in the log
```

### Anything in the Host logs that might be useful for us?

```txt
found nothing in the log
```

jherby2k, 2022-09-15:
same issue Pi 4 2GB

agners, 2022-09-15:
I need logs of the previous boot to debug this. Currently, those can only be accessed by accessing the host shell, e.g. via [ssh on port 22222](https://developers.home-assistant.io/docs/operating-system/debugging#ssh-access-to-the-host). If you have access to the host shell, can you share the output of the following command?

```
journalctl -b -1
```

jherby2k, 2022-09-15:
Actually just tried a second time and it worked without issue.

davebeckster, 2022-09-15:
Appear to have same issue with Generic X86-64 installation.  Skill level (so far) proven inadequate to obtain requested logs. 
Home Assistant 2022.9.4
Supervisor 2022.08.6
Operating System 8.5
Frontend 20220907.2 - latest

davebeckster, 2022-09-15:
I have tried many times.  Appears to try to load 9.0 and fall back to 8.5 ( very long boot times).  
Home Assistant 2022.9.4
Supervisor 2022.08.6
Operating System 8.5
Frontend 20220907.2 - latest

agners, 2022-09-16:
Very long boot time appears there is a problem with NTP, possibly DNS. Can you try setting a DNS server manually? E.g.

```
ha network info
``` 

Check the `interface` name, then use
```
ha network update <interface> --ipv4-nameserver 8.8.8.8
```

C0D3-IT, 2022-09-16:
> I need logs of the previous boot to debug this. Currently, those can only be accessed by accessing the host shell, e.g. via [ssh on port [...]

Hey, finally I was able to connect to the host.
Unfortunately theres seems a mistake with the command you gave.
[image]

kukulich, 2022-09-16:
> My raspberry pi wont boot anymore after update from OS 8.5 to 9. Everything is dead :/
> Any idea how to debug or even repair that system in that state?

Same on my Raspberry.

agners, 2022-09-16:
> Hey, finally I was able to connect to the host.
> Unfortunately theres seems a mistake with the command you gave.

That command is correct, but it seems that your device does not store any logs anymore. That is weird! Can you run `journalctl --list-boots`?

ramphex, 2022-09-16:
8.5 to 9.0 broke my VM install on Proxmox. It now only boots into the UEFI Shell.

[image]

agners, 2022-09-16:
@ramphex please open a new issue, as this is a different platform. Also share all the settings of your VM along with the host platform configuration.

davebeckster, 2022-09-17:
As requested, opened new problem for X86.  

➜  ~ ha network info                                                                                                   
docker:                                                                                                                  
   address: 172.30.32.0/23                                                                                                
   dns: 172.30.32.3                                                                                                       
   gateway: 172.30.32.1                                                                                                   
   interface: hassio                                                                                                     
host_internet: true                                                                                                    
interfaces:                                                                                                            
- connected: true                                                                                                        
   enabled: true                                                                                                          
   interface: enp2s0                                                                                                      
   ipv4:                                                                                                                    
     address:                                                                                                                -
     -  192.168.86.27/24                                                                                                      
     gateway: 192.168.86.1                                                                                                  
     method: auto                                                                                                            
    nameservers:                                                                                                            
     - 192.168.86.1                                                                                                          
     ready: true                                                                                                          
   ipv6:                                                                                                                    
     address:                                                                                                                
      - fd06:f20d:c411:0:999f:f980:d871:1b5f/64                                                                               
      - fe80::86fb:ce55:b428:4c4a/64                                                                                         
      gateway: null                                                                                                           
      method: auto                                                                                                            
      nameservers: []                                                                                                        
     ready: true                                                                                                          
   primary: true                                                                                                          
   type: ethernet                                                                                                          
   vlan: null                                                                                                              
   wifi: null                                                                                                            
supervisor_internet: true

C0D3-IT, 2022-09-18:
> journalctl --list-boots

ok, got it. It seems there is only one entry:
`# journalctl --list-boots`
`0 7030e68e84204fb89304440bea41ee25 Fri 2022-03-11 07:33:33 UTC—Sun 2022-09-18 09:40:15 UTC`

The output of this is:
`journalctl -b 0`
[output.log](https://github.com/home-assistant/operating-system/files/9593542/output.log)

davebeckster, 2022-09-18:
Tried your suggestion resetting name server, didn't resolve the issue.  Was
not successful executing "journalctl --list-boots" likely due to lack of
experience using HA CLI.   Can I obtain needed information another way?  As
I experienced some challenges with bios setting on initial Dell install, is
it possible your new release is more sensitive to bios settings than the
8.5?

kukulich, 2022-10-07:
I've still problem to upgrade to 9.0+.

I tried upgrade from 8.5 via SSH -> Raspberry dead.
I tried fresh installation via BalenaEtcher -> Raspberry dead.

Fresh installation of 8.5 - HA is starting.

C0D3-IT, 2022-10-23:
@agners: Any news on this? Do you need more information?

agners, 2022-10-24:
@C0D3-IT hm, that log looks fairly normal, but that is also only from the current boot without an update attempt, so I can't really tell by that why the update fails.

What is really odd is that also old logs are missing. It feels as  if your storage is no longer taking any writes. Is your system running fine otherwise? It could be an indication of a failing SD card. Maybe moving to a new SD card helps.

davebeckster, 2022-10-25:
FYI.  In my HAOS x86 installation, 9.0 will not run if my hard drive is
installed in the chassis (I boot/run  from ssd).  The HD was not being
used.

kukulich, 2022-10-25:
My system looks normal - I've already upgraded HA to 2022.10.5. I'm just not able to upgrade to HAOS 9+. How can I check state of my SSD card?

C0D3-IT, 2022-10-30:
> @C0D3-IT hm, that log looks fairly normal, but that is also only from the current boot without an update attempt, so I can't really tell by that why the update fails.
>
> What is really odd is that [...]

My System works normal, except the update issue.... I moved my data location to an external USB SSD drive. Maybe this is the reason for the log issue?

davebeckster, 2023-01-29:
I've been personally out of commission for a few weeks so I haven't tried
to install the second drive.  Should be able to confirm fix or not tomorrow.

davebeckster, 2023-01-31:
Appears the problem has NOT been resolved.  My x86 (Dell ) system will not
complete loading haos if the hard drive is installed.  This started with
the installation of 9.0.  No changes except simply installation of 9.x.
Booted from ssd before with unused HD installed (future data disk).  Hangs
afterwords.  Remove HD, system boots normally.  Confident HD is fine.   I'm
operating with only ssd today.  Will provide any assistance I'm capable of
but I'm a user not administrator or developer.

davebeckster, 2023-01-31:
Failed to mention, currently running the latest releases of all packages I
use.  No updates available.

C0D3-IT, 2023-05-01:
For me the Bug was not resolved. I reinstalled the whole system a while ago. Now it works again

closed 2023-06-26 by agners