home-assistant/operating-system #2137: OS V9.0 fails to install on X86-64

davebeckster, 2022-09-17:
### Describe the issue you are experiencing

Repeated attempts results in no installation.  The installation process appears to take longer than normal before it reboots back into 8.5.  No error messages.  

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

Operating System 8.5

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Click the 9.0 operating system install button
2.  Wait for system shutdown and restart.
3.  Note that it is still running 8.5
...

### Anything in the Supervisor logs that might be useful for us?

```txt
(I removed many Zigbee2MQTT Warnings, otherwise complete)

22-09-16 18:29:00 INFO (MainThread) [supervisor.api.middleware.security] /backups/d882fd8c/info access from cebe7a76_hassio_google_drive_backup
22-09-16 18:29:00 INFO (MainThread) [supervisor.api.middleware.security] /backups/02bbfe82/info access from cebe7a76_hassio_google_drive_backup
22-09-16 18:29:00 INFO (MainThread) [supervisor.api.middleware.security] /backups/b2102526/info access from cebe7a76_hassio_google_drive_backup
22-09-16 18:29:00 INFO (MainThread) [supervisor.api.middleware.security] /backups/51e23567/info access from cebe7a76_hassio_google_drive_backup
22-09-16 18:29:01 INFO (MainThread) [supervisor.api.middleware.security] /backups/d882fd8c/download access from cebe7a76_hassio_google_drive_backup
22-09-16 18:29:01 INFO (MainThread) [supervisor.api.backups] Downloading backup d882fd8c
22-09-16 18:29:04 INFO (MainThread) [supervisor.misc.tasks] All core tasks are scheduled
22-09-16 18:29:04 INFO (MainThread) [supervisor.core] Supervisor is up and running
22-09-16 18:29:04 INFO (MainThread) [supervisor.host.info] Updating local host information
22-09-16 18:29:04 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
22-09-16 18:29:04 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
22-09-16 18:29:04 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.IPV4_CONNECTION_PROBLEM/ContextType.SYSTEM
22-09-16 18:29:04 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
22-09-16 18:29:04 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
22-09-16 18:29:04 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR
22-09-16 18:29:05 INFO (MainThread) [supervisor.host.services] Updating service information
22-09-16 18:29:05 INFO (MainThread) [supervisor.host.network] Updating local network information
22-09-16 18:29:05 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
22-09-16 18:29:05 INFO (MainThread) [supervisor.host.manager] Host information reload completed
22-09-16 18:29:06 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_FAILED/ContextType.DNS_SERVER
22-09-16 18:29:06 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
22-09-16 18:29:06 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for 
22-09-16 18:29:06 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_IPV6_ERROR/ContextType.DNS_SERVER
22-09-16 18:29:06 INFO (MainThread) [supervisor.resolution.check] System checks complete
22-09-16 18:29:06 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
22-09-16 18:29:08 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-09-16 18:29:08 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
22-09-16 18:29:08 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
22-09-16 18:33:43 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for
```

### Anything in the Host logs that might be useful for us?

```txt
Difficult for me to know what may be useful. Size limitation prevented inclusion of log.
```

### System Health information

## System Information

version | core-2022.9.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.60
arch | x86_64
timezone | America/Los_Angeles
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4896
Installed Version | 1.27.2
Stage | running
Available Repositories | 1121
Downloaded Repositories | 15

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 17

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | February 5, 2023 at 4:00 PM
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
remote_server | us-west-2-0.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.5
-- | --
update_channel | stable
supervisor_version | supervisor-2022.08.6
agent_version | 1.2.1
docker_version | 20.10.14
disk_total | 916.2 GB
disk_used | 113.6 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (10.0.0), Acurite to Home Assistant (0.3.16b), ESPHome (2022.3.1), File editor (5.4.1), Frigate NVR (3.1), Home Assistant Google Drive Backup (0.108.4), Mosquitto broker (6.1.3), SSH & Web Terminal (12.0.2), VLC (0.1.3), Zigbee2MQTT (1.27.2-1), TasmoAdmin (0.20.0), Plex Media Server (3.1.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 7
-- | --
resources | 6
views | 45
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | September 15, 2022 at 12:33 AM
-- | --
current_recorder_run | September 16, 2022 at 6:28 PM
estimated_db_size | 1329.29 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

### Additional information

I am primarily a UI user with limited linux and CLI skills.  I am willing to try any diagnostics that would be helpful but detailed instructions may be needed.

davebeckster, 2022-09-18:
As no one else appears to have this problem, assume its a local issue.  However I could still use some help resolving it.  I realise you need more information and I can access the journal data with DiskInternals Linux Reader but don't know what to look for and its location.  I'm guessing this could be a BIOS setting issue but it hasn't been an issue with previous release.  Appreciate any wisdom available.  These are areas I only dabble in when necessary and certainly have little expertise.

agners, 2022-09-19:
Do you have a screen connected? If so, can you monitor what happens after you click on updating? Is the system rebooting multiple times?

davebeckster, 2022-09-20:
Monitor installed, yes starts to boot, dosent get very far, goes blank,
starts over, comes up in 8.5.  Dies too fast, disappear instantly, cannot
read last last reports.  Need to know how to get you that data from logs.

davebeckster, 2022-09-20:
My logic for suspecting bios is that it is the only thing I personally had
to change to get it to run on X86.  I played with it until it booted.
Never touched it again.   Made several changes based on comments from
unknown sources.   Perhaps some were unnecessary and perhaps detrimental.
Otherwise I do add-ons, integrations, hasc stuff and automation.   No touch
system.

shaowned, 2022-09-22:
I have this problem as well

davebeckster, 2022-10-04:
Could still use some help resolving this one.

davebeckster, 2022-10-09:
I have discovered how to get 9.0 to boot on my system.  I currently boot from a M-2 SSD but in the chassie is also a hard drive.  I used it as a backup drive for a SSD failure (it happened) and eventually intended to use it as a data drive.  It was my only drive before acquiring the SSD.  Under 8.5 I booted and ran from the SSD and the hard drivet showed up as sd0 when I looked at all hardware.  9.0 will not boot with the hard drive present.  Remove it and it boots.  I still have dreams of a data drive.  How should I proceed?

closed 2023-01-14 by github-actions[bot]