home-assistant/operating-system #2139: Invalid JSON in response

dnestico, 2022-09-18:
### Describe the issue you are experiencing

I updated to 2022.08.6 and now my  [view_config.yml.zip](https://github.com/home-assistant/operating-system/files/9592999/view_config.yml.zip) view_config.yaml is busted, I can't see my dashboard anymore. I even went through my configuration.yaml  [configuration.yml.zip](https://github.com/home-assistant/operating-system/files/9593000/configuration.yml.zip) file and tried updating it like crazy and still broken.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

9.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. updated to 9.0
2. dashboard now shows that JSON error

### Anything in the Supervisor logs that might be useful for us?

```txt
Here cuz too long https://pastebin.com/UDdWQ2qt
```

### Anything in the Host logs that might be useful for us?

```txt
Here cuz too long https://pastebin.com/tsVSApJU
```

### System Health information

can't find this HA OS.

### Additional information

The dashboard is gone it just says "Invalid JSON in response" now.

`ERROR (MainThread) [frontend.js.latest.202209072]`

agners, 2022-09-19:
The operating system does nothing to the data partition, where Home Assistant Core and your configuration is located on. So I doubt this is related to the operating system update itself.

It could be due to a file system corruption of some sort, possible caused by a broken SD card. Make sure you create a backup and download it. 

Maybe the Core container somehow corrupted, you can try to recreate it using `ha core rebuild`.

dnestico, 2022-09-19:
I will make an issue on HA core github then thanks.

I should hopefully have a backup too cuz of the google drive addon but how do I do this code thing at the bottom? Will it delete my whole HA?

dnestico, 2022-12-03:
And happened again... help pls

closed 2023-03-03 by dnestico