home-assistant/operating-system #2141: Zigbee Home Automation fails to start after HA OS 9.0 upgrade

fromonesource, 2022-09-18:
### Describe the issue you are experiencing

Upgraded to OS 9.0 and ZHA will no longer start. Error log:

```Logger: homeassistant.config_entries
Source: components/zha/core/gateway.py:172
First occurred: 1:58:29 PM (2 occurrences)
Last logged: 6:07:23 PM

Error setting up entry Zigbee for zha
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy_znp/api.py", line 984, in request
    response = await response_future
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 357, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 112, in async_setup_entry
    await zha_gateway.async_initialize()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 185, in async_initialize
    raise exc
  File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 172, in async_initialize
    self.application_controller = await app_controller_cls.new(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 138, in new
    await app.startup(auto_form=auto_form)
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 118, in startup
    await self.connect()
  File "/usr/local/lib/python3.10/site-packages/zigpy_znp/zigbee/application.py", line 111, in connect
    await znp.connect()
  File "/usr/local/lib/python3.10/site-packages/zigpy_znp/api.py", line 697, in connect
    await self.nvram.determine_alignment()
  File "/usr/local/lib/python3.10/site-packages/zigpy_znp/nvram.py", line 31, in determine_alignment
    rsp = await self.znp.request(c.UTIL.AssocFindDevice.Req(Index=0))
  File "/usr/local/lib/python3.10/site-packages/zigpy_znp/api.py", line 982, in request
    async with async_timeout.timeout(self._znp_config[conf.CONF_SREQ_TIMEOUT]):
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError```

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

9.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Install and configure ZHA
2. Upgrade OS from 8.5 to 9.0
3. Check Integrations page to see "Failed to set up"

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing
```

agners, 2022-09-19:
What kind of Zigbee adapter are you using? Do you use other USB dongles on that system?

Did you use device path with `by-id` (see https://www.home-assistant.io/integrations/zha/#configuration---gui).

jmoransalama, 2022-10-19:
I'm having the same problem, using an Electrolama ZZH. Tried using different usb extension cables and ports but it didn't work. I'm also using another usb port for booting from a hard drive.

Downgrading HA OS with
`ha os update --version 8.5`
fixed the issue for now.

nigotiator, 2022-12-21:
Yep, just updated from 8.2 to 9.4 and it didn't find the conbee2 stick. removed the integration and added again and now all devices are renamed... Need to restore the full backup for now... I'm using the label:board/generic-x86-64 tho...