home-assistant/operating-system #2142: Bluetooth not working on virtualbox VM

ZTHawk, 2022-09-18:
### Describe the issue you are experiencing

Bluetooth is not working with HA.
Using `bluetoothctl scan on` I get "No default controller available"
In host console I see:
```
Bluetooth: hci0: command 0xfc01 tx timeout
Bluetooth: hci0: FW download error recovery failed (-110)
```

I tried an image with HAOS 7.6. Using the command `bluetoothctl scan on` there I get "Discovery started" followed by a list of all found devices.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

9.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Start a VM with attached Bluetooth device.
In host console I get:
```
Bluetooth: hci0: FW download error recovery failed (-110)
```

When disabling the Bluetooth device and then reenabling it I get the following in host console:
```
Bluetooth: hci0: command 0xfc01 tx timeout
Bluetooth: hci0: FW download error recovery failed (-110)
```

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

### System Health information

N/A

### Additional information

N/A

agners, 2022-09-19:
What make/type of Bluetooth device are you using? You are saying 7.6 used to work still, did you try HAOS 8.5?

agners, 2022-09-19:
-110 means timeout. It seems the device somehow does not respond. Maybe removing it and plugging back in helps?

ZTHawk, 2022-09-19:
I just did some more tests:
1) HA OS 7.6 and Core 2022.6.7 -- Bluetooth scan `bluetoothctl scan on` is working (it can find devices)
2) HA OS 7.6 and Core 2022.9.5 -- Bluetooth scan is working
3) HA OS 8.0 and Core 2022.9.4 -- Bluetooth scan is NOT working (No default controller available)
4) HA OS 8.0 and Core 2022.6.7 (after downgrade) -- Bluetooth scan is NOT working
5) HA OS 8.5 and Core 2022.6.7 -- Bluetooth scan is NOT working

ZTHawk, 2022-10-08:
Just for completness: My bluetooth device (integrated in wireless module): Intel Wireless AC 9560

ZTHawk, 2022-10-26:
Update: HA 9.2 (an 9.3) + Core 2022.10.5 -- Bluetooth scan is NOT working

ZTHawk, 2022-12-15:
I have bought a separate Bluetooth-Stick to fix this issue but this resulted in a weird behaviour.
The moment I attached the Bluetooth-Stick to windows host, the integrated bluetooth module started to work within HA. Windows host had bluetooth via the new stick. So suddenly everthing was working fine.
I have removed the bluetooth stick. HA was still fine. I have not reattached the stick.

I have then upgraded to OS 9.4 and also upgraded to 2022.12. Still everything was fine.

I have restarted the host and integrated bluetooth module is not working in HA anymore. Tried to reproduce the behaviour with the stick but with no luck.

If I unassign the bluetooth module (on Intel Wifi chip) and then I assign the stick to HA then it crashes.
If I try to restart it, it will crash.
Switching back to integrated bluetooth allows HA to start again but with no bluetooth.

PS: 7.6 is working but 8.0 RC1 is not.

ZTHawk, 2023-01-29:
Update: HA OS 9.5 (an 9.4) + Core 2023.1.7 -- Bluetooth scan is NOT working

HA OS 7.6 is the last working version. Any vesion after that has bluetooth issues.
Tested with clean installations.

ZTHawk, 2023-04-18:
Update: HA OS 10.0 + Core 2023.4.4 -- Bluetooth scan is NOT working

ZTHawk, 2023-04-27:
Update: HA OS 10.1 + Core 2023.4.6 -- Bluetooth scan is not working

[image]
When Bluetooth device "(Intel Corp.) [0002]" is not selected in USB device filters (see screenshot above) and I then try to add the Bluetooth device (see screenshot below)
[image]
Then I get the error:
```
Bluetooth: hci0: FW download error recovery failed (-19)
```

When adding the tick in the first screenshot and then trying to add the Bluetooth device like in second screenshot, then I get the error:
```
Bluetooth: hci0: command 0xfc01 tx timeout
Bluetooth: hci0: FW download error recovery failed (-110)
```

ZTHawk, 2023-06-01:
Update: HA OS 10.1 + Core 2023.5.3
Still not working

madbrain76, 2023-06-06:
I am seeing the same problem with an Intel Bluetooth controller on my X560 TUF Gaming Plus motherboard, Virtualbox 7.0.8 and latest HAOS 10.2 .

agners, 2023-06-06:
Hardware initialization/firmware is something handled by the kernel. What would be interesting is if the same error appears on a different distribution (e.g. Arch Linux with the LTS kernel installated). If that is the case, you could try report this to the Linux kernel folks (e.g. via https://bugzilla.kernel.org/).

ZTHawk, 2023-06-20:
Update: HA OS 10.2 + Core 2023.6.1
> Still not working

Installed OS 10.3 + Core 2023.6.2 today. Had issues with system not responding. Did a hard reset. After everything started up I have added the bluetooth device. It is working now.
> Device was not in USB filter

Next I have added it to USB filter and I have done a shutdown. Booted the VM: Bluetooth still working.

Rebooted host system. Result in HA: Bluetooth still working.

I will not yet see this as fixed untill a longer period of runtime/more restarts (see above comments where it was working once for a short time and then stopped again).

intercrypto, 2023-06-20:
Hi, @ZTHawk.

I did my upgrade now and the bluetooth does not work. My versions are the same. I have tested with and without USB filter in Virtualbox.

I´m using a U59 Pro Intel mini computer. The bluetooth works in the Windows host but in VM HA does not work.

Please, some help.

ZTHawk, 2023-06-21:
@intercrypto 
Unfortunately I cannot help. It just suddenly started to work.
That is also the reason why I do not see it as fixed yet.

PS: My installed drivers are (Asus PN41):
Wifi: 22.220.0.4 (date: 29.03.2023)
BT: 22.220.1.1 (date: 18.04.2023)

remme123, 2023-07-20:
> I have bought a separate Bluetooth-Stick to fix this issue but this resulted in a weird behaviour. The moment I attached the Bluetooth-Stick to windows host, the integrated bluetooth module started [...]

I successfully started Bluetooth in VBox using the same method. My environment is VBox running on Windows 10, with the virtual machine running ArchLinux.

I speculate that the issue is caused by Windows 10 occupying the Bluetooth device, which results in it being also occupied when assigned to VBox.

ZTHawk, 2023-08-22:
Today (at least I noticed today) it stopped working.
With a new additional error:
```
Bluetooth: hci0: command 0xfc01 tx timeout
Bluetooth: hci0: FW download error recovery failed (-110)
Bluetooth: hci0: Reading Intel version command failed (-32)
```

ZTHawk, 2023-08-22:
Reverting back to HAOS 10.3 and driver 22.220.1 fixed the issue.

nnathan, 2023-09-02:
Not a Home Assistant user, but I was trying to redirect my Intel AX210 Bluetooth USB device to a VM using virt-manager in a stock version of Ubuntu 22.04 Linux. It kept failing with the same error messages as OP posted.

The bandaid solution was to download the latest Linux firmware for iwlwifi which I think is used to transfer firmware to the AX210 BT device (*I think*). Just download the latest firmware [from here](https://cdn.kernel.org/pub/linux/kernel/firmware/) and copy the extracted contents to `/lib/firmware`. Then I tried redirecting the USB device and it succeeded. In the case of OP you might need to either reboot or use a tool like `usbreset` to reset the device so that the firmware can load.

ZTHawk, 2023-12-02:
Still applicable

bobvmierlo, 2024-01-28:
Running into the same issue here. Running HAOS 11.4, Supervisor 2024.01.0 and Core 2024.1.5 inside a Virtualbox VM on an Intel NUC (NUC13ANHi5) and using [this bluetooth dongle
](https://www.trust.com/en/product/18187-manga-bluetooth-4-0-adapter) from Trust

Some lines from the host log:
```
Jan 28 20:03:52 homeassistant systemd[1]: Stopped target Bluetooth Support.
Jan 28 20:03:52 homeassistant kernel: usb 1-2: reset full-speed USB device number 5 using xhci_hcd
Jan 28 20:03:52 homeassistant kernel: usb 1-2: unable to get BOS descriptor or descriptor too short
Jan 28 20:03:52 homeassistant systemd[1]: Reached target Bluetooth Support.
Jan 28 20:03:52 homeassistant kernel: Bluetooth: hci0: Reading Intel version command failed (-32)
Jan 28 20:03:53 homeassistant kernel: Bluetooth: hci0: Reading Intel version command failed (-32)
Jan 28 20:03:58 homeassistant systemd[1]: Stopped target Bluetooth Support.
Jan 28 20:03:58 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-3639b5d244c015189d3d78972a017ba2d46ef650763710c2ed8db2d33b57e433-runc.AKRUYk.mount: Deactivated successfully.
Jan 28 20:03:58 homeassistant kernel: usb 1-2: reset full-speed USB device number 5 using xhci_hcd
Jan 28 20:03:58 homeassistant kernel: usb 1-2: unable to get BOS descriptor or descriptor too short
Jan 28 20:03:58 homeassistant systemd[1]: Reached target Bluetooth Support.
Jan 28 20:03:58 homeassistant kernel: Bluetooth: hci0: Reading Intel version command failed (-32)
```

txteclipse, 2024-02-20:
I believe I am also experiencing this problem. Bluetooth device is a mobo-integrated Intel module. Host OS is Windows 10 v. 22H2. These are the errors when I use the usb device filter in VirtualBox.
[image]

ZTHawk, 2024-05-15:
Switched to Proxmox with HA as VM. Passed Bluetooth device to HA. Restored from backup.
Bluetooth was working.
Restarted HA virtual machine, Bluetooth is not working anymore.

ZTHawk, 2024-07-06:
Test done with Proxmox:
1. Removed bluetooth from HA
2. Assigned to Windows 11 (VM in Proxmox)
3. Bluetooth is working fine (devices can be found)
4. Removed from Win11
5. Assigned to HA
6. Bluetooth is not working

I get this message the moment I connect the bluetooth device to HA:
[image]

behold81, 2024-10-08:
I am having the same issue with the following BT module. it was working and now its not. 8087:0aaa Intel Corp. Bluetooth 9460/9560

Using Proxmox. BT is blacklisted in the host, VM is using q35 and the USB Passthrough added. I see the device in the VM but will not initialise with:
[image]

I have tried alot to sort this with no hope. using 2024.10.0 and 13.1 OS.

mrdunu, 2024-11-23:
Also having a similar issue with Home Assistant OS 13.2 and core-2024.11.3 using the onboard Intel NUC bluetooth:

Bluetooth: hci0: Reading Intel version command failed (-32)

fofyga, 2024-11-25:
same issue with bluetooth dongle on 13.1

jellekropman, 2024-12-02:
> Test done with Proxmox:
> 
>     1. Removed bluetooth from HA
> 
>     2. Assigned to Windows 11 (VM in Proxmox)
> 
>     3. Bluetooth is working fine (devices can be found)
> 
>     4. Removed from Win11
> 
>     5. Assigned to HA
> 
>     6. Bluetooth is not working
> 
> 
> I get this message the moment I connect the bluetooth device to HA: ![grafik](https://private-user-images.githubusercontent.com/23618799/346289437-f6f2851b-3523-43ca-bc97-512e43c8bc3e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMxNTM3NjgsIm5iZiI6MTczMzE1MzQ2OCwicGF0aCI6Ii8yMzYxODc5OS8zNDYyODk0MzctZjZmMjg1MWItMzUyMy00M2NhLWJjOTctNTEyZTQzYzhiYzNlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjAyVDE1MzEwOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRmNDc3OGQ0NzdiOWI3YTJkZGI0NTEwZmI5OTliNzEyZmM3OTY5NzU5ZTEyOTU3MDYxY2I0MjJmOTQ1YjkxNGImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Hj5kV7vEuZrzRf4zqqlKyLVZ2u46laxHyy0Ef0u2lN0)

Found the solution in this post on the proxmox forum:
https://forum.proxmox.com/threads/pve-7-4-home-assistant-intel-ax200-bluetooth-issues.126014/post-654997

This worked for me!

madbrain76, 2024-12-06:
I got my Intel 9260 bluetooth controller to work in Home Assistant under VMWare. However, it only works if I start HAOS from the GUI. If I start is as a service, which is the way I normally do, the VM fails to see the device. I have 3 USB devices attached to the VM successfully, even when running as a service. But for some reason, the Bluetooth fails.

Is anyone else running VMWare ?

I posted in the VMWare community, about this 3 days ago, but it appears to be a very sleepy one.

https://community.broadcom.com/vmware-cloud-foundation/question/intel-9260-bluetooth-controller-not-visible-if-vm-is-running-as-a-service

madbrain76, 2024-12-17:
I also tried it on Virtualbox 7.1.4 . I am getting similar symptoms as when running HAOS under VMWare as a service. The Bluetooth device bounces between the guest and the host. Ultimately, control returns to the host, and it shows up in Device manager once again.

The only configuration that I did get it to work in is using the VMWare GUI, not running the VM as a service.

Nilossy, 2025-03-26:
Having this same problem; basically copy and paste txteclipse's screenshot, exact same readout. Tried a hundred things and just throwing different errors that amount to failure to detect the adapter.

Version: 2025.3.4, Windows 10 -> VirtualBox
Adapter: Intel ax210
HAOS accurately detects the Intel adapter enough to show the prompt to add BT integration, but then simply says "Failed Setup".

madbrain76, 2025-04-01:
I tried with an Asus BT 4.0 USB adapter. I tried it in Virtualbox, VMWare Workstation Pro (using GUI), VMWare Workstation Pro (as a service) on a Windows 11 host. It only works in the second case, with VMWare and launching the VM from GUI.
I suspect there is some kind of contention between the Windows host OS and VM for the bluetooth controller. Has anyone tried on a Linux / Mac / Solaris host ?

Pascal66, 2026-01-01:
Up. Shame. 4 years without any solution.